OpenEtherCATsociety/SOEM #442: SOEM on control servo(IS620N), Can't change servo from SafeOP to OP

summerlotus513, 2020-08-31:
I'm trying to use SOEM to control the servo(IS620N), but when I set the servo to OP, It couldn't set the servo to the OP state. The servo is in safeop and program exit with "No error".
Is there something wrong with my process,? Can someone give me a code reference that can completely enable servo? 
Be deeply grateful!
[my slaveinfo]
[image]
[my program print]
[image]

ArthurKetels, 2020-08-31:
Simple_test is not a good program to control servo slaves. I would suggest to read the tutorial. Also read the slave manual to see the exact requirements to enable operation. Some slaves need the sync0/1 pulses to be active and a master that send PDO packets synchronized with the sync pulses.

Where you able to control simpler slaves like digital I/O or analog outputs with success?

summerlotus513, 2020-09-01:
> Simple_test is not a good program to control servo slaves. I would suggest to read the tutorial. Also read the slave manual to see the exact requirements to enable operation. Some slaves need the [...]

Yes, I try update the simple_test to control the slave(AX58100 & LAN9252) in IO,it work.
But when I refer to some programs on the network to guide my servo(IS620N) ，it didn't work. I tried a lot of programming，but failed.
Somebody tell me when servo in safeop, I should enable the servo, then change to OP.
My servo follow the standard cia402 protocol. (The slave manual say: Boot the servo driver according to the standard cia402 protocol).
[image]
I tried, but faild. servo can't be enabled in safeOP. 
So, I want konw, how to change servo(CiA402) to OP,or how to enable the servo.....Are there any procedures or information that I can refer to...
Thank you so much.

toshisanro, 2021-08-05:
You need to change the   ecx_context.manualstatechange  value  from  0 to 1.  
[image]

toshisanro, 2021-08-05:
@summerlotus513   I have been engaged in the development of EtherCAT master （SOEM） for more than two years.Based on soem, I have realized multiple master , and I  can control Panasonic （A6B），elmo，STEP，inovance (IS620N)，tsino-dynatron（cooldrive  rc)  servo driver ,and beckhoff  IOs ( EK1100 ,EL1008,EL2008 etc). If you have technical requirements, please contact me.

gvcode1, 2021-08-15:
@toshisanro Hi, how to contact you? Do you have email?

toshisanro, 2021-08-16:
@gvcode1 my  email : toshisanro@163.com

nakarlsson, 2022-01-14:
close due to inactivity

closed 2022-01-14 by nakarlsson

brunogold, 2024-12-13:
@summerlotus513  were you able to get this resolved. I use the same AC drive and I believe I am in the same state :) Appreciate any help on this.