OpenEtherCATsociety/SOEM #444: Controlling Panasonic MINAS A6 Servo Controller

Houman-HM, 2020-09-09:
I have recently acquired a Panasonic MINAS A6 to control a servo motor.
I have managed to run the simple_test and slave reaches operational without any errors.
I'm completely new to EtherCATs and as well as SOEM.
Using the EC-Engineer software, I put the controller to operational mode and then change the PDO for operation mode to 1 and the PDO for for control word to 0x0F. After setting the control word to 0x0F, the relay in the driver switches on and then I can write the PDO for Target Position.

I want to achieve the same scenario using SOEM, however, there is no tutorial available on how to do this.
So far, I have used the ec_SDOwrite(1, 0x1600, 0x02, FALSE, objectSize, &objectValue, EC_TIMOUTRXM) to write the PDOs and it does return 1 after writing each Service data object. However, it does not switch on the relay and finishes without any errors. 
I confirm that in case an error happens, I can clear the error by changing the control word to 0x86 using ec_SDOwrite.
I was wondering if you can guide me through how to switch on the relay in the driver so that I can control the motor. 

Furthermore, how's it possible to have the controller in the operational mode constantly? 
For example, in the simple_test node, the driver is set to operational and some process data are sent and received and then, it is set to INIT state, after the which the socket is closed and the program would end. Is there a way to keep the connection on and have the driver always in the operational mode? (Because, using EC-Engineer software, once you put the EtherCAT to Operational mode it stays on the same mode until and error happens or is manually to go another state).

I truly appreciate it if you could guide me through this.

ArthurKetels, 2020-09-09:
There is a tutorial available. Obviously named tutorial.txt ;-)
You do not use SDO's to transfer PDO's. First read up on the basic principles of EtherCAT.

Houman-HM, 2020-09-09:
Hey @ArthurKetels,  thanks for the quick reply. 
I have gone through the tutorial you're referring to and that's how I have reached this level. I am really new to using EtherCATs and I yet to grasp it properly. 
I would try to read more about EtherCATs, hoping to be able to solve the probelm. 

Thanks again and sorry if I caused any inconveniece by asking those trivial questions.

Houman-HM, 2020-09-19:
Thanks for your suggestions @ArthurKetels, I managed to get it working by reading more about the concepts of EtherCAT and how  the PDO are transferred.

closed 2020-09-19 by Houman-HM