OpenEtherCATsociety/SOEM #445: SM2 and SM3 seem inverted

alessio76, 2020-09-14:
Hi everybody,
I'm trying to create a master application for the Mecademic Meca500 robot but even the test tools like slaveinfo, simple_test or red_test give errors. The error I get, after have reached SAFE_OP state, is: 

SOEM (Simple Open EtherCAT Master)
MECA test
Starting MECA test
ec_init on eno1 succeeded.
1 slaves found and configured.
..
Slave:1 Name:Robots Output size:1056bits Input size:288bits State: 2 delay:0.1
Calculated workcounter 3
Slave:1 Name:Robots Output size:1056bits Input size:288bits State: 4 delay:0.1
Request operational state for all slaves
Not all slaves reached operational state.
Slave 1 State=0x04 StatusCode=0x0000 : No error
End MECA test, close socket
End program
[master_meca_test_wireshark.pcapng.txt](https://github.com/OpenEtherCATsociety/SOEM/files/5217799/master_meca_test_wireshark.pcapng.txt)

While with simple_test and red_test I get 
Not all slaves reached operational state.
Slave 1 State=0x12 StatusCode=0x0011 : Invalid requested state change.

Here are the results from the slaveinfo -sdo e -map commands. 
[slaveinfo_nopatch_map.txt](https://github.com/OpenEtherCATsociety/SOEM/files/5217681/slaveinfo_nopatch_map.txt)
[slaveinfo_nopatch_sdo.txt](https://github.com/OpenEtherCATsociety/SOEM/files/5217682/slaveinfo_nopatch_sdo.txt)

Here you can see that the object 0x1c00 have the values 2,1,3,4 and, for some reason, the first three subindices of the object 0x1c13 (SM3) become 0 even if I've just mapped them.

It seems, reading other issues, that SM2 and SM3 are inverted, so I tried the advice of issue #224, but this didn't solve the problem, so I'm planning to contact mecademic for support. Are there any other solutions besides fixing the firmware? 
Thanks in advance.

ArthurKetels, 2020-09-15:
What advice did you implement? With the current firmware in your slave, SM2 and SM3 are reversed. So either update the slave firmware or patch SOEM as per example in #224. Looking at the wireshark dump you did not implement either.

alessio76, 2020-09-16:
After Mecademic response, I've succesfully implement the patch of comment #224; even if is only temporary is the best solution we have for now waiting for a definitive fix in the slave stack. 
Thanks for your quick reply.

closed 2020-09-16 by alessio76