OpenEtherCATsociety/SOEM #446: Servo control issue, relay clicking, beginner

SamuelJamesFisher, 2020-09-15:
Hello, I'm new to SOEM, trying to control actuator with a Panasonic minas a6 servo driver. My issue is that the relay is clicking on and off, I know there will be something simple wrong in my code which i have been bashing my head on for the last week. Please could you take some time out of your day and have a look at this hideous beast.

```
void CALLBACK RTthread(UINT uTimerID, UINT uMsg, DWORD_PTR dwUser, DWORD_PTR dw1, DWORD_PTR dw2)
{

    if (inOP)
    {
        
        enableSlave(1);
        slaveOutputArray[1]->targetPos = 10;
        homing(1);
    }

    IOmap[0]++;

    ec_send_processdata();
    wkc = ec_receive_processdata(EC_TIMEOUTSAFE);
    rtcnt++;
   
    //runMotor();
    //15 14 13 12 11 10 9 8 7 6 5 4 3 2 1 0
}
```
```
bool enableSlave(int slave)
{
        //int sizeOfInt = sizeof(uint16_t);
        //uint16_t test;

        printf("wkc %d", wkc);

        uint16_t buffer = slaveInputArray[slave]->StatusWord; //ec_slave[1].inputs[1];

        for (int i = 15; i >= 0; --i) {
            printf("%d", (bool)(buffer & (1U << i)));
        }

        printf("\nstatusword = %#x\n", (buffer & 0x4f));
        printf("statusword = %#x\n", (buffer & 0x6f));

        //0123 4567 89 10 11 12
        //0000 0010 00 1  1  1 000
        //15 14 13 12 11 10 98 7654 3210
        //0  0  0  1  0  1  10 001 1 0111

        if ((buffer & 0x4F) == 0x00)
        {
            printf("off 1 wait\n");
        }
        else if ((buffer & 0x4F) == 0x40)
        {
            printf("switch on disabled\n");
            slaveOutputArray[slave]->controlWord = std::uint16_t(0x06);
        }
        else if ((buffer & 0x6F) == 0x21)
        {
            printf("ready to switch on\n");
            slaveOutputArray[slave]->controlWord = std::uint16_t(0x07);
        }
        else if ((buffer & 0x6F) == 0x23)
        {
            printf("enable op\n");

            //slaveOutputArray[slave]->targetPos = slaveInputArray[slave]->pos + 20;
            
            slaveOutputArray[slave]->controlWord = std::uint16_t(0x0F);
        }
        else if ((buffer & 0x6F) == 0x27)
        {
[... 281 lines cut ...]
 FMMU1 Ls:00000029 Ll:  23 Lsb:0 Leb:7 Ps:1600 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 256 rd: 256 MBX protocols : 04
 CoE details: 0f FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0009.0] 0x6040:0x00 0x10 UNSIGNED16   Controlword
  [0x000B.0] 0x6060:0x00 0x08 INTEGER8     Modes of operation
  [0x000C.0] 0x607A:0x00 0x20 INTEGER32    Target position
  [0x0010.0] 0x60B8:0x00 0x10 UNSIGNED16   Touch probe function
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0029.0] 0x603F:0x00 0x10 UNSIGNED16   Error code
  [0x002B.0] 0x6041:0x00 0x10 UNSIGNED16   Statusword
  [0x002D.0] 0x6061:0x00 0x08 INTEGER8     Modes of operation display
  [0x002E.0] 0x6064:0x00 0x20 INTEGER32    Position actual value
  [0x0032.0] 0x60B9:0x00 0x10 UNSIGNED16   Touch probe status
  [0x0034.0] 0x60BA:0x00 0x20 INTEGER32    Touch probe pos1 pos value
  [0x0038.0] 0x60F4:0x00 0x20 INTEGER32    Following error actual value
  [0x003C.0] 0x60FD:0x00 0x20 UNSIGNED32   Digital inputs
End slaveinfo, close socket
End program

nakarlsson, 2020-09-15:
One common mistake for this sample code is not removing the cyclicly adding of ++ to IOmap byte 0. For you it means the controlword of drive 1 will alter every RTthread call. 
If you also experience issues with drive 2, a wireshark would help, NOTE: Windows is not ideal when it comes to realtime, and you might experience "glitches" in communcation on NICs.

void CALLBACK RTthread(UINT uTimerID, UINT uMsg, DWORD_PTR dwUser, DWORD_PTR dw1, DWORD_PTR dw2)
..
IOmap[0]++;  <--- this will alter your slave 1 control word

nakarlsson, 2020-10-23:
Still a problem?

SamuelJamesFisher, 2020-10-25:
Sorry for the delay. I have solved it. "IOmap[0]++" Was not helping. Other things I learned to help others out, Don't bother trying to do sync or any Realtime stuff with a windows laptop, Panasonic minas a6 servo driver and probably others need constant communication, READ READ the servo driver ethercat manual THOROUGHLY there could be a small line that you hadn't done which causes the whole thing not to work.

closed 2020-10-25 by SamuelJamesFisher