OpenEtherCATsociety/SOEM #447: AX5206 and SOEM

2b-t, 2020-09-19:
Hey everybody,
I am trying to use SOEM in combination with a **Beckhoff AX5206**. I have already seen that there are a few post concerning this topic: An [issue](https://github.com/OpenEtherCATsociety/SOEM/issues/127) over here has been closed and re-opened several times but seems to be never have been fully resolved but it seems like [LinuxCNC was able to get it to work](https://forum.linuxcnc.org/24-hal-components/37643-ethercat-ax5206-stuck-in-safeop).

From what I have understood so far, the particularity of this drive seems to be that it requires a **start-up list** to be sent and furthermore it takes quite a while to **transition between the different states**.
What I have tried so far is modifying the SOEM example code (actually a C++ wrapper for it) by including a routine that reads the start-up list from TwinCAT (`.xml` or `.csv`), extracts the data to be written, saves it to a vector
`std::vector<uint8> data`
and uses the IDN to extract the the drive number (A = 0, B = 1) as well as the corresponding IDN number and maps the corresponding (S) or (P) IDNs to a variable called `mappedIdn` which considers the offset of `UINT16_MAX/2+1 = 32768` for the P-parameters. Then they are ordered by their transition I->P, P->S, S->O and send to the slave with [SoE write](https://openethercatsociety.github.io/doc/soem/ethercatsoe_8c.html#ae7ea660f01cb95ab4f37479bbc2f017e) commands of the form
`ec_SoEwrite(slaveId, driveNo, 64, mappedIdn, data.size(), data.data(), EC_TIMEOUTRXM );`
I have tried to play around with the timeouts, included pauses in between the different commands and after them, tried to send the data as small and big endian but the drive does not reach `SafeOP`. 

Has somebody been able to get this type of drives to work with SOEM and or has some general advice on how to proceed?
*Any help is highly appreciated!*

ArthurKetels, 2020-09-20:
First thing to do is to make a wireshark capture of the start-up sequence. Perhaps some part did not succeed. Did you check all return values from the ec_SoEwrite() functions?

Further, why do people choose AX5 series drives? We all know how difficult they are to use. There are so many drives that give a much better user experience.

Any drive that takes seconds to go in the next state is broken by design in my opinion.

2b-t, 2020-09-20:
Hey @ArthurKetels
*thanks a lot* for your reply and your efforts on SOEM in general.

Sadly I cannot choose other drives. It is an existing prototype system that previously has been running on TwinCAT and now should be ported for more flexibility.

I have first tried it with what @robochime has listed as a *minimal configuration*: S-0-0001 (A), S-0-0002 (A), S-0-0015 (A, B), S-0-0016 (A, B), S-0-0024 (A) and (B), S-0-0032 (A, B) as well as P-0-0010 (A), P-0-0201 (A), P-0-0202 (A), P-0-0203 (A), P-0-0204 (A), P-0-0304 (A, B). With such a start-up list I am able to bring a drive that has already been powered up and sent the full start-up list at some earlier point to OP in TwinCAT. (Nonetheless such a minimal start-up list is not able to bring a drive that has been powered off to OP: In such a case it will also be stuck at PreOP!)

When using SOEM the ec_SoEwrite succeeds for all of the essential SOEs listed above and I can read out the values successfully with ec_SoEread but yet the drive is not able to go beyond PreOP. If I send the full start-up list I just noticed that quite a few of them actually do not succeed: *Only 68 out of 106 return a success*.

ArthurKetels, 2020-09-20:
It is not unthinkable that some parameter changes in the drive take a bit of time. If you then would continue uploading other parameters it could be that they are dropped AFTER the mailbox protocol has accepted them. You can not assume there is a proper queue for parameter changes in the drive. Twincat has a very specific timing for uploading parameters to the drive.

I recommend either copying the timing or do a parameter read check after each write.

2b-t, 2020-09-20:
Thanks a lot @ArthurKetels for the advice. I will look into it.

2b-t, 2020-09-25:
@ArthurKetels I finally had a chance to look into it and figured out I had made an error with reading the parameters from the start-up list. Now every single `ec_SoEwrite` returns success but if I try to read the values again using `ec_SoEread` I can see that the _values for the IDNs S-0016 (A, B), S-0024 (A, B), P-0053 (A, B) and P-0054 (A, B) are actually not written successfully_. I have tried to fix it by sending the commands in a loop with variable timings and different numbers of max iterations but it would not succeed anyways.

ArthurKetels, 2020-09-25:
It would help if you could post the start-up list and also a wireshark trace of the configuration.

The S-0016 and S-0024 are a problem because they are the AT and MTD mapping. I do not know if you can actually read them back. But what is send, and what is read back? Are you sure you are in the right EtherCAT state (pre-OP)?

The P-0053 and P-0054 are just ascii strings describing the motor and drive names. It does not hurt.

If you do not already know it, a complete description of IDNs and error messages for the AX5000 series:
https://infosys.beckhoff.de/english.php?content=../content/1033/ax5000_diag_messages/7912810507.html&id=6197924645771836358

2b-t, 2020-09-25:
Hey @ArthurKetels 
_thanks again_ for your quick reply... Here is the start-up list as csv-file [startup_list.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5283755/startup_list.zip) but I can only add the Wireshark trace mid of next week.

I issue all the `ec_SoEwrite` commands in _pre-OP_ which I checked with SOEM as well as manually on the built-in display. For every single command
- I write the IDN with `ec_SoEwrite` making sure I use the correct data type and Endian (a vector of `uint8` as well as `uint16` or `uint32` values) and check if the resulting write working counter is > 0.
- I check with `ec_SoEread` if it has succeeded by again making sure the read working counter is > 0 and that the read data equals the previously written data.

This works for all IDNs apart from the ones mentioned before: S-0-0016, S-0-0024, P-0-0053 and P-0-0054. In this case the write command returns a working counter > 0 but If I read them out again I get a working counter > 0 but values that are different from 0x00 ... but are not the same as the ones I have written.
Then I set the cycle times 0 and 1 to 250000 and 2000000 ns (which is the same as S-0-0001 and S-0-0002 and a multiple of Sync1 as required) respectively with a cycle shift of 0 using `ec_dcsync01`.

If I read out the last error codes from IDN P-0-0300, I get several different ones:
- _F408, F16D, F11F, F4A5_ which seem to be all in one way or the other related to the two IDNs S-0-0016 and S-0-0024 not being set, which are responsible for the AT and MTD mapping.
- _FC03, FD11, FD43_: Undervoltage (?)
- _F415_: Some error regarding distributed clocks (but I think I also had that with TwinCAT3 from time to time).
- _F11F_: "The advanced parametrization introduced with firmware 2.10 is required" (I am running version 2.12) which seems to be a warning only and not an error.

_Thanks_ again for assisting me with this slightly off-topic question.

2b-t, 2020-10-01:
Hey @ArthurKetels
I finally had the opportunity to take wireshark traces of sending the start-up list to a single AX5206 drive with **Beckhoff TwinCAT3** (Windows 10), as well as with **SOEM** (same computer but on Ubuntu 16.04), one where I only send the data to the drive **without reading the corresponding IDNs out again** and one where I try to **read out the IDNs afterwards** to see if the sending succeeded. Here the corresponding traces:

[WiresharkTraces.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5313133/WiresharkTraces.zip)

For these particular traces I sent every IDN only once with 100 ms between them in the order given by the .csv-file in PreOP. I have already experimented with sorting them, grouping them, writing with several thousand retries, increasing the timeouts by a factor of 1000, introducing pauses of up to 10 seconds in between write commands, moving the distributed clock commands before and after sending the IDNs but sadly none of it seems to help: IDNs S-0016 (A, B), S-0024 (A, B), P-0053 (A, B) and P-0054 (A, B) still seem to fail...
When inspecting the EtherCAT telegrams (`ecat_mailbox.soe_idn == `) I can that at least see that in all of the three files (TwinCAT vs SOEM) there are messages with the same type (FPWR - configured address write) that contain the same SoE data but the messages themselves else are quite different.

ArthurKetels, 2020-10-01:
Thanks for the wireshark traces. Really helpful.

1) The SoE commands are not in the correct order according to your start-up list. First do all IP transition parameters, then the PS transition parameters. TwinCAT does it this way too. P-0-0010 comes first.
2) IDNs S-0016 (A, B), S-0024 (A, B), P-0053 (A, B) and P-0054 (A, B) are correct. They are all record IDNs with Actual Length and Max Length. No matter what you write to them, when you read the IDN they always have the firmware Max Length and NOT your Max Length. Except for this field all other fields ARE the same. So here you have been chasing the wrong rabbit.
3) BUG in ecx_readIDN map(). Already fixed and merged in SOEM master. The mapping size started from zero again at each new drive number. Please pull latest. 
4) After configuring and mapping the slaves, but before requesting SAFEOP, start synchronous PDOs and SYNC0/1. Then request SAFEOP. I am not sure this is needed (according to the standard SYNC0/1 can be done after the slaves are in SAFEOP), but just to be safe and mirror the TwinCAT behavior. 
5) Wait at least 7 seconds before bailing out. This is how long it took in the TwinCAT trace from SAFEOP request to SAFEOP confirmed.

2b-t, 2020-10-01:
Hey @ArthurKetels 
your remarks are as always really helpful. _Thanks a lot_, I will look into it!

2b-t, 2020-10-02:
Hey @ArthurKetels 
I have implemented everything you proposed:

1. I _ordered the SoE commands by examining the Wireshark trace_ closely, now they are sent in the same order as the TwinCAT Wireshark traces. It has a very interesting logic, it writes P-0010 (A), P-0304 (A), P-0010 (B), P-0304 (B), S-0015 (A), S-0016 (A), S-0015 (B), S-0016 (B), S-0024 (A), S-0024 (B), then comes a long list of commands for drive 0 which have again some internal order, then another for drive 1 (which has the same order as the ones for drive 0) and finally P-0556 (A), P-0556 (B), S-0104 (A) and S-0104 (B) are sent.
3. I _pulled the latest commit_ of SOEM.
4. I _set SYNC0/1_ and then in a loop I _start sending process data_ to the slaves in a loop, and request SafeOP.
5. I _wait for at least 10 seconds_ after requesting it before I finally give up. I had never tried that before. I had actually always paused before requesting it in my previous attempts.

My code now works as follows:

```
if (ec_init(ifname)) {
  /// Read start-up list from .csv/.xml and sort according to TwinCAT logic
  /// ...

  if (ec_config_init(FALSE) > 0) {
    ec_statecheck(SLAVE_ID, EC_STATE_PRE_OP, EC_TIMEOUTSTATE);

    if (ec_slave[SLAVE_ID].state == EC_STATE_PRE_OP) {
      /// Send start-up list
      /// ...

      uint32 cycleTime0 {  250000 };
      uint32 cycleTime1 { 2000000 };
      int32  cycleShift {       0 };
      ec_configdc();
      ec_dcsync01(SLAVE_ID, TRUE, cycleTime0, cycleTime1, cycleShift);
      ec_config_map(&IOmap);                    

      auto const endTime = std::chrono::system_clock::now() + std::chrono::seconds(10);

      while ((std::chrono::system_clock::now() < endTime) && 
             (ec_slave[SLAVE_ID].state != EC_STATE_SAFE_OP)) {
        ec_send_processdata();
        ec_receive_processdata(EC_TIMEOUTRET);
        ec_slave[SLAVE_ID].state = EC_STATE_SAFE_OP;
        ec_writestate(SLAVE_ID);
        ec_statecheck(SLAVE_ID, EC_STATE_SAFE_OP, EC_TIMEOUTSTATE);
      }

      if (ec_slave[SLAVE_ID].state == EC_STATE_SAFE_OP) {
        /// The drive fails to get here...
        /// ...

```

Nonetheless I am _not able to bring the drives to SafeOP_: Some seconds after requesting SafeOP the drive gives me error code _F412_ - invalid cycle time.

Here again a Wireshark trace of the process: 
[SOEM_Update1.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5319191/SOEM_Update1.zip)

I was not able to isolate the messages in the trace that set the sync times to check if I am doing everything correctly but I have taken the times from the distributed clock settings for the drive in TwinCAT3 (see screenshot below), furthermore SYNC1 is a multiple of SYNC0 and equal to S-0001 and S-0002 as required.

[image]

_Thanks a lot!_

ArthurKetels, 2020-10-02:
Well, actually you achieved quite a bit. The IDN mapping seems to work OK now.

That the slave bails out with a PDO timing issue makes total sense. In your code you enable SYNC0/1 with the correct timing but fail to do the same with your PDO cycle. Have a look at the sample code in red_test.c. It shows how to set-up a real-time task that outputs a synchronous PDO cycle and locks to the reference slave clock with a PLL. You start the PDO cycle right after:
`ec_config_map(&IOmap);`

This PDO cycle needs to run for a few seconds to get exactly in the middle of the sync0/1 window. For SoE servo drives it is extremely important to have very tight PDO cycle timing tolerances. Your sync unit cycle is 2000us or 2ms. You have to send a PDO cycle exactly every 2ms. You also need an offset from the sync1 firing, a good starter is halfway (1ms). Look again at red_test.c how to do this.

In TwinCAT you see a burst of slave clock syncs in the beginning of the trace (ARMW at 0x0910). This is not needed for a single slave. The reason is to sync all slave clocks among each other.  When using a large number of slaves it could otherwise take quite some time to get all the clocks aligned. Remember this is only about the slave clocks. It is your responsibility to do the same for the master clock.

2b-t, 2020-10-02:
_Thanks a lot_ @ArthurKetels 
you are indeed a _walking encyclopedia of Ethercat technology_.
I will have a look at it!
_Have a nice weekend!_

nick42h, 2020-10-09:
Hello @2b-t, 

thank you for the work, i am really happy to have found this topic. Could you please update this topic if you managed it to bring your AX5206 to OP with SOEM? 
I've just started working with an AX5112 and facing similar problems you discribed. For a beginning I will follow the steps mentioned here (https://github.com/OpenEtherCATsociety/SOEM/issues/447#issuecomment-702831871)

2b-t, 2020-10-09:
_Hey @ArthurKetels_ 
with _your help I was able to get the drive to OP_ now.

By looking at the two Wireshark traces I realised that actually **Sync1** in the TwinCAT was actually set to **1750000** while I was setting it to 2000000. Already by fixing that I was able with the code above to send the drive to SafeOP.
For going to OP I turned to the `red_test.c` example and started sending PDOs in the correct timing. First I implemented my own routine with C++ syntax (`std::thread` with `std::this_thread::sleep_for`). Even though I had isolated the thread on a proprietary CPU-core of the real-time patched operating system with maximum priority, the **jitter** would be almost the magnitude of the sync time resulting in an _F415 error_ (Sync lost). Already switching to a
````
auto startPoint = std::chrono::steady_clock::now();
while (std::chrono::steady_clock::now() - startPoint < waitTime) continue;
```` 
was an order of magnitude more precise. _Copy and pasting the relevant part of `red_test.c` keeping a **Posix-thread** would have a jitter two magnitudes lower than the initial C++ syntax and would finally **bring the drive to OP**._
Thanks a lot again!

2b-t, 2020-10-09:
_Hey @nick42h_
Funny enough that you write precisely at the same time I was composing my final comment.
In case my final comment does answer your question here a couple of hints from my side for working with these drives:
**Use Wireshark** to get a trace of the **start-up sequence** in TwinCAT and compare your SOEM output incrementally to it (Use the corresponding Wireshark filters to filter for IDNs and the sync times). Check the display of your drives to double check the current state and **read out errors**.

The **start-up list has to be sent in PreOP** using the ec_SoEwrite() function, potentially maintaining the same weird order of the parameters as TwinCAT (I still have to check if it works without as well). Pay attention to the **endian** and the **sync times**. Then ask for the state change to SafeOP **giving the slave 10 seconds or so to respond** similar to pseudo-code given in [my comment](https://github.com/OpenEtherCATsociety/SOEM/issues/447#issuecomment-702831871).
If this succeeds start a real-time thread that similar to `red_test.c` sends PDOs to the slave in the correct timing and gradually tries to sync the reference clock to the master clock. I did this for a couple of seconds before finally requesting OP. Pay attention to the **jitter**, the AX5206 seems to be really sensible to it.

Good luck! :+1:

closed 2020-10-09 by 2b-t

nick42h, 2020-10-15:
I got the AX5112 to Operational state too. Thanks again for the detailed description @ArthurKetels and @2b-t . 

The F415 "jitter" problem was also the last thing I had to struggle. To deal with the jitter the following helped (it was new to me, maybe it helps other too):  Execute your script not only with ./simple_test ifname, but with sudo chrt -f 80 ./simple_test ifname, to set the real time rights to a high position (99 is highest).

ArthurKetels, 2020-10-15:
Watch out with elevating real-time priorities above 49. This is the kernel priority for interrupt and driver handling. Your application might see an improvement but if you then block on a kernel function it is no use.

The main reason is that simple_test is not designed for lowest jitter. Please take a look at red_test, there the correct real-time priorities are used. Fine-tuning your system is no easy task and needs very good understanding of your hardware.

2b-t, 2020-10-15:
Hey @nick42h 
really glad to hear that and _good job_ for being that fast. It took me several dozen hours to reach the same result. I am setting the thread priority in a similar way in order to decrease the jitter but on an isolate CPU core.

nick42h, 2020-10-19:
Hello @2b-t, 
have you manged to turn the drive unit? 

I send the following data to theese IDNs, but the drive is not following my commands (IDN-List for the AX5000 found [here](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&ved=2ahUKEwi-66Oy_r_sAhWIy6QKHVHpB5cQFjAAegQIBxAC&url=http%3A%2F%2Fftp.beckhoff.com%2Fdownload%2Fdocument%2Fmotion%2Fax5000-idn-descriptions_en.pdf&usg=AOvVaw35pRSbV-HB6ab3NzUjTAcK)):

- S-0-0032 Primary operation mode: 0x11 --> "pos ctrl feedback 1 lag less" (during PreOperational-State with the rest of the start up list)
- S-0-0134 Master control word: 16bit binary and I send zeros for the first 12 bits and ones for the 13th, 14th and 15th bit (Restart drive, enable drive and drive on) when the AX5000 reached OP 
- S-0-0051 Position feedback value 1: reading this value and safe it as the _startposition_ in the first cycle
- S-0-0047 Position command value: commanding new position and add the _startposition_ 

But the drive is not moving. When I read "S-0-0135 Drive status word" the last to bits correspond with a 2, meaning that the drive is ready, but not ready to operate. Also the 3rd bit "Status command value processing"  is zero, meaning that the drive ignores set values. The beckhoff support mentioned, that when no safety or other hardware components are included the last two bits of the Drive status word (S-0-0134), should correspond with a 3 when, as soon when the master control word (S-0-0134) says enable drive and drive on. 

Do you send anything else during your cyclic loop when the AX5000 reached OP-state to move the drive? I can not test how TwinCat is controlling the drive, because my laptop does not satisfy the real-time-requirements, which is immediately leading to an F415 error before I can test anything.

2b-t, 2020-10-21:
Hey @nick42h 
sorry for the late reply. I sadly didn't have the time to work on it ever since bringing it to OP mode and therefore _actually never verified if it is moving correctly_. I should have some time in the following days though.
The first thing I can do is _make a Wireshark trace_ of [putting the drive into manual mode](https://infosys.beckhoff.com/english.php?content=../content/1033/ax5000_system_doku_hw2/18014400028207883.html&id=) and [moving it around](https://infosys.beckhoff.com/english.php?content=../content/1033/ax5000_system_doku_hw2/18014400028207883.html&id=) and share it with you (if that helps). I will also check the "Drive status word" S-0-0135 and let you know what mine looks like.

nick42h, 2020-10-22:
Thank you, but I could borrow a real time computer and do some tests myself. 

I also found out why my attempt [here](https://github.com/OpenEtherCATsociety/SOEM/issues/447#issuecomment-711641308)  did not work:
Actually it is not possible to control the drive by using the ec_SoEwrite function and wirite a value to "S-0-0047 Position command value", because this value is not changeable in any ethercat state. This is also written in the IDN-List for the AX5000 found [here](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&ved=2ahUKEwi-66Oy_r_sAhWIy6QKHVHpB5cQFjAAegQIBxAC&url=http%3A%2F%2Fftp.beckhoff.com%2Fdownload%2Fdocument%2Fmotion%2Fax5000-idn-descriptions_en.pdf&usg=AOvVaw35pRSbV-HB6ab3NzUjTAcK). But the value can be transferred cyclically via the MDT (Master Data Telegram). The same applies for "S-0-0134 Master control word". It can not be changed via ec_SoEwrite, but can be transferred cyclically via MDT. 

The values that are cyclically transferred are set during sending the start up list, by the IDNs "S-0-0016 Configuration list of AT" and "S-0-0024 Configuration list of MDT". 
The default start up list for the AX5112 looks like this [AX5112_start_up_list.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5422621/AX5112_start_up_list.zip).
As you can see in the csv-file to S-0-0016 "04 00 04 00 33 00 BD 00" is send. When you read the bits from left to right you get 0x00BD = 189, than 0x0033 = 51 in decimal. Theese values correspond with IDN "S-0-0189 Following distance" and "S-0-0051 Position feedback value 1 (motor feedback)". 
For "S-0-0024 Configuration list of MDT" the following bit pattern is send: 02 00 02 00 2F 00. 0x002F = 42 --> "S-0-0047 Position command value". 
I think this is why the values can be seen in TwinCat in the highlightes window too. There seem to be some other values by default. I don't know what the bits 02 00 02 00 and 04 00 04 00 do in S-0-0024/16 do or why they are necessary there.
[image]

The MDT and AT data can be accessed via SOEM by the slaves inputs and outputs in your cyclic loop during OperationalState: 

```
                    ec_send_processdata();
//                    ec_send_overlap_processdata();

                    wkc = ec_receive_processdata(EC_TIMEOUTRET);

                    // read AT values received from slave
                    AX5112_PDO_ReceiveFromSlave* incoming_data = (AX5112_PDO_ReceiveFromSlave*) ec_slave[SLAVE_ID].inputs;
//                    print_AX5112_PDO_ReceiveFromSlave(incoming_data);

                    // fill the outputs with the MDT values
                    AX5112_PDO_SendToSlave *to_slave = (AX5112_PDO_SendToSlave*) ec_slave[SLAVE_ID].outputs;
                    to_slave->MasterControlWord = static_cast<uint16>(0xE000);
                    to_slave->PositionCommandValue = static_cast<int32>(commanded_position);
```

When I send the commanded_position like this, the drive is following the given position. 

(The structs are defined in the following way)
```
struct AX5112_PDO_ReceiveFromSlave{
    uint16              StatusWord;                         
    int32               PositionFeedback1;              
    int32               followingDistance;                 
    uint8               WcState:1;
    uint8               InputToggle:1;
    uint16              State;
    uint64              AdsAddr;
    uint8               Chn0;
    int32               DcOutputShift;
    int32               DcInputShift;
}__attribute__((packed));

struct AX5112_PDO_SendToSlave{
    uint16              MasterControlWord;                       
    int32               PositionCommandValue;                   
}__attribute__((packed));
```

Probably not everything I wrote here is correct, but at the moment it works. 

Next I will try to modify S-0-0024 and S-0-0016 to be able to command "S-0-0036 Velocity command value" or "S-0-0080 Torque command value" for velocity and torque control or read  "S-0-0040 Velocity feedback value 1" or  "S-0-0084 Torque feedback value" for velocity and torque control.

2b-t, 2020-10-22:
Hey @nick42h 
I just had noted down the following paragraph to share it with you together with the Wireshark trace but again you were faster.

> From what I understand (tab "Process data" in the TwinCAT drive menu) each axis expects cyclically S-0-0134 (Master control word) and S-0-0047 (Position command value) as MTD while it should return S-0-0135 (Drive status word), S-0-0051 (Position feedback 1 value), S-0-0189 (Following distance) and S-0-0084 (Torque/force feedback value) as AT when the primary operation mode (S-0-0032) is set to 0x000B (11). [AT and MTD should actually be sent cyclically as a single datagram](https://www.sciencedirect.com/science/article/pii/S1877705811055822) while the same EtherCAT frame might contain additional acyclic data. The corresponding IDNs themselves are marked in their documentation as write only and are only supposed to be accessed with AT/MTD.
> I think what we have to do is to first check if `ec_readIDNmap()` is actually the correct size of the AT/MTD and then simply map the information (length, data type, offset) from the "Process data" tab with the corresponding offsets and data sizes to the PDOs like for any non-Sercos drive.

It seems like this is precisely what you have done. Thanks for sharing your approach in such detail! 👍

2b-t, 2020-10-22:
_From what I understand the first digits of S-0-0016/S-0-0024 are just used for passing the number of corresponding IDNs used in the AT and MTD telegrams_ (actual and max length, therefore it is twice the same value):  Four (04 00) for AT (see my comment above: S-0-0135, S-0-0051, S-0-0189, S-0-0084) and two (02 00) for MTD (S-0-0134, S-0-0047).

2b-t, 2020-10-22:
For the sake of documentation to future readers:
The PDO mapping can be found in the TwinCAT settings and is similar to other non-Sercos drives. There you will find an entry for S-0-0016 and S-0-0024 for each individual axis: If you highlight them they will show you which IDNs will be transferred in the AT/MTD telegrams.
For me they are:

> S-0-0016 (AT):
> Index       | Size | Offset | Type
> S-0-0135 |   2   |     0     | UINT
> S-0-0051 |   4   |     2     | DINT
> S-0-0189 |   4   |     6     | DINT
> S-0-0084 |   2   |    10    |   INT
> 
> S-0-0024 (MTD):
> Index       | Size | Offset | Type
> S-0-0134 |   2   |     0     | UINT
> S-0-0047 |   4   |     2     | DINT

The field "Type" shows how the underlying type should be interpreted (see [Beckhoff Standard Data Type documentation](https://infosys.beckhoff.com/english.php?content=../content/1033/tcplccontrol/html/tcplcctrl_plc_data_types_overview.htm&id)), the field "Size" its length in multiples of 1 Byte (8 bit, which clearly has to match the aforementioned data type) and the field "Offset" holds information about additional padding that might be used (in this case none as it only corresponds to the sum of the sizes of the previous entries). This way you can construct the structs given @nick42h above for [different operation modes (S-0-0032) other than mode 11 (pos ctrl feedback 1 lag less)](http://ftp.beckhoff.com/download/document/motion/ax5000-idn-descriptions_en.pdf#S-0-0032) if you already have a working TwinCAT configuration. If you do not have such a configuration you can reconstruct the transferred IDNs from the data in S-0-0016 directly as pointed out by @nick42h in [this comment](https://github.com/OpenEtherCATsociety/SOEM/issues/447#issuecomment-714472754) and determine the length of each individual IDN by looking at the corresponding IDN documentation (datatype, data length in bit, min and max value).

nick42h, 2020-10-23:
Thank you for the correction of the AT-Telegram.

pkmcnc, 2021-12-23:
> Further, why do people choose AX5 series drives? We all know how difficult they are to use. There are so many drives that give a much better user experience.

Hello Arthur.
Can you point me to another Ethercat drive series/manufacturer. Requirements are 380V, 12-15A nominal, Endat 2.1 support, better available from ebay.
I found Beckhoff AX5000 series so far, other manufacturers doesn't fit or are impossible to purchase or wildly expensive, tool long lead time etc.
Thank you!

Zahir-Khan, 2021-12-23:
Yaskawa Mitsubishi and Omron are pretty good. You can get affordable Maxine drives from vitalsystem.com

I've had good experience with Maxon drives as well. You will have to research requirements yourself.

pkmcnc, 2021-12-23:
Well I asked 6 Yaskawa representatives in my country about Ethercat drives 3KW with Endat encoder modules. No one even quoted so far, this is too unusual order. They rather sell complete servo sets with motors.
AMC Ethercat drive 30A peak costs more than $3000 with 6 month lead time.
Collmorgens are not cheaper.
Maxsine EP3E looks good but never heard of Endat protocol.
Maxon drives are too weak... any other suggestion?

ArthurKetels, 2021-12-23:
Some European manufacturers:
SEW-Eurodrive
Stoeber Antriebstechnik
KEB combivert
ESR Pollmeier
Moog