OpenEtherCATsociety/SOEM #448: Issue with output status persistence on Beckhoff E1859 EtherCAT Terminal

lucairsara, 2020-09-19:
Hello,
I m using SOEM on raspbian (master) to control a slave E1859 EtherCAT Terminal (8 digital in, 8 digital out), over an EK1100 bus coupler.
I am able to write to the outputs and read from the inputs of the E1859.

The code is something like:
```
...
for(i = 0; i < 8; i++){
    ...
    ec_send_processdata();
    wkc = ec_receive_processdata(EC_TIMEOUTRET);
    ec_slave[0].outputs[0] = pow(2, i); // turn on the digital outputs in sequence... 1, 2, 4, 8, ....
    usleep(10000000); // sleep for 10 seconds before sending the next datagram
    ...
}
...
```

**I expect that the digital outputs (in sequence) stay turned on for 10 seconds (until the next datagram is sent from the master to the slave), but instead they only turn on for few millisecs and then turn off again.**

Can anyone tell me how to keep/preserve the status of the digital outputs (on/off) when a datagram is sent from the master, until the next datagram is sent?

Thank you very much for your help!
Luca

nakarlsson, 2020-09-20:
You must send/recv processdata continuously while waiting for the 10s to elapse. Sending a single send/recv in 10s will cause the slave to enter SAFEOP + Error state due to the Watchdog getting triggered. Default you need to send datagrams at least < 100ms plus some margin for Jitter in you master.

lucairsara, 2020-09-23:
Thank you very much for the prompt answer!
It works perfectly!
Best regards,
Luca

closed 2020-09-23 by lucairsara