OpenEtherCATsociety/SOEM #449: ATI force gamma sensor,Slave 1 State=12 StatusCode=  1e : Invalid input configuration

Jacklove-G, 2020-09-28:
I'm a green hand in the field of ethercat. And now, I have a ATI force sensor, and I've tried to get the data with soem for several days, however, I still can't  succeed. So, I wonder to know what should I do on the simpletest.c.

Jacklove-G, 2020-09-28:
here is the slaveinfo imformation

Slaveinfo
Starting slaveinfo
ec_init on enx081f71202d91 succeeded.
1 slaves found and configured.
Calculated workcounter 3
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  1e : Invalid input configuration

Slave:1
 Name:ATI EtherCAT F/T Sensor
 Output size: 64bits
 Input size: 224bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000732 ID: 26483052 Rev: 00010111
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:   8 F:00010064 Type:3
 SM3 A:1c00 L:  28 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   8 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000008 Ll:  28 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 23 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program

Jacklove-G, 2020-09-28:
And here is the ESI file
[ATIEtherCATFT.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5289551/ATIEtherCATFT.zip)

Jacklove-G, 2020-09-28:
And here is the slaveinfo imformation

Slaveinfo
Starting slaveinfo
ec_init on enx081f71202d91 succeeded.
1 slaves found and configured.
Calculated workcounter 3
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  1e : Invalid input configuration

Slave:1
 Name:ATI EtherCAT F/T Sensor
 Output size: 64bits
 Input size: 224bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000732 ID: 26483052 Rev: 00010111
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:   8 F:00010064 Type:3
 SM3 A:1c00 L:  28 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   8 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000008 Ll:  28 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 23 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x7010:0x01 0x20 UNSIGNED32   Control 1
  [0x0004.0] 0x7010:0x02 0x20 UNSIGNED32   Control 2
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0008.0] 0x6000:0x01 0x20 INTEGER32    Fx/Gage0
  [0x000C.0] 0x6000:0x02 0x20 INTEGER32    Fy/Gage1
  [0x0010.0] 0x6000:0x03 0x20 INTEGER32    Fz/Gage2
  [0x0014.0] 0x6000:0x04 0x20 INTEGER32    Tx/Gage3
  [0x0018.0] 0x6000:0x05 0x20 INTEGER32    Ty/Gage3
  [0x001C.0] 0x6000:0x06 0x20 INTEGER32    Tz/Gage3
  [0x0020.0] 0x6010:0x00 0x20 UNSIGNED32   SubIndex 000
  [0x0024.0] 0x6020:0x00 0x20 UNSIGNED32   SubIndex 000
End slaveinfo, close socket
End program

nakarlsson, 2020-09-28:
Please provide a wireshark capturing the failing startup

Jacklove-G, 2020-09-28:
@nakarlsson Here is the data

[ATI_wireshark.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5290153/ATI_wireshark.zip)

Jacklove-G, 2020-09-28:
And here is my code 

#include <stdio.h>
#include <string.h>
#include <inttypes.h>

#include "ethercat.h"

#define EC_TIMEOUTMON 500

char IOmap[4096];
OSAL_THREAD_HANDLE thread1;
int expectedWKC;
boolean needlf;
volatile int wkc;
boolean inOP;
uint8 currentgroup = 0;

static int RS3_write8(uint16 slave, uint16 index, uint8 subindex, uint8 value)
{
int wkc=0;
wkc += ec_SDOwrite(slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
return wkc;
}

static int RS3_write16(uint16 slave, uint16 index, uint8 subindex, uint16 value)
{
int wkc=0;
wkc += ec_SDOwrite(slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
return wkc;
}

static int RS3_write32(uint16 slave, uint16 index, uint8 subindex, uint32 value)
{
int wkc=0;
wkc += ec_SDOwrite(slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
return wkc;
}

static int RS3_setup(uint16 slave)
{
int wkc=0;

printf("RS3 drive setup\n");
wkc += RS3_write8(slave, 0x1C12, 0, 0);
wkc += RS3_write8(slave, 0x1C13, 0 , 0);

wkc += RS3_write8(slave,  0x1601, 0 , 0);//output Rx
wkc += RS3_write32(slave, 0x1601, 1, 0x70100010);
wkc += RS3_write32(slave, 0x1601, 2, 0x70100020);
wkc += RS3_write8(slave,  0x1601, 0, 2);

wkc += RS3_write8(slave,  0x1A00, 0, 0);//input  Tx
wkc += RS3_write32(slave, 0x1A00, 1, 0x60000010); 
wkc += RS3_write32(slave, 0x1A00, 2, 0x60000020); 
wkc += RS3_write32(slave, 0x1A00, 3, 0x60000030); 
wkc += RS3_write32(slave, 0x1A00, 4, 0x60000040); 
wkc += RS3_write32(slave, 0x1A00, 5, 0x60000050); 
wkc += RS3_write32(slave, 0x1A00, 6, 0x60000060); 
wkc += RS3_write32(slave, 0x1A00, 7, 0x60100000); 
wkc += RS3_write32(slave, 0x1A00, 8, 0x60200000); 
wkc += RS3_write8(slave, 0x1A00, 0, 8);

wkc += RS3_write16(slave, 0x1C12, 1, 0x1601);
wkc += RS3_write16(slave, 0x1C12, 0, 1);

wkc += RS3_write16(slave, 0x1C13, 1, 0x1A00);
wkc += RS3_write16(slave, 0x1C13, 0, 1);

//Explicitly set flags that are (probably) invalid in EEPROM
[... 213 lines cut ...]
    }
    osal_usleep(10000);
}
}

int main(int argc, char *argv[])
{
printf("SOEM (Simple Open EtherCAT Master)\nSimple test\n");

if (argc > 1)
{
/* create thread to handle slave error handling in OP /
// pthread_create( &thread1, NULL, (void ) &ecatcheck, (void) &ctime);
osal_thread_create(&thread1, 128000, &ecatcheck, (void) &ctime);
/* start cyclic part */
simpletest(argv[1]);
}
else
{
printf("Usage: simple_test ifname1\nifname = eth0 for example\n");
}

printf("End program\n");
return (0);
}

Jacklove-G, 2020-09-28:
#include <stdio.h>
#include <string.h>
#include <inttypes.h>

#include "ethercat.h"

#define EC_TIMEOUTMON 500

char IOmap[4096];
OSAL_THREAD_HANDLE thread1;
int expectedWKC;
boolean needlf;
volatile int wkc;
boolean inOP;
uint8 currentgroup = 0;

static int RS3_write8(uint16 slave, uint16 index, uint8 subindex, uint8 value)
{
   int wkc=0;
   wkc += ec_SDOwrite(slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
   return wkc;
}

static int RS3_write16(uint16 slave, uint16 index, uint8 subindex, uint16 value)
{
   int wkc=0;
   wkc += ec_SDOwrite(slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
   return wkc;
}

static int RS3_write32(uint16 slave, uint16 index, uint8 subindex, uint32 value)
{
   int wkc=0;
   wkc += ec_SDOwrite(slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
   return wkc;
}

static int RS3_setup(uint16 slave)
{
   int wkc=0;

   printf("RS3 drive setup\n");
   wkc += RS3_write8(slave, 0x1C12, 0, 0);
   wkc += RS3_write8(slave, 0x1C13, 0 , 0);

   wkc += RS3_write8(slave,  0x1601, 0 , 0);//output Rx
   wkc += RS3_write32(slave, 0x1601, 1, 0x70100010);
   wkc += RS3_write32(slave, 0x1601, 2, 0x70100020);
   wkc += RS3_write8(slave,  0x1601, 0, 2);

   wkc += RS3_write8(slave,  0x1A00, 0, 0);//input  Tx
   wkc += RS3_write32(slave, 0x1A00, 1, 0x60000010); 
   wkc += RS3_write32(slave, 0x1A00, 2, 0x60000020); 
   wkc += RS3_write32(slave, 0x1A00, 3, 0x60000030); 
   wkc += RS3_write32(slave, 0x1A00, 4, 0x60000040); 
   wkc += RS3_write32(slave, 0x1A00, 5, 0x60000050); 
   wkc += RS3_write32(slave, 0x1A00, 6, 0x60000060); 
   wkc += RS3_write32(slave, 0x1A00, 7, 0x60100000); 
   wkc += RS3_write32(slave, 0x1A00, 8, 0x60200000); 
   wkc += RS3_write8(slave, 0x1A00, 0, 8);

   wkc += RS3_write16(slave, 0x1C12, 1, 0x1601);
   wkc += RS3_write16(slave, 0x1C12, 0, 1);

   wkc += RS3_write16(slave, 0x1C13, 1, 0x1A00);
   wkc += RS3_write16(slave, 0x1C13, 0, 1);

   //Explicitly set flags that are (probably) invalid in EEPROM
   ec_slave[slave].SM[2].SMflags = 0x10024;

[... 211 lines cut ...]
    }
    osal_usleep(10000);
   }
}

int main(int argc, char *argv[])
{
   printf("SOEM (Simple Open EtherCAT Master)\nSimple test\n");

   if (argc > 1)
   {
   /* create thread to handle slave error handling in OP /
   // pthread_create( &thread1, NULL, (void ) &ecatcheck, (void) &ctime);
   osal_thread_create(&thread1, 128000, &ecatcheck, (void) &ctime);
   /* start cyclic part */
   simpletest(argv[1]);
   }
   else
   {
   printf("Usage: simple_test ifname1\nifname = eth0 for example\n");
   }

   printf("End program\n");
   return (0);
}

Jacklove-G, 2020-09-28:
#include <stdio.h>
#include <string.h>
#include <inttypes.h>

#include "ethercat.h"

#define EC_TIMEOUTMON 500

char IOmap[4096];
OSAL_THREAD_HANDLE thread1;
int expectedWKC;
boolean needlf;
volatile int wkc;
boolean inOP;
uint8 currentgroup = 0;

static int RS3_write8(uint16 slave, uint16 index, uint8 subindex, uint8 value)
{
   int wkc=0;
   wkc += ec_SDOwrite(slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
   return wkc;
}

static int RS3_write16(uint16 slave, uint16 index, uint8 subindex, uint16 value)
{
   int wkc=0;
   wkc += ec_SDOwrite(slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
   return wkc;
}

static int RS3_write32(uint16 slave, uint16 index, uint8 subindex, uint32 value)
{
   int wkc=0;
   wkc += ec_SDOwrite(slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
   return wkc;
}

static int RS3_setup(uint16 slave)
{
   int wkc=0;

   printf("RS3 drive setup\n");
   wkc += RS3_write8(slave, 0x1C12, 0, 0);
   wkc += RS3_write8(slave, 0x1C13, 0 , 0);

   wkc += RS3_write8(slave,  0x1601, 0 , 0);//output Rx
   wkc += RS3_write32(slave, 0x1601, 1, 0x70100010);
   wkc += RS3_write32(slave, 0x1601, 2, 0x70100020);
   wkc += RS3_write8(slave,  0x1601, 0, 2);

   wkc += RS3_write8(slave,  0x1A00, 0, 0);//input  Tx
   wkc += RS3_write32(slave, 0x1A00, 1, 0x60000010); 
   wkc += RS3_write32(slave, 0x1A00, 2, 0x60000020); 
   wkc += RS3_write32(slave, 0x1A00, 3, 0x60000030); 
   wkc += RS3_write32(slave, 0x1A00, 4, 0x60000040); 
   wkc += RS3_write32(slave, 0x1A00, 5, 0x60000050); 
   wkc += RS3_write32(slave, 0x1A00, 6, 0x60000060); 
   wkc += RS3_write32(slave, 0x1A00, 7, 0x60100000); 
   wkc += RS3_write32(slave, 0x1A00, 8, 0x60200000); 
   wkc += RS3_write8(slave, 0x1A00, 0, 8);

   wkc += RS3_write16(slave, 0x1C12, 1, 0x1601);
   wkc += RS3_write16(slave, 0x1C12, 0, 1);

   wkc += RS3_write16(slave, 0x1C13, 1, 0x1A00);
   wkc += RS3_write16(slave, 0x1C13, 0, 1);

   //Explicitly set flags that are (probably) invalid in EEPROM
   ec_slave[slave].SM[2].SMflags = 0x10024;

   //Explicitly disable sync managers that are activated by EEPROM
   ec_slave[slave].SM[4].StartAddr = 0;
   ec_slave[slave].SM[5].StartAddr = 0;

   //set a slave name

   strncpy(ec_slave[slave].name, "ATI EtherCAT F/T Sensor", EC_MAXNAME);

   if(wkc != 20)
   {
    printf("RS3 setup failed\n %d\n",wkc);
    return -1;
   }
   return 0;
}

Jacklove-G, 2020-09-28:
Here is my code

#include <stdio.h>
#include <string.h>
#include <inttypes.h>

#include "ethercat.h"

#define EC_TIMEOUTMON 500

char IOmap[4096];
OSAL_THREAD_HANDLE thread1;
int expectedWKC;
boolean needlf;
volatile int wkc;
boolean inOP;
uint8 currentgroup = 0;

static int RS3_write8(uint16 slave, uint16 index, uint8 subindex, uint8 value)
{
   int wkc=0;
   wkc += ec_SDOwrite(slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
   return wkc;
}

static int RS3_write16(uint16 slave, uint16 index, uint8 subindex, uint16 value)
{
   int wkc=0;
   wkc += ec_SDOwrite(slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
   return wkc;
}

static int RS3_write32(uint16 slave, uint16 index, uint8 subindex, uint32 value)
{
   int wkc=0;
   wkc += ec_SDOwrite(slave, index, subindex, FALSE, sizeof(value), &value, EC_TIMEOUTRXM);
   return wkc;
}

static int RS3_setup(uint16 slave)
{
   int wkc=0;

   printf("RS3 drive setup\n");
   wkc += RS3_write8(slave, 0x1C12, 0, 0);
   wkc += RS3_write8(slave, 0x1C13, 0 , 0);

   wkc += RS3_write8(slave,  0x1601, 0 , 0);//output Rx
   wkc += RS3_write32(slave, 0x1601, 1, 0x70100010);
   wkc += RS3_write32(slave, 0x1601, 2, 0x70100020);
   wkc += RS3_write8(slave,  0x1601, 0, 2);

   wkc += RS3_write8(slave,  0x1A00, 0, 0);//input  Tx
   wkc += RS3_write32(slave, 0x1A00, 1, 0x60000010); 
   wkc += RS3_write32(slave, 0x1A00, 2, 0x60000020); 
   wkc += RS3_write32(slave, 0x1A00, 3, 0x60000030); 
   wkc += RS3_write32(slave, 0x1A00, 4, 0x60000040); 
   wkc += RS3_write32(slave, 0x1A00, 5, 0x60000050); 
   wkc += RS3_write32(slave, 0x1A00, 6, 0x60000060); 
   wkc += RS3_write32(slave, 0x1A00, 7, 0x60100000); 
   wkc += RS3_write32(slave, 0x1A00, 8, 0x60200000); 
   wkc += RS3_write8(slave, 0x1A00, 0, 8);

   wkc += RS3_write16(slave, 0x1C12, 1, 0x1601);
   wkc += RS3_write16(slave, 0x1C12, 0, 1);

   wkc += RS3_write16(slave, 0x1C13, 1, 0x1A00);
   wkc += RS3_write16(slave, 0x1C13, 0, 1);

   //Explicitly set flags that are (probably) invalid in EEPROM
[... 213 lines cut ...]
    }
    osal_usleep(10000);
   }
}

int main(int argc, char *argv[])
{
   printf("SOEM (Simple Open EtherCAT Master)\nSimple test\n");

   if (argc > 1)
   {
   /* create thread to handle slave error handling in OP /
   // pthread_create( &thread1, NULL, (void ) &ecatcheck, (void) &ctime);
   osal_thread_create(&thread1, 128000, &ecatcheck, (void) &ctime);
   /* start cyclic part */
   simpletest(argv[1]);
   }
   else
   {
   printf("Usage: simple_test ifname1\nifname = eth0 for example\n");
   }

   printf("End program\n");
   return (0);
}

nakarlsson, 2020-09-28:
Try skipping the explicit PDO assignment,, it should be active by default.
The PDO assignment objects seems to be read-only

Jacklove-G, 2020-09-28:
@nakarlsson Is just commentting out the sentence "slave->PO2SOconfig = RS3_setup;"?    And it doesn't work.

Jacklove-G, 2020-09-28:
Fome the User manual，it seems that the senor would be in state operational automatically, so I am not sure if it is necessary to let the device enter the SAFE_OP state first.

nakarlsson, 2020-09-28:
post a wireshark of that startup

Jacklove-G, 2020-09-28:
@nakarlsson 
Here is the data after commentting out the sentence "slave->PO2SOconfig = RS3_setup;". And it contains data from powering up the device to running the   @@simpletest.c.
[ATI2.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5290700/ATI2.zip)

Jacklove-G, 2020-09-28:
@nakarlsson 
Here is the data after commentting out the sentence "slave->PO2SOconfig = RS3_setup;". And it contains data from powering up the device to running the simpletest .c.
[ATI2.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5290723/ATI2.zip)

Jacklove-G, 2020-09-28:
[image]
And from the ESI file, the default sizes of SM2 and SM3 should be 2 and 10, however, in the result of slaveinfo, the lengths of SM2 and SM3 are 8 and 28. 
[image]

nakarlsson, 2020-09-28:
yes, best guess is that the slave expect size 32 for SM3/FMMU1. Try that using the "configlist" used by etehrcatconfig -> ecx_config_from_table.

Jacklove-G, 2020-09-28:
[image]
[image]
Is it right?  I don't know what dtype, SM2f, SM3f, FM0ac and FM1ac should be. And Can I find them in ESI?
 And  in simpletest.c, I 've changed "ec_config_init(FALSE)" to "ec_config_init(TRUE)".

Jacklove-G, 2020-09-28:
[image]
[image]
Is it right? And I don't know what dtype, SM2f, SM3f, FM0ac and FM1ac should be. Can I find them in ESI file?
And in simpletest.c, I've change "ec_config_init(FALSE)" to "ec_config_init(TRUE)".

Jacklove-G, 2020-09-28:
Is there a tool which can change ESI file to eeporm data?

Jacklove-G, 2020-09-28:
Is there a tool which can change ESI file to eeporm data? Then I could use eepormtool.c, which may work.

Jacklove-G, 2020-09-28:
[image]
After the change, the PDO mapping seems to be wrong.

Jacklove-G, 2020-09-28:
And after the change, the PDO mapping seems to be wrong.

SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enx081f71202d91 succeeded.
1 slaves found and configured.
Calculated workcounter 3
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  1e : Invalid input configuration

Slave:1
 Name:ATI EtherCAT F/T Sensor
 Output size: 16bits
 Input size: 80bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000732 ID: 26483052 Rev: 00010111
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:   2 F:00010064 Type:3
 SM3 A:1c00 L:  10 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   2 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000002 Ll:  10 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 23 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x7010:0x01 0x20 UNSIGNED32   Control 1
  [0x0004.0] 0x7010:0x02 0x20 UNSIGNED32   Control 2
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0002.0] 0x6000:0x01 0x20 INTEGER32    Fx/Gage0
  [0x0006.0] 0x6000:0x02 0x20 INTEGER32    Fy/Gage1
  [0x000A.0] 0x6000:0x03 0x20 INTEGER32    Fz/Gage2
  [0x000E.0] 0x6000:0x04 0x20 INTEGER32    Tx/Gage3
  [0x0012.0] 0x6000:0x05 0x20 INTEGER32    Ty/Gage3
  [0x0016.0] 0x6000:0x06 0x20 INTEGER32    Tz/Gage3
  [0x001A.0] 0x6010:0x00 0x20 UNSIGNED32   SubIndex 000
  [0x001E.0] 0x6020:0x00 0x20 UNSIGNED32   SubIndex 000
End slaveinfo, close socket
End program

Jacklove-G, 2020-09-28:
Set Ibits to 256, and it works

nakarlsson, 2020-09-28:
yes, that is as expected, 8x32. I see now the last entry 0x6020:0x00 is missing when reading 0x1a00.
try running without the configlist and block complete access.

uset the following in the  RS3_setup , eg  CoEdetails &=~ECT_COEDET_SDOCA
`context->slavelist[slave].CoEdetails & ECT_COEDET_SDOCA) /* has Complete Access */`

Jacklove-G, 2020-09-28:
I'm sorry, and I can not understand why you say 0x6020:0x00 is missing, could you explain a little?

Jacklove-G, 2020-09-28:
@nakarlsson 
I'm sorry, and I can not understand why you say 0x6020:0x00 is missing, could you explain a little?

Jacklove-G, 2020-09-28:
@nakarlsson 
I'm sorry, and I can not understand why you say 0x6020:0x00 is missing, could you explain a little? thank you.

nakarlsson, 2020-09-28:
**In the capture, ATI2.pcapng frame No. 289, the slave response to the read of object 0x1a00 doesn't include 0x6020:0x00 and that would explain why SOEM is missing 4Bytes. 
This frame is a complete access response, I suggest you'd try to read it without complete access and see if SOEM can retrieve the proper input length 8Bytes.**

nakarlsson, 2020-12-22:
@Jacklove-G , Can this be closed?

closed 2021-02-09 by nakarlsson