OpenEtherCATsociety/SOEM #450: The operational

ziafon, 2020-09-28:
Hello, I'm new to SOEM. I‘m trying to control a motor with the nidec serve driver, and i use  jetson tx2 as master. My issue is that the tx2 can recieve data from the driver but could not enable it by sending data. Is there something wrong with my code?

	    if (ec_slavecount > 0)
	    {
            motor1=1;
            motor2=2;
            motor_setup(motor1);
            printf("Found %s at position %d\n", ec_slave[motor1].name, motor1);
            motor_setup(motor2);
            printf("Found %s at position %d\n", ec_slave[motor2].name, motor2);
	    }

	    /* If CA disabled => auto-mapping work */
            ec_config_map(&IOmap);

	    /* Let DC off for the time being */
//            ec_configdc();

            printf("Slaves mapped, state to SAFE_OP.\n");
            /* wait for all slaves to reach SAFE_OP state */
            ec_statecheck(0, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE * 4);

            oloop = ec_slave[0].Obytes;
            if ((oloop == 0) && (ec_slave[0].Obits > 0)) oloop = 1;
            if (oloop > 8) oloop = 8;
            iloop = ec_slave[0].Ibytes;
            if ((iloop == 0) && (ec_slave[0].Ibits > 0)) iloop = 1;
            if (iloop > 8) iloop = 8;

            printf("segments : %d : %d %d %d %d\n",ec_group[0].nsegments ,ec_group[0].IOsegment[0],ec_group[0].IOsegment[1],ec_group[0].IOsegment[2],ec_group[0].IOsegment[3]);

            printf("Request operational state for all slaves\n");
            expectedWKC = (ec_group[0].outputsWKC * 2) + ec_group[0].inputsWKC;
            printf("Calculated workcounter %d\n", expectedWKC);

	    /* Going operational */
            ec_slave[0].state = EC_STATE_OPERATIONAL;
            /* send one valid process data to make outputs in slaves happy*/
            ec_send_processdata();
            ec_receive_processdata(EC_TIMEOUTRET);

            /* request OP state for all slaves */
            ec_writestate(0);
            chk = 40;
            /* wait for all slaves to reach OP state */
            do
            {
                ec_send_processdata();
            	ec_receive_processdata(EC_TIMEOUTRET);
            	ec_statecheck(0, EC_STATE_OPERATIONAL, 50000);
            }
            while (chk-- && (ec_slave[0].state != EC_STATE_OPERATIONAL));

        if (ec_slave[0].state == EC_STATE_OPERATIONAL )
            {
            printf("Operational state reached for all slaves.\n");
            inOP = TRUE;

		/* Output torque and velocity values to a file */
/*
		FILE *fp = NULL;
		char *filename = "/home/protectli/velocity.dat";
		remove(filename);
		fp = fopen(filename, "w");
*/
	  	/**
           * 
	  	* Initialize static parameters by setting SDO
	 	*/
	    motor_init(motor1);
    	    motor_init(motor2);

        WRITE_SDO(1, 0x6040, 0, BUF16, 15, "*Control word: motor1*");
        WRITE_SDO(2, 0x6040, 0, BUF16, 15, "*Control word: motor2*");
        
		out_motor_t *out_motor1 = (out_motor_t *)ec_slave[motor1].outputs;
		in_motor_t *in_motor1 = (in_motor_t *)ec_slave[motor1].inputs;

                out_motor_t *out_motor2 = (out_motor_t *)ec_slave[motor2].outputs;
		in_motor_t *in_motor2 = (in_motor_t *)ec_slave[motor2].inputs;

		/* Initialize origin */
		ec_send_processdata();
               	wkc = ec_receive_processdata(EC_TIMEOUTRET);

                osal_usleep(500000);
                inOP = FALSE;
//		fclose(fp);
            }

nakarlsson, 2020-09-29:
The init part writing SDOs should be done before entering OP and the process data loop. In general, don't write SDOs in the process data loop, that traffic is asynchronous and indeterministic since it blocks while waiting for the slave to respond.
The control word 0x6040 should be mapped as part of the process data, use the process data control word to control the drives.

```
motor_init(motor1);
motor_init(motor2);

/* enter loop*/

out_motor1->control_word = 0xabcd /* 0x6040 mapped as process data */
send
/rcv
```

nakarlsson, 2021-02-09:
Seems inactive

closed 2021-02-09 by nakarlsson