OpenEtherCATsociety/SOEM #453: ec_DCtime not as expected

crispyDyne, 2020-10-08:
Hello,

When running the SOEM "simple_test", I expect the time output to be the number of nano seconds since Jan 1, 2000, or Unix time minus 946684800s. 

If my math is correct, the value I'm getting corresponds to the year 2106!
3348366266032364800 (19 digits)

Today is October 8, 2020, so I'm expecting something like this:
~655422############ (18 digits)

Why is the ethercat clock offset?

I'm not very experienced with C, so my efforts to track down the issue have not been fruitful. Any help would be appreciated! 

Cheers, 
Chris

### Example output from simple_test
> SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on eth0 succeeded.
2 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 18 0 0 0
Request operational state for all slaves
Calculated workcounter 2
Operational state reached for all slaves.
Processdata cycle 10000, WKC 2 , O: I: 01 00 00 80 ba 00 00 80 T:3348366266032364800
Request init state for all slaves
End simple test, close socket
End program

### Theories

- I am misunderstanding the time value
- The time is incorrectly set, and there is a way to set it that I am not aware of. 
- Integer casting error at the printf statement
- There is a different clock I should be looking at
- ?

### Info

- I'm running SOEM on a Raspberry Pi
- EK1101 with an EL3154 analog input 
- All other communication is as expected
- I am able to record data from the EL3154 
- System clock on the Pi reads correctly (Thu 08 Oct 2020 07:40:12 PM BST, 1602182421).

nakarlsson, 2020-10-09:
Are you running Linux?

crispyDyne, 2020-10-09:
Thank you very much. It all makes sense now.  

Yes, I am on Linux, and when I switch to CLOCK_REALTIME I get the expected time.  

I'll use CLOCK_MONOTONIC going forward and just note the initial system time as a display offset. 

Cheers

closed 2020-10-09 by crispyDyne

huyu168, 2021-05-31:
hello,i have the same problem with you,and finally i find that a variable  is overflowed