OpenEtherCATsociety/SOEM #455: How to read or write on DI/DO (Wago 750-354)?

Valneltin, 2020-10-19:
Hello !

I have a Wago 750-354 coupler with one module DO Wago 750-1504 and one module DI 750-1405.
For now, I'm using only the Wago DO module 750-1504.

Here my configuration :

- Ubuntu 18.04
- Wago 750-354 Coupler
- Wago 750-1504 DO module (16 DO)
- Wago 750-1405 DI module  (16 DI)
- Wago 750-600 End of line

I have done the slave_info and a simple_test and everything seems to work well.
Now I just want to control a DO. So I've seen two ways to write but I'm not sure to understand :

int ec_SDOwrite(uint16 Slave, uint16 Index, uint8 SubIndex, boolean CA, int psize, void *p, int Timeout)

OR

void set_output_int16 (uint16 slave_no, uint8 module_index, int16 value)
{
   uint8 *data_ptr;
   /* Get the IO map pointer from the ec_slave struct */
   data_ptr = ec_slave[slave_no].outputs;
   /* Move pointer to correct module index */
   data_ptr += module_index * 2;
   /* Read value byte by byte since all targets can handle misaligned
    * addresses
    */
   *data_ptr++ = (value >> 0) & 0xFF;
   *data_ptr++ = (value >> 8) & 0xFF;
}

So I put it in the simple_test script, after the module are in Operational state, in the loop.
But with the the ec_SDOwrite I have a segmentation fault. With the set_output_init16, nothing seems to append.

Someone have an idea ?
Thank you.

- **slave_info**

sudo ./test/linux/slaveinfo/slaveinfo eno1 -map
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eno1 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:WAGO Series 750 Bus Coupler
 Output size: 48bits
 Input size: 32bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000021 ID: 07500354 Rev: 00000002
 SM0 A:1000 L: 512 F:00010026 Type:1
 SM1 A:1400 L: 512 F:00010022 Type:2
 SM2 A:1800 L:   6 F:00010064 Type:3
 SM3 A:2400 L:   4 F:00010000 Type:4
 FMMU0 Ls:00000000 Ll:   6 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000006 Ll:   4 Lsb:0 Leb:7 Ps:2400 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 512 rd: 512 MBX protocols : 04
 CoE details: 33 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0xF200:0x01 0x01 BOOLEAN      K-Bus Cycle Overrun Flag Disable
  [0x0000.1] 0xF200:0x02 0x01 BOOLEAN      Input Process Data Hold Request
  [0x0000.2] 0xF200:0x03 0x01 BOOLEAN      Output Process Data Hold Request
  [0x0000.3] 0xF200:0x04 0x01 BOOLEAN      Output Process Data Clear Request
  [0x0000.4] 0x0000:0x00 0x0C
  [0x0002.0] 0xF200:0x05 0x10 UNSIGNED16   Diagnostics Control Word
  [0x0004.0] 0x7000:0x01 0x01 BOOLEAN      Channel 1 Data
  [0x0004.1] 0x7000:0x02 0x01 BOOLEAN      Channel 2 Data
  [0x0004.2] 0x7000:0x03 0x01 BOOLEAN      Channel 3 Data
  [0x0004.3] 0x7000:0x04 0x01 BOOLEAN      Channel 4 Data
  [0x0004.4] 0x7000:0x05 0x01 BOOLEAN      Channel 5 Data
  [0x0004.5] 0x7000:0x06 0x01 BOOLEAN      Channel 6 Data
  [0x0004.6] 0x7000:0x07 0x01 BOOLEAN      Channel 7 Data
  [0x0004.7] 0x7000:0x08 0x01 BOOLEAN      Channel 8 Data
  [0x0005.0] 0x7000:0x09 0x01 BOOLEAN      Channel 9 Data
  [0x0005.1] 0x7000:0x0A 0x01 BOOLEAN      Channel 10 Data
  [0x0005.2] 0x7000:0x0B 0x01 BOOLEAN      Channel 11 Data
  [0x0005.3] 0x7000:0x0C 0x01 BOOLEAN      Channel 12 Data
  [0x0005.4] 0x7000:0x0D 0x01 BOOLEAN      Channel 13 Data
  [0x0005.5] 0x7000:0x0E 0x01 BOOLEAN      Channel 14 Data
  [0x0005.6] 0x7000:0x0F 0x01 BOOLEAN      Channel 15 Data
  [0x0005.7] 0x7000:0x10 0x01 BOOLEAN      Channel 16 Data
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0006.0] 0xF100:0x01 0x01 BOOLEAN      K-Bus Cycle Overrun Flag
  [0x0006.1] 0xF100:0x02 0x01 BOOLEAN      Input Process Data Hold Ack.
  [0x0006.2] 0xF100:0x03 0x01 BOOLEAN      Output Process Data Hold Ack.
  [0x0006.3] 0xF100:0x04 0x01 BOOLEAN      Output Process Data Clear Ack.
  [0x0006.4] 0x0000:0x00 0x0B
  [0x0007.7] 0x10F3:0x04 0x01 BOOLEAN      New Message Available
  [0x0008.0] 0xF100:0x05 0x10 UNSIGNED16   Diagnostics Status Word
End slaveinfo, close socket
End program

- **simple_test**

sudo ./test/linux/simple_test/simple_test eno1
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on eno1 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 10 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Operational state reached for all slaves.
Processdata cycle 10000, WKC 3 , O: 00 00 00 00 00 00 I: 00 00 00 00 T:3348359585107514440
Request init state for all slaves
End simple test, close socket
End program

Valneltin, 2020-10-20:
Update !

I saw some values changed when I use a set_output_int16(n°salve, n°module, value) but my module didn't blink the LED of digital outputs ...

sudo ./test/linux/simple_test/simple_test eno1

SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on eno1 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 10 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Operational state reached for all slaves.
Processdata cycle 3000, WKC 3 , O: 00 00 00 00 10 10 I: 00 00 00 00 T:3348372107334227000
Request init state for all slaves
End simple test, close socket
End program

void simpletest(char *ifname)
{
    int i, j, oloop, iloop, chk;
    needlf = FALSE;
    inOP = FALSE;

   printf("Starting simple test\n");

   /* initialise SOEM, bind socket to ifname */
   if (ec_init(ifname))
   {
      printf("ec_init on %s succeeded.\n",ifname);
      /* find and auto-config slaves */

       if ( ec_config_init(FALSE) > 0 )
      {
         printf("%d slaves found and configured.\n",ec_slavecount);

         ec_config_map(&IOmap);

         ec_configdc();

         printf("Slaves mapped, state to SAFE_OP.\n");
         /* wait for all slaves to reach SAFE_OP state */
         ec_statecheck(0, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE * 4);

         oloop = ec_slave[0].Obytes;
         if ((oloop == 0) && (ec_slave[0].Obits > 0)) oloop = 1;
         if (oloop > 8) oloop = 8;
         iloop = ec_slave[0].Ibytes;
         if ((iloop == 0) && (ec_slave[0].Ibits > 0)) iloop = 1;
         if (iloop > 8) iloop = 8;

         printf("segments : %d : %d %d %d %d\n",ec_group[0].nsegments ,ec_group[0].IOsegment[0],ec_group[0].IOsegment[1],ec_group[0].IOsegment[2],ec_group[0].IOsegment[3]);

         printf("Request operational state for all slaves\n");
         expectedWKC = (ec_group[0].outputsWKC * 2) + ec_group[0].inputsWKC;
         printf("Calculated workcounter %d\n", expectedWKC);
         ec_slave[0].state = EC_STATE_OPERATIONAL;
         /* send one valid process data to make outputs in slaves happy*/
         ec_send_processdata();
         ec_receive_processdata(EC_TIMEOUTRET);
         /* request OP state for all slaves */
         ec_writestate(0);
         chk = 200;
         /* wait for all slaves to reach OP state */
         do
         {
            ec_send_processdata();
            ec_receive_processdata(EC_TIMEOUTRET);
            ec_statecheck(0, EC_STATE_OPERATIONAL, 50000);
         }
         while (chk-- && (ec_slave[0].state != EC_STATE_OPERATIONAL));
         if (ec_slave[0].state == EC_STATE_OPERATIONAL )
         {
            printf("Operational state reached for all slaves.\n");
            inOP = TRUE;
                /* cyclic loop */

            for(i = 1; i <= 3000; i++)
            {
               **set_output_int16(0x0001, 0x7002, 0x1010);**
               ec_send_processdata();
               wkc = ec_receive_processdata(EC_TIMEOUTRET);

                    if(wkc >= expectedWKC)
                    {
                        printf("Processdata cycle %4d, WKC %d , O:", i, wkc);

                        for(j = 0 ; j < oloop; j++)
                        {
                            printf(" %2.2x", *(ec_slave[0].outputs + j));
                        }

                        printf(" I:");
                        for(j = 0 ; j < iloop; j++)
                        {
                            printf(" %2.2x", *(ec_slave[0].inputs + j));
                        }
                        printf(" T:%"PRId64"\r",ec_DCtime);
                        needlf = TRUE;
                    }
                    osal_usleep(5000);
                }
                inOP = FALSE;
            }
            else
            {
                printf("Not all slaves reached operational state.\n");
                ec_readstate();
                for(i = 1; i<=ec_slavecount ; i++)
                {
                    if(ec_slave[i].state != EC_STATE_OPERATIONAL)
                    {
                        printf("Slave %d State=0x%2.2x StatusCode=0x%4.4x : %s\n",
                            i, ec_slave[i].state, ec_slave[i].ALstatuscode, ec_ALstatuscode2string(ec_slave[i].ALstatuscode));
                    }
                }
            }
            printf("\nRequest init state for all slaves\n");
            ec_slave[0].state = EC_STATE_INIT;
            /* request INIT state for all slaves */
            ec_writestate(0);
        }
        else
        {
            printf("No slaves found!\n");
        }
        printf("End simple test, close socket\n");
        /* stop SOEM, close socket */
        ec_close();
    }
    else
    {
        printf("No socket connection on %s\nExcecute as root\n",ifname);
    }
}

ArthurKetels, 2020-10-20:
I suggest to read the readme from SOEM.

Valneltin, 2020-10-20:
Hello,

Thanks for your answer.
I've already red the tutorial and the readme and many other issues. But I still don't understand the part "Accessing data thought IOmap" in the tutorial

For the void set_output_int16 : 

The index of the slave for me is 0x0001 because I have just 1 slave and I want to control the salve n°1.
The index of the module is 0x7000 but in the tutorial it indicates "module index as index internal to the slave in case more than one channel" . I don't understand this.
And for the value, this is the value I want to set to the module.

Please help me. I'm really blocked. 
Thanks for your help.

ArthurKetels, 2020-10-20:
The IOmap is a buffer where all inputs and outputs from all slaves are copied to/from. You provided the mapping.

> [0x0000.0] 0xF200:0x01 0x01 BOOLEAN K-Bus Cycle Overrun Flag Disable
[0x0000.1] 0xF200:0x02 0x01 BOOLEAN Input Process Data Hold Request
[0x0000.2] 0xF200:0x03 0x01 BOOLEAN Output Process Data Hold Request
[0x0000.3] 0xF200:0x04 0x01 BOOLEAN Output Process Data Clear Request
[0x0000.4] 0x0000:0x00 0x0C
[0x0002.0] 0xF200:0x05 0x10 UNSIGNED16 Diagnostics Control Word
[0x0004.0] 0x7000:0x01 0x01 BOOLEAN Channel 1 Data
[0x0004.1] 0x7000:0x02 0x01 BOOLEAN Channel 2 Data
[0x0004.2] 0x7000:0x03 0x01 BOOLEAN Channel 3 Data
[0x0004.3] 0x7000:0x04 0x01 BOOLEAN Channel 4 Data
[0x0004.4] 0x7000:0x05 0x01 BOOLEAN Channel 5 Data
[0x0004.5] 0x7000:0x06 0x01 BOOLEAN Channel 6 Data
[0x0004.6] 0x7000:0x07 0x01 BOOLEAN Channel 7 Data
[0x0004.7] 0x7000:0x08 0x01 BOOLEAN Channel 8 Data
[0x0005.0] 0x7000:0x09 0x01 BOOLEAN Channel 9 Data
[0x0005.1] 0x7000:0x0A 0x01 BOOLEAN Channel 10 Data
[0x0005.2] 0x7000:0x0B 0x01 BOOLEAN Channel 11 Data
[0x0005.3] 0x7000:0x0C 0x01 BOOLEAN Channel 12 Data
[0x0005.4] 0x7000:0x0D 0x01 BOOLEAN Channel 13 Data
[0x0005.5] 0x7000:0x0E 0x01 BOOLEAN Channel 14 Data
[0x0005.6] 0x7000:0x0F 0x01 BOOLEAN Channel 15 Data
[0x0005.7] 0x7000:0x10 0x01 BOOLEAN Channel 16 Data

This is as direct as anyone can make it. The address between [] is the actual byte offset in IOmap. The number after . is the bit offset. For example you want to control the 16 channels of output.

>uint16 *outchannels = (uint16 *)&IOmap[4];
>
>(your PDO loop)
> *outchannels = 0xffff; // all bits on

In the tutorial.txt you can also find:

>typedef struct PACKED
{
	int16	   outvalue1;
	int16	   outvalue2;
} out_EL4132t;
out_EL4132t		*out_EL4132;
...
   /* connect struct pointers to slave I/O pointers */
   out_EL4132 = (out_EL4132t*) ec_slave[3].outputs;
   out_EL4132->outvalue2 = 0x3FFF;

Each ec_slave[n].outputs is a pointer into IOmap starting at that slave n output section. In your case you can use ec_slave[1].outputs.

Valneltin, 2020-10-21:
Hello,

Thank you very much. My Wago was broken... I've try with a Beckhoff and it works perfectly.

closed 2020-10-21 by Valneltin