OpenEtherCATsociety/SOEM #456: Cannot reach OPERATIONAL with SAFE OP + ERROR

luoqiqiqi75, 2020-10-21:
Hi & Servus,
I tried _simple_test_ under Win10 with other EtherCAT slaver and it worked smoothly, but our slaver (ET1100) never changed into _OPERATIONAL_ state, after _ec_writestate(int)_ for a while, the slaver state went from _SAFE_OP_ to _ERROR_ + _SAFE_OP_. I can write or read SDO in _PRE_OP_ or _SAFE_OP_, only by changing to _OPERATIONAL_ it will fail.

The EC-Engineer also works fine, I configured _PO2SO_ with CoE (should be) correctly, so I think there might be some important parameters in ENI xml file I missed, however it is like fishing for a needle in the ocean. 

As follows is my slaver info, is there any clue or what does state _ERROR_ + _SAFE_OP_ mean?

> SOEM (Simple Open EtherCAT Master)
> Slaveinfo
> Starting slaveinfo
> ec_init on \\Device\\NPF_{DBC9D679-B4BF-42C7-AD76-B85278A83A04} succeeded.
> 1 slaves found and configured.
> Calculated workcounter 3
> 
> Slave:1
>  Name:INTERNAL_NAME
>  Output size: 192bits
>  Input size: 160bits
>  State: 4
>  Delay: 0[ns]
>  Has DC: 1
>  DCParentport:0
>  Activeports:1.0.0.0
>  Configured address: 1001
>  Man: 00078557 ID: 00015370 Rev: 00020970
>  SM0 A:1000 L: 128 F:00010026 Type:1
>  SM1 A:1080 L: 128 F:00010022 Type:2
>  SM2 A:1100 L:  20 F:00010064 Type:4
>  SM3 A:1400 L:  24 F:00010020 Type:3
>  FMMU0 Ls:00000000 Ll:  24 Lsb:0 Leb:7 Ps:1400 Psb:0 Ty:02 Act:01
>  FMMU1 Ls:00000018 Ll:  20 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:01 Act:01
>  FMMUfunc 0:1 1:2 2:3 3:0
>  MBX length wr: 128 rd: 128 MBX protocols : 04
>  CoE details: 2f FoE details: 00 EoE details: 00 SoE details: 00
>  Ebus current: 0[mA]
>  only LRD/LWR:0
> PDO mapping according to CoE :
>   SM2 inputs
>      addr b   index: sub bitl data_type    name
>   [0x0018.0] 0x607A:0x00 0x20 INTEGER32    T
>   [0x001C.0] 0x60FF:0x00 0x20 INTEGER32    T
>   [0x0020.0] 0x60B1:0x00 0x20 INTEGER32    V
>   [0x0024.0] 0x6071:0x00 0x10 INTEGER16    T
>   [0x0026.0] 0x60B2:0x00 0x10 INTEGER16    T
>   [0x0028.0] 0x6040:0x00 0x10 UNSIGNED16   C
>   [0x002A.0] 0x6060:0x00 0x08 INTEGER8     M
>   [0x002B.0] 0x2001:0x00 0x08 UNSIGNED8    r
>   SM3 outputs
>      addr b   index: sub bitl data_type    name
>   [0x0000.0] 0x6064:0x00 0x20 INTEGER32    P
>   [0x0004.0] 0x606C:0x00 0x20 INTEGER32    V
>   [0x0008.0] 0x60F4:0x00 0x20 INTEGER32    F
>   [0x000C.0] 0x6077:0x00 0x10 INTEGER16    T
>   [0x000E.0] 0x6041:0x00 0x10 UNSIGNED16   S
>   [0x0010.0] 0x2401:0x00 0x10 UNSIGNED16   A
>   [0x0012.0] 0x2402:0x00 0x10 UNSIGNED16   A
>   [0x0014.0] 0x603F:0x00 0x10 UNSIGNED16   E
>   [0x0016.0] 0x6061:0x00 0x08 INTEGER8     M
>   [0x0017.0] 0x2000:0x00 0x08 UNSIGNED8    t
> End slaveinfo, close socket
> End program
>

ArthurKetels, 2020-10-21:
You have a bug in your slave firmware (Beckhoff slave sample code) similar as #445 .
Either patch the slave firmware or patch SOEM.

luoqiqiqi75, 2020-10-23:
@ArthurKetels Thanks a lot for quick response! Yet I am not in our hardware team, they are sure using sample code. Unfortunately their slave firmware cannot be updated for many company reasons. I did the patch metioned at #224 unwillingly, the BRD _ec_slave[0]_ went to OP and the only slave _ec_slave[1]_ went back to PRE_OP. In my own program, after _ec_checkstate_ its state changes to SAFE_OP + ERROR again.

ArthurKetels, 2020-10-23:
From the picture above all seems to be OK, all slaves are in OP. The simpletest program is supposed to run in a long loop when in OP (not infinite).

When in doubt send a wireshark trace of the OK simpletest run. Then I can confirm correct configuration and state transitions.

If you want you can implement your own code to read ESI files and configure SOEM that way. SOEM is just giving you the building blocks to make your own master to your own liking. 

> Unfortunately their slave firmware cannot be updated for many company reasons.

This is of course a bogus argument. The code has a bug that prevents the slave from being EtherCAT compliant. It will not pass the mandatory conformance test. Simple, bad firmware -> bad product.

luoqiqiqi75, 2020-10-26:
The "Operational state reached for all slaves." is an inaccurate log in my case, _ec_slave[0]_ is _0x08_ but _ec_slave[1]_ remains _0x02_ . Now I am still doubt about the reason of "Big Endian", so here is my outputs with/without patch & wireshark traces:

ArthurKetels, 2020-10-26:
I think I know what your issue is. It has nothing to do with your slave state. The wireshark trace of ET1100_ok_simple_test is 100% OK. The slave does indeed go to operational and reports no errors. 

What you have to keep in mind is that ec_slave[1].state is not an accurate representation of the current state of the slave. This is an optimization issue. SOEM only reads the slave state when an working counter error is reported back in the PDO. It is guaranteed that the slaves remain in OP as long as the working-counter equals the calculated working-counter. Constantly reading the per slave state would just use up bandwidth for no use. 

Second, when you request to read the slave state per ec_readstate() you have to be careful to keep the PDO exchange running. Otherwise you would get a watchdog timeout. And this results in reading 0x14 as state (probably with watchdog error reported in the slave error state register). To properly read the per slave stae see how the function ecatcheck() works in simple_test.c. When simple_test is running simply unplug the slave and then reconnect. You can see ecatcheck starting to work to rediscover and bring the slave back to OP.

luoqiqiqi75, 2020-10-27:
Thanks a lot! Now the slaver is running as desired from last day!

Conclusions:
1. The "Big Endian Patch" #224 did work.
2. Solution in [https://www.ethercat.org/memberarea/forum.asp?../memberarea/forum/free-download-slave-stack-code-v5-12_topic1590.html](url) currently takes no obvious effect.
3. Sorry for wasting you too much time that I mixed up _test/linux/simple_test_ keeping OP with long last loop and _test/win32/simple_test_ with _timeSetEvent_.

I wish you a nice week (*°∀°)=3

closed 2020-10-27 by luoqiqiqi75