OpenEtherCATsociety/SOEM #457: SOEM Distributed Clock management

mictas, 2020-10-23:
Hello,

we are using SOEM with ROS to manage EtherCAT drivers. 
The operating system used is Ubuntu 18.04 LTS with ROS "Melodic".
The communication works and we can command movements to the motors.
We are trying to understand better how "Distributed Clock" mechanism works in SOEM.

We need the following clarifications:

1) If the following sequence of commands is used in our "ROS EtherCAT Master" (it is a roscpp node based on simple_test.cpp which uses SOEM):

ec_dcsync0( 1, TRUE, 1000000U, 0U )
ec_configdc()

is "Distributed Clock" automatically setted for all connected slaves and the reference clock adopted for the entire system is the one of the 1st slave connected to the EtherCAT Master?

2) In some examples found on the web sometimes "ec_configdc()" is written before "ec_dcsync0" while in other cases after it.
   Is there a correct order for writing these two commands?

3) Typing the commands listed in 1), we deduce that automatically Master initiates DC feature of slaves setting as Reference  Clock the Local clock of slave 1.
   Did we understand it right?

4) How "Propagation Delay", "Offset", and "Drift" measure and compensation are managed? Are they automatically implemented and executed by the "SOEM library" or the user has to implement a proper code?

5) Is it necessary to set some specific registers or Object Dictionary PDO entries in each slave to activate DC functionalities on it?

Waiting for a your response, thank you.

ArthurKetels, 2020-10-23:
First, ec_dcsync0() and ec-sync01() program the slave onboard pulse generators. In itself they have nothing to do with distributed clocks. Each slave has its own local clock from which the sync pulses are generated. You can program a single pulse or a repetitive pulse. There are two generators 0 and 1, generator 0 (called sync0) has an absolute start time and a repetition time. Sync1 is similar but act as a slave of sync0 (for details see Beckhoff slave controller datasheet).

Second, each slave (that is DC capable as are most) has a hardware implemented mechanism to synchronize its local clock with an external clock. From the master perspective this is very simple. To get the local clock value read the slave local clock register. To sync on an external clock write the value of the external clock in the slave local clock register. The save then calculates the drift and offset (including programmed propagation delay) and corrects its local clock. In order to have a stable synchronization this read once and write multiple clock (multiple slaves) must be repeated on a regular basis. The most obvious spot being the PDO cycle.

When first and second system are combined we can generate system wide synchronized pulses.

SOEM tries to abstract most of the difficult bookkeeping regarding DC and Sync. When ec_configdc() is executed a lot of measuring takes place.
1. Scan all DC capable slaves.
2. Measure propagation delays between each slave. This is a lot more involved than that simple sentence would infer.
3. Programs each propagation delay into the proper slave DC register.
4. Designate the reference slave. This is the master clock. For now this is the first DC capable slave in the slave topology.
5. Pre-set all slave local clocks to the master clock (rough synchronization).
6. Set proper flags so the PDO cycle knows how to distribute the clocks.

When the PDO cycle is running it now does all DC work automatically. After a short time all slaves have received enough reference clock data to synchronize themselves. It is up to the user to verify if all slaves are synchronized and remain synchronized. To do this you can monitor the DC status registers of each slave. It is up to you to define proper thresholds, when is a slave regarded as "synchronized".

Following step 5 above it is not wise to execute ec_dcsync0() before ec_configdc(). The clock is programmed at absolute time and then that same clock is changed. Proper behavioure of the sync0 is not guaranteed in this case.

Slave local clock offset and drift are taken care of by the slave hardware. Propagation delays are managed by ec_configdc(). The sync0 offset is a parameter of ec_dcsync0(). The definition is : the offset added to the absolute time modulo the repetition time. For example, when the repetition time is set at 1ms and the offset at zero, then sync0 fires at each whole multiple of 1000000ns. When the offset is +50000ns then sync0 is fired at 1050000ns, 2050000ns, 3050000ns etc. 

The slave hardware is now set-up for DC synchronization and pulse generation. However there could be extra steps to take in a slave to guarantee proper functioning. For this you have to consult the slave documentation.

Also DC and sync do not cover the master synchronization to the reference slave. It is optional, but many slaves will complain if the master is not properly synchronized. On how to do this look at the red_test example in SOEM.

Of course there is a lot more to tell about details and possible strategies, but that would make this post far too long.

mictas, 2020-12-07:
Thank you very much for your response and sorry for my late reply.

closed 2020-12-07 by mictas