OpenEtherCATsociety/SOEM #459: state = 11, Statuscode=16 : Invalid mailbox configuration

foreversilver, 2020-11-09:
[image]

Hi everyone,
I encountered this problem when I try to connect a EtherCAT slave with SOEM, and this is the output of slave info. When I try to debug the source code I saw that is because the mailbox size and offset read from eeprom are all ZERO. But I can connect the device with TwinCAT successfully with ESI file. I want  to ask that whether I can configure the mailbox info before I start the SOEM master station and how can I do it?

Thank you very much!

ArthurKetels, 2020-11-09:
Theoretically you could do that, but the EEPROM on the slave has more issues. In the TxPDO you can see that the 3rd entry has data type 0x0000, this is also invalid. The SM types of SM0 and SM1 are 0, and they should be 1 and 2. 
If you could post the ESI xml file and the EEPROM bin file (retrieved with eepromtool) then I can try to rectify the EEPROM data.

The best solution of course would be an update from the slave vendor. This is again an example of a non conformant slave.

foreversilver, 2020-11-09:
Hi Ketels, 
Thank so much for answering me. Yes I found the two problem also, and I can upload the ESI xml now, but I can only post the EEPROM bin file tomorrow. What do you mean by rectifying the EEPROM data? Do you mean rectify the EEPROM data and write it back to the EEPROM of the device? I  think I can also do this through TwinCAT and that would be easier right? Also, I do not want to do that because I do not know if it will affect other from connecting it with another EtherCAT master. Can you help me by configuring the PDO and MBX during the initialization of SOEM master station? It can be done in the INIT state right? Or maybe we can do it by change the source code.

Thanks a lot again!
[XPR_SPI_slave_ECAT_Plasma.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5511392/XPR_SPI_slave_ECAT_Plasma.zip)

ArthurKetels, 2020-11-09:
Thanks for the ESI file. This file looks fine. And from its content you can generate the correct SII EEPROM data. If you have TwinCAT that would be the easiest path (it can generate the SII automatically from the ESI). Updating the EEPROM is safe for use in other masters. As you have noticed already, most masters do not look in the EEPROM for other data than the slave identification. SOEM does not use ESI and therefore has to rely on the SII.

If you want to correct the data in SOEM then you have to indeed do it in INIT. First you set manualstatechange to 1, then call ecx_config_init(). All slaves are now enumerated and basic data from SII is put in the ec_slave struct. The manualstatechange needs to be set to prevent the state change to pre-OP (slave has wrong maibox SM configuration).
You now have to identify the correct slave number. Then modify the ec_slave struct.
For example : 
`               ec_slave[slave].SM[0].StartAddr = htoes(0x1000);`
`               ec_slave[slave].SM[0].SMlength = htoes(0x0080);`
`               ec_slave[slave].SM[0].SMflags = htoel(EC_DEFAULTMBXSM0);`
`               ec_slave[slave].SMtype[0] = 1; `
You have to search for and replace all struct items that have the wrong values (see ethercatconfig.c).
Then rewrite the SM configuration : ecx_FPWR(context->port, configadr, ECT_REG_SM0, sizeof(ec_smt) * 2, &(context->slavelist[slave].SM[0]), EC_TIMEOUTRET3);
Then put all slaves in pre-OP: ecx_FPWRw(context->port, configadr, ECT_REG_ALCTL, htoes(EC_STATE_PRE_OP | EC_STATE_ACK),       EC_TIMEOUTRET3);
Last you can continue the normal mapping with : ecx_config_map().

foreversilver, 2020-11-12:
Hi AuthurKetels,
I have done the changes to my code according to your suggestion and it works now. I read the bootstrap mailbox data from EEPROM and assigned it to standard mailbox. Then I configure PDO assignment to correct the wrong data type of TxPDO entry.
Thank you very much for your response!

closed 2020-11-12 by ArthurKetels