OpenEtherCATsociety/SOEM #461: PDO Mapping and motor control

jayanthendo, 2020-11-12:
Hello guys,
i am new to EtherCat and SOEM library i have tried the simple test of linux version on my technosoft IPOS-8020 BX-CAT and these are the results
```
`SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on enp1s0 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 22 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Operational state reached for all slaves.
Processdata cycle    1, WKC 3 , O: 00 00 00 00 00 00 I: 48 02 00 00 00 00 00 00 Processdata cycle    2, WKC 3 , O: 00 00 00 00 00 00 I: 48 02 00 00 00 00 00 00 Processdata cycle    3, WKC 3 , O: 00 00 00 00 00 00 I: 48 02 00 00 00 00 00 00 Processdata cycle    4, WKC 3 , O: 00 00 00 00 00 00 I: 48 02 00 00 00 00 00 00 Processdata cycle    5, WKC 3 , O: 00 00 00 00 00 00 I: 48 02 00 00 00 00 00 00 Processdata cycle    6, WKC 3 , O: 00 00 00 00 00 00 I: 48 02 00 00 00 00 00 00 Processdata cycle    7, WKC 3 , O: 00 00 00 00 00 00 I: 48 02 00 00 00 00 00 00 Processdata cycle    8, WKC 3 , O: 00 00 00 00 00 00 I: 48 02 00 00 00 00 00 00 Processdata cycle    9, WKC 3 , O: 00 00 00 00 00 00 I: 48 02 00 00 00 00 00 00 Processdata cycle   10, WKC 3 , O: 00 00 00 00 00 00 I: 48 02 00 00 00 00 00 00 T:3348286611138349440
Request init state for all slaves
End simple test, close socket
End program
`
```
And for slave info this is the result 
```
`SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enp1s0 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:iPOS8020 BX-CAT
 Output size: 48bits
 Input size: 128bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 000001a3 ID: 01bae7ad Rev: 3531354a
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:   6 F:00010024 Type:3
 SM3 A:1c00 L:  16 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   6 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000006 Ll:  16 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0e
 CoE details: 0f FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program
`
```
Now my doubts are

1. the workcounter is giving 3 but i have only one slave
2. is it necessary to map pdo or it will be mapped automatically.
3. i am trying to map the PDO after preoperational state using the hook POSOconfig as mentioned in tutorial.txt  but the changes are not being the reflected in sm2 and sm3 .
4. and after mapping the pdo's correctly how can i control the motor, in the tutorial is mentioned that we can access using the IOmap and ec_slave[1].outputs so is it correct?

Here is the code that i am trying 
```
`#define control_word 0x6040
#define status_word 0x6041
#define mod_op 0x6060
#define sync_ch_2 0x1C12
#define sync_ch_3 0x1C13
#define Tx_PDO1 0x1A00
#define Tx_PDO2 0x1A01
#define Tx_PDO3 0x1A02
#define Tx_PDO4 0x1A03
#define Rx_PDO1 0x1600
#define Rx_PDO2 0x1601
#define Rx_PDO3 0x1602
#define Rx_PDO4 0x1603
int setup_drive(uint16_t slave){
  //sync 2,3
  ob2 = 0x0000;
  ec_SDOwrite(slave,sync_ch_2,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x0000;
  ec_SDOwrite(slave,sync_ch_3,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  //TXPDO1
  ob2 = 0x656C630;
  ec_SDOwrite(slave,Tx_PDO1,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x60410016;
  ec_SDOwrite(slave,Tx_PDO1,01,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x60640032;
  ec_SDOwrite(slave,Tx_PDO1,02,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x60770016;
  ec_SDOwrite(slave,Tx_PDO1,03,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x776F643;
  ec_SDOwrite(slave,Tx_PDO1,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x656C630;
  //TXPDO2
  ec_SDOwrite(slave,Tx_PDO2,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x60F40032;
  ec_SDOwrite(slave,Tx_PDO2,01,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x60FD0032;
  ec_SDOwrite(slave,Tx_PDO2,02,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x776F642;
  ec_SDOwrite(slave,Tx_PDO2,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  //TXPDO3
  ob2 = 0x656C630;
  ec_SDOwrite(slave,Tx_PDO3,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x606C0032;
  ec_SDOwrite(slave,Tx_PDO3,01,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x776F641;
  ec_SDOwrite(slave,Tx_PDO3,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x776F642;
  ec_SDOwrite(slave,Tx_PDO3,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  //Tx_PDO4
  ob2 = 0x656C630;
  ec_SDOwrite(slave,Tx_PDO4,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x6061008;
  ec_SDOwrite(slave,Tx_PDO4,01,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x776F641;
  ec_SDOwrite(slave,Tx_PDO4,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  //Rx_PDO1
  ob2 = 0x656C630;
  ec_SDOwrite(slave,Rx_PDO1,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x60400016;
  ec_SDOwrite(slave,Rx_PDO1,01,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x607A0032;
  ec_SDOwrite(slave,Rx_PDO1,02,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x776F642;
  ec_SDOwrite(slave,Rx_PDO1,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  //Rx_PDO2
  ob2 = 0x656C630;
  ec_SDOwrite(slave,Rx_PDO2,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x60FE0132;
  ec_SDOwrite(slave,Rx_PDO2,01,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x60FE0232;
  ec_SDOwrite(slave,Rx_PDO2,02,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x776F642;
  ec_SDOwrite(slave,Rx_PDO2,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  //Rx_PDO3
  ob2 = 0x656C630;
  ec_SDOwrite(slave,Rx_PDO3,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x6060008;
  ec_SDOwrite(slave,Rx_PDO3,01,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x60710016;
  ec_SDOwrite(slave,Rx_PDO3,02,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x776F642;
  ec_SDOwrite(slave,Rx_PDO3,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  //Rx_PDO4
  ob2 = 0x656C630;
  ec_SDOwrite(slave,Rx_PDO4,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x60810032;
  ec_SDOwrite(slave,Rx_PDO4,01,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x60830032;
  ec_SDOwrite(slave,Rx_PDO4,02,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x776F642;
  ec_SDOwrite(slave,Rx_PDO4,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  //sync 2,3
  ob2 = 0x01;
  ec_SDOwrite(slave,sync_ch_2,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x1A00;
  ec_SDOwrite(slave,sync_ch_3,01,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x1A01;
  ec_SDOwrite(slave,sync_ch_3,02,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  ob2 = 0x02;
  ec_SDOwrite(slave,sync_ch_3,00,FALSE,sizeof(ob2),&ob2,EC_TIMEOUTRXM);
  return 0;
}
`
```
`ec_slave[1].PO2SOconfig = setup_drive;`
Please help me i am struck here

nakarlsson, 2020-11-12:
1. the workcounter is giving 3 but i have only one slave 
- It is as expected, the command LRW (Logic read write) generate +1 for inputs + 2 for outputs = 3 WCK

2. is it necessary to map pdo or it will be mapped automatically.
- You should get a default mapping, check the manual/or ESI or do a slaveinfo -map dump.

3. i am trying to map the PDO after preoperational state using the hook POSOconfig as mentioned in tutorial.txt but the changes are not being the reflected in sm2 and sm3 .
- The code snippet above got some issues. The write sizeof must macth the targeted SDO.

Write SDO 0x1c12:00 with size of that subundex, uint8
Write SDO 0x1c12:01 with size of that subindex uint16.

If you check the result of the ec_SDOwrite it should return > 0 on success . You'll probably also see some SDO aborts in a wireshark if you do a capture.

4. and after mapping the pdo's correctly how can i control the motor, in the tutorial is mentioned that we can access using the IOmap and ec_slave[1].outputs so is it correct?
- Correct. ec_slave[1].outputs points to the IOmap output portion for slave 1.

Pseudo
Say that a CiA401 Control/Statusword is mapped as first PDO
uint16_t * control_word = (uint16_t *)ec_slave[1].outputs;
uint16_t * status_word = (uint16_t *)ec_slave[1].inputs;

Search for older issues and you'll find samples.

*control_word = SWITCH_ON
send_processdata
receive_processdata
printf "current status is %x\n", *status_word

jayanthendo, 2020-11-13:
Thanks @nakarlsson for the reply
i have seen status in pdo mapping and 2 are returning 0 so i have modified it and mapped correctly 
But after mapping and in operational mode when i am able to change the pdo values but if i want to change the objects other than in pdo using ec_SDOwrite its giving segmentation fault
can i change the values other than in pdo or is there any other way to do it to drive the motor.

nakarlsson, 2020-11-13:
can you take a wireshark log, it might help understanding

jayanthendo, 2020-11-13:
i have observed that when i am writing then no error but its changing its state and this is the log file

[ipos_log.txt](https://github.com/OpenEtherCATsociety/SOEM/files/5536386/ipos_log.txt)

when i insert read command its segmentation fault and this is the log file

[motor_log.txt](https://github.com/OpenEtherCATsociety/SOEM/files/5536398/motor_log.txt)

nakarlsson, 2020-11-23:
post wireshark logs in pcapng format please

jayanthendo, 2020-11-24:
@nakarlsson  i am uploading the log files, please check them 
[logs.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5588465/logs.zip)

jayanthendo, 2020-12-02:
@nakarlsson  have you found anything from the wireshark logs?

nakarlsson, 2020-12-02:
I can't see any obvious what could cause a segmentation fault, I guess it is much easier for you to find where you have the memory violate .

Other than that, you shouldn't mix Asynchronous and Synchronous communication, e.g. remove the SDO read/write from the cyclic process data loop. Also remove the get status, if you check ecatcheck that does the diagnostics for you.

So, clean up the code. Seperate the process data from SDO and diagnostics and you should get it to work.

jayanthendo, 2020-12-02:
@nakarlsson Thanks for reply
i have updated the code but still i am unable to run the motor 
here is my updated code 
```
void simpletest(char *ifname)
{
    int i, j, oloop, iloop, chk;
    needlf = FALSE;
    inOP = FALSE;
    
    int usedmem;
    out_EL4132t *out_EL4132;

   printf("Starting simple test\n");

   /* initialise SOEM, bind socket to ifname */
   if (ec_init(ifname))
   {
      printf("ec_init on %s succeeded.\n",ifname);
      /* find and auto-config slaves */

       if ( ec_config_init(FALSE) > 0 )
      {
         printf("%d slaves found and configured.\n",ec_slavecount);
         ec_slave[1].PO2SOconfig = speed_profile_startup;

         printf("Name: %s EEpMan: %d eep_id: %d configadr: %d aliasadr: %d State %d\n\r",ec_slave[1].name,ec_slave[1].eep_man,\
         ec_slave[1].eep_id,ec_slave[1].configadr,ec_slave[1].aliasadr,ec_slave[1].state );//osal_usleep(100000);

         usedmem = ec_config_map(&IOmap);
         
         printf("usedmem :%d sizeofIOmap %d\n\r",usedmem,sizeof(IOmap));
         
         si_map_sdo(1);

         //ec_configdc();

         printf("Slaves mapped, state to SAFE_OP.\n");
         /* wait for all slaves to reach SAFE_OP state */
         ec_statecheck(1, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE * 4);

         oloop = ec_slave[0].Obytes;
         if ((oloop == 0) && (ec_slave[0].Obits > 0)) oloop = 1;
         if (oloop > 8) oloop = 8;
         iloop = ec_slave[0].Ibytes;
         if ((iloop == 0) && (ec_slave[0].Ibits > 0)) iloop = 1;
         if (iloop > 8) iloop = 8;

         printf("segments : %d : %d %d %d %d\n",ec_group[0].nsegments ,ec_group[0].IOsegment[0],ec_group[0].IOsegment[1],ec_group[0].IOsegment[2],ec_group[0].IOsegment[3]);
         
         sp_out = (Speed_outputs *)ec_slave[1].outputs;
         sp_in = (Speed_inputs *)ec_slave[1].inputs;

         //printf("State %d status_word %d %X\n\r",ec_slave[1].state, sp_in->status_word, sp_in->status_word);
         printf("Request operational state for all slaves\n");
         //expectedWKC = (ec_group[0].outputsWKC * 2) + ec_group[0].inputsWKC;
         //printf("Calculated workcounter %d\n", expectedWKC);
         ec_slave[1].state = EC_STATE_OPERATIONAL;
         /* send one valid process data to make outputs in slaves happy*/
         ec_send_processdata();
         ec_receive_processdata(EC_TIMEOUTRET);
         /* request OP state for all slaves */
         ec_writestate(1);
         chk = 200;
         /* wait for all slaves to reach OP state */
         do
         {
            ec_send_processdata();
            ec_receive_processdata(EC_TIMEOUTRET);
            ec_statecheck(0, EC_STATE_OPERATIONAL, 50000);
[... 107 lines cut ...]
  [0x000D.0] 0x606C:0x00 0x20 INTEGER32    
  [0x0011.0] 0x6064:0x00 0x20 INTEGER32    
  [0x0015.0] 0x60FD:0x00 0x20 UNSIGNED32   
Slaves mapped, state to SAFE_OP.
segments : 1 : 25 0 0 0
Request operational state for all slaves
Operational state reached for all slaves.
State 8 status_word 592
State 8 status_word 592 Velocity_actual 0 
State 8 status_word 5712 Velocity_actual 0 
State 8 status_word 5712 Velocity_actual 0 
State 8 status_word 5712 Velocity_actual 0 
State 8 status_word 5712 Velocity_actual 0 
State 8 status_word 5712 Velocity_actual 0 
State 8 status_word 5712 Velocity_actual 0 
State 8 status_word 5712 Velocity_actual 0 
State 8 status_word 5712 Velocity_actual 0 
State 8 status_word 5712 Velocity_actual 0 

Request init state for all slaves
End simple test, close socket
End program

```
After that the drive is going to fault

nakarlsson, 2020-12-02:
This sequence below is not correct, the IOmap is only sent to the ethercat slaves when you do  ec_send_processdata();

So in theory , you would at least do somethink like

            sp_out->mode_of_operation = 3; 
            ec_send_processdata();
           wkc = ec_receive_processdata(EC_TIMEOUTRET);
            sp_out->control_word = 6; 
            ec_send_processdata();
            wkc = ec_receive_processdata(EC_TIMEOUTRET);
            sp_out->control_word = 7;
            ec_send_processdata();
            wkc = ec_receive_processdata(EC_TIMEOUTRET);
            sp_out->control_word = 0x0F; 
            ec_send_processdata();
            wkc = ec_receive_processdata(EC_TIMEOUTRET);
            sp_out->target_velocity = 0x00140000;
            ec_send_processdata();
            wkc = ec_receive_processdata(EC_TIMEOUTRET);

The above code is very simplifyed, you'll need to implement  a state machine to bring the moter to switched  on.

jayanthendo, 2020-12-02:
ok i have modified it and this is the updated one
```
if (ec_slave[1].state == EC_STATE_OPERATIONAL )
         {
            printf("Operational state reached for all slaves.\n");
            inOP = TRUE;
            sp_out->mode_of_operation = 0x3;
            ec_send_processdata();
            wkc = ec_receive_processdata(EC_TIMEOUTRET); 
            sp_out->control_word = 0x6;
            ec_send_processdata();
            wkc = ec_receive_processdata(EC_TIMEOUTRET);  
            sp_out->control_word = 0x7;
            ec_send_processdata();
            wkc = ec_receive_processdata(EC_TIMEOUTRET); 

            sp_out->control_word = 0x0F;
            ec_send_processdata();
            wkc = ec_receive_processdata(EC_TIMEOUTRET);  
            sp_out->target_velocity = 0x00140000;
            ec_send_processdata();
            wkc = ec_receive_processdata(EC_TIMEOUTRET); 

            printf("State %d status_word %d\n\r",ec_slave[1].state, sp_in->status_word);

            
            /* cyclic loop */
            for(i = 1; i <= 10; i++)
            {
               ec_send_processdata();
               wkc = ec_receive_processdata(EC_TIMEOUTRET);
               sp_out->target_velocity = 0x00140000;

               printf("State %d status_word %d Velocity_actual %d \n\r",ec_slave[1].state, sp_in->status_word, sp_in->velocity_actual);   
               osal_usleep(5000);

            }
                inOP = FALSE;
            }
```
But same problem continues
I am attaching the wireshark log file maybe this helps you more 
[ipos_log1.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5628725/ipos_log1.zip)

and this is the manual of driver
[P091.064.EtherCAT.iPOS_.UM_.pdf](https://github.com/OpenEtherCATsociety/SOEM/files/5628732/P091.064.EtherCAT.iPOS_.UM_.pdf)

nakarlsson, 2020-12-02:
I think the "manual of the driver"  is for you to read.

As I said, my code was simplified to show that you need to send the processdata, Stepping through a state machine as shown below is not the way to go. you should request a state change, wait and check that it is done. Then request next state. On errors to step up the state machine, consult the manual to find out what is the problem...

            sp_out->mode_of_operation = 0x3;
            ec_send_processdata();
            wkc = ec_receive_processdata(EC_TIMEOUTRET); 
            sp_out->control_word = 0x6;
            ec_send_processdata();
            wkc = ec_receive_processdata(EC_TIMEOUTRET);  
            sp_out->control_word = 0x7;
            ec_send_processdata();
            wkc = ec_receive_processdata(EC_TIMEOUTRET);

jayanthendo, 2020-12-03:
@nakarlsson Thank you for your advice after giving wait time, the drive started to drive the motor.
Thank you for assisting me. and i will close this issue.

closed 2020-12-03 by jayanthendo

bharatgohil, 2025-07-05:
Hi All,

Using this thread, I am able Run "SG-ASE" drive but It is working in CSV mode not in CSP mode. 
What are modification in above code to run in CSP mode?
Same drive working with Twincat and able to control the position.