OpenEtherCATsociety/SOEM #462: Is it necessary to develop a PDO access function like SDO_Write/Read?

foreversilver, 2020-11-13:
Hi everyone,
How do you usually access the PDO data? I created a struct and assign the outputs pointer of the slave to the struct, so I  can access the IOmap directly to get PDO. For example, I create a struct like:
```
struct Output {
uint16 control_;
uint32 contro2_;
}

struct Output *output = ec_slave[slave].outputs;
uint16 control_data = output->control;
```
But when I have multiple slaves this needs lots of lock to protect the data.

So I  want to ask if I can develop two api like PDO_Wrtite/Read(uint8 slave, uint16 index, uint8 subindex, ...);
Or Can you share what strategy do you think  is better to access the PDO data?

Thanks a lot!

nakarlsson, 2020-11-13:
Can't you just have a single lock and swap the IOmap or batch read the required slaves?

briansoe66, 2020-12-03:
I think accessing the PDO that struct only works on little endian systems. On big endian systems, you will need to byte swap each value.

If you do all the data processing within the master thread while loop, then you don't need locks.

```
while(1){
clock_nanosleep();
ec_send_processdata();
ec_receive_processdata();
my_data_processing();
```

david9991, 2020-12-23:
We use IOmap directly instead. It mapping to the concept of Process Image which mentioned in both EtherCAT and CANopen standard documentations, and each element (defined by its `offset` and `size`) is calculated according to ENI or DCF files and mapping to an logical memory area (sometimes even non-continuously and hard to be defined as a `struct`), so I think, accessing them in IOmap or ec_slave[slave].outputs/inputs by their calculated `offset` and `size` is more reasonable. 

So in your example: the API `PDO_Wrtite/Read(uint8 slave, uint16 index, uint8 subindex, ...);`, how do you map Index, SubIndex to a location inside Process Image without knowledge the full content of ENI or DCF files?
And you mentioned SDO_Write() is something very different. SDO_write() is nothing about the Process Image, the CoE AL can send the command with Index, Subindex, Data directly without knowing where it is in Process Image, that's a different story.

Our solution is compile the ENI or DCF files to a linkable object, it contains all object(by global variables) those defined in ENI/DCF and a callback function _(void *processImage) which bidirectional update global variables and Process Image according to the ENI/DCF files, So the program can update process images directly without knowing the content of ENI/DCF files, so Apps only need to do is read and write these global variables.

foreversilver, 2020-12-23:
Thanks every boby, I got your points and I am going to close this issue.

closed 2020-12-23 by foreversilver