OpenEtherCATsociety/SOEM #463: Problem of cannot find any slave.

miao-lastwish, 2020-11-13:
HI, all

I have a custom AM335x busybox board, which use kernel:
`Linux 4.9.147-ti-rt-r121 #1 SMP PREEMPT RT Thu May 23 21:52:09 UTC 2019 armv7l GNU/Linux`

This board have two network port eth0/eth1, it works well as common network usage.

I just want use eth0 as ethercat master port to connect BECKHOFF EK1100/EL2809 via `simple_test`

But it returns me:
```
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on eth0 succeeded.
No slaves found!
End simple test, close socket
End program
```

I notice that eth0 port only has TX packages and with 0 RX packages.
```
eth0      Link encap:Ethernet  HWaddr 28:EC:9A:FE:0E:74
          inet6 addr: fe80::2aec:9aff:fefe:e74/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
                    ~~~
          TX packets:504 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:76231 (74.4 KiB)
          Interrupt:176
```

I have tried to use eth1, get the same result.

Besides, while boot up, eth0 eth1 is all added to br0, so I use `brctl delif br0 eth0` to remove eth0 from br0.
After that the net info is:
```
/etc/network # ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: can0: <NOARP40000> mtu 16 qdisc noop qlen 10
    link/[280]
3: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq qlen 1000
    link/ether 28:ec:9a:fe:0e:74 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::2aec:9aff:fefe:e74/64 scope link
       valid_lft forever preferred_lft forever
4: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP100> mtu 1500 qdisc mq master br0 qlen 1000
    link/ether 28:ec:9a:fe:0e:76 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::2aec:9aff:fefe:e76/64 scope link
       valid_lft forever preferred_lft forever
5: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue qlen 1000
    link/ether 28:ec:9a:fe:0e:76 brd ff:ff:ff:ff:ff:ff
    inet 172.16.7.7/20 scope global br0
       valid_lft forever preferred_lft forever
    inet 172.16.8.8/8 scope global br0
       valid_lft forever preferred_lft forever
    inet6 fe80::2aec:9aff:fefe:e74/64 scope link
       valid_lft forever preferred_lft forever
```

Any help would be great appreciated.

nakarlsson, 2020-11-13:
IS this related to #441?

miao-lastwish, 2020-11-16:
> Is this related to #441?

Thanks, I'll read it.

miao-lastwish, 2020-12-04:
I changed the kernel to 4.9.45 and the problem fixed, it should be the bug of kernel.

closed 2020-12-04 by miao-lastwish