OpenEtherCATsociety/SOEM #464: Setting digital Outputs with PDOs

joseyy, 2020-11-18:


ArthurKetels, 2020-11-18:
I guess you answered your own question. Often DS402 slaves need to have the local state machine in an OP state in order to set digital outputs. But strictly it is undefined behavior and can change between different vendors.

joseyy, 2020-11-18:
Thanks for replying back. As you said it might be undefined behavior but I thought since I am able to toggle the digital outputs using SDOwrite it wouldn't be a problem doing so using the IOmap. 

The problem is that I haven't been able to make the slave switch to Operation Enabled (which is the last state). I can, however, switch the slave all the way to Switch ON (the state before Operation Enabled) but when I send the command to switch to "Operation Enabled" it goes into the "Fault" state (this is more of a DS402 question than SOEM), **what are some of the things I should look at to find a solution to this problem?**

Note: The motor is connected to the slave and we are able to send commands to it using the software supplied by the vendor.

ArthurKetels, 2020-11-19:
A bit more info could help to understand your problems. What slave? Post, ESI file of the slave, your code to bring the slave to OP and command the DS402 state machine, Wireshark dump of your last attempt.

As always, when something does not work, simplify. Try to set outputs on a simple slave with only digital outputs.

nakarlsson, 2021-02-26:
Can we close this issue?

joseyy, 2021-02-26:
Yes you can close it

closed 2021-02-26 by nakarlsson