OpenEtherCATsociety/SOEM #465: Error Sending PDO to Elmo controller

ddrouin17, 2020-11-20:
I am using an Elmo Gold Solo Double Twitter controller 

I am trying to use pdo entry Physical Outputs to enable Digital Output.
  
SYSCTRL-ZCU102-root@[13:05:13]> slaveinfo eth0 -map   
SOEM (Simple Open EtherCAT Master)                    
Slaveinfo                                             
Starting slaveinfo
ec_init on eth0 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:? M:0000009a I:00030924
 Output size: 80bits
 Input size: 80bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 0000009a ID: 00030924 Rev: 00010420
 SM0 A:1800 L: 140 F:00010026 Type:1
 SM1 A:1900 L: 140 F:00010022 Type:2
 SM2 A:1100 L:  10 F:00010064 Type:3
 SM3 A:1180 L:  10 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  10 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000000a Ll:  10 Lsb:0 Leb:7 Ps:1180 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 140 rd: 140 MBX protocols : 0e
 CoE details: 3f FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x607A:0x00 0x20 INTEGER32    Target position
  [0x0004.0] 0x60FE:0x01 0x20 UNSIGNED32   Physical outputs
  [0x0008.0] 0x6040:0x00 0x10 UNSIGNED16   Controlword
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x000A.0] 0x6064:0x00 0x20 INTEGER32    Position actual value
  [0x000E.0] 0x60FD:0x00 0x20 UNSIGNED32   Digital inputs
  [0x0012.0] 0x6041:0x00 0x10 UNSIGNED16   Statusword
End slaveinfo, close socket
End program

I am able to get the configure, and map the controller and get it to EC_OPERATIONAL_STATE

I am using the following sequence to try to enable the output:

uint32_t *data_ptr = (uint32_t *) ec_slave[1].outputs;

data_ptr++;
*data_ptr = 0xFFFFFFFF;

I do not see the output get enabled and when I check the state using the following sequence:

 ec_readstate();
 ec_send_processdata();
 ec_receive_processdata(EC_TIMEOUTRET);

 printf("\n\nEnabling outputs using Pdo state:0x%x\n", ec_slave[1].state);

The state is 0x14 which I think indicates that the state is now Safe-Operational (0x4) and either EC_STATE_ACK or EC_STATE_ERROR has occured.

I have a check for errors

  while(EcatError)
  {
    printf("Error received after enabling output using pdo %s", ec_elist2string());
  }

But I don't see any.

Why would writing to a PDO change the state?

Also when I try to enable the output using sdo using the following code the outputs work as expected

 uint32_t value_to_send = 0xFFFFFFFF;
 uint16_t index = 0x60FE;
 uint8_t subIndex = 1;

  if (ec_SDOwrite(1, index, subIndex, FALSE, sizeof(value_to_send),
                  &value_to_send, EC_TIMEOUTTXM))

I am new to Ethercat and SOEM so I would appreciate any help I can get.

ddrouin17, 2020-11-20:
Here is the ESI file for the device
[Elmo ECAT 00010420 V10.txt](https://github.com/OpenEtherCATsociety/SOEM/files/5573988/Elmo.ECAT.00010420.V10.txt)

ArthurKetels, 2020-11-20:
You are new to EtherCAT and start with a complex DS402 drive that has some quirks? Perhaps not the wisest of choices.
Please use the search function in issues and search for Elmo.
Second, read the tutorial.txt (I know, it is against the rules of cool, but anyway, give it a try).

ddrouin17, 2020-11-20:
Thanks for getting back to me.  
I have commented out ec_configdc() and added ec_slave[1].CoEdetails &= ~ECT_COEDET_SDOCA; based on fixes from prior issues I read about using Elmo controllers but I am still having the same issue.  
[consoleOut.txt](https://github.com/OpenEtherCATsociety/SOEM/files/5576004/consoleOut.txt)

[test_app.txt](https://github.com/OpenEtherCATsociety/SOEM/files/5576005/test_app.txt)

I attached the code I am running and the console output.  I have read the tutorial.txt but will reread.  Thanks for your help.

ArthurKetels, 2020-11-24:
In the tutorial is explained how you make a PDO cycle. Use a separate task to do the ec_send_processdata() and ec_receive_processdata(). Also have a look at the sample files simple_test and even better red_test. 

In order to keep a slave in OP mode you have to send processdata continuously, otherwise the slave watchdog timer kicks in (at around 100ms) and throws the slave back to safe-OP+error as is shown in your console output. You can not simply wait for 10s.

And again, if you would have made a Wireshark capture of your test program it is easy to see where you go wrong.

ddrouin17, 2020-12-02:
Thanks for your help.  I added a new thread to send and receive processdata and am now able to stay in OP mode.  I also got the PDO output to work by sending and SDO write to configure the SM2 PDO assignment as shown in your tutorial.  One question I still have is what triggers receipt of Emergency packets?

ddrouin17, 2020-12-02:
It seems like EcatError is only update after I do an sdoRead.

ArthurKetels, 2020-12-02:
Emergency CoE packets is a CANopen legacy. You can do perfectly fine without. This is why SOEM does handle Emergency packets but does not actively go look for them. Any CoE transfer will fetch pending Emergency packets. So your observation is correct.

ddrouin17, 2020-12-02:
Thanks again.

closed 2020-12-02 by ddrouin17