OpenEtherCATsociety/SOEM #466: Async SDO upload and download

briansoe66, 2020-12-03:
This is a feature request for async sdo download and upload functions, that can be used during OP. ec_SDOwrite and ec_SDOread are blocking, and not intended to be used during OP. 

Or is it thread safe to use ec_SDOwrite and ecSDOread on a separate thread while the cyclic thread is in OP? Do I need to lock anything?

nakarlsson, 2020-12-04:
it is thread safe to use ec_SDOwrite and ecSDOread. 
If you need locks depends on the user data you provide to ec_SDOwrite and ecSDOread, it that needs to be thread safe.
If there is a single thread using the result from ec_SDOwrite and ecSDOread you should be fine.

closed 2020-12-04 by briansoe66

briansoe66, 2020-12-04:
Adding async sdo download and upload to SOEM is not needed. Instead, the user can create a new thread to perform the ec_SDOwrite and ecSDOread, which will prevent it from blocking the cyclical PDO loop. Probably, this thread should be realtime but with priority less than 40, because the cyclical PDO loop created using osal_thread_rt uses priority 40.