OpenEtherCATsociety/SOEM #467: Reading EtherCAT-Specific Diagnostics from SOEM

mictas, 2020-12-07:
Hello,
I'm using a ROS SOEM package in a laptop with Ubuntu 18.04 LTS and ROS Melodic.
Through SOEM I command movements to EtherCAT slaves connected to motors.
During the tests, some EtherCAT communication errors occurred and I found this useful document in the web:

https://www.ethercat.org/download/documents/EtherCAT_Diagnosis_For_Users.pdf

With the aim to solve some communication problems, I would cyclically read the following registers for each connected slave:

- ECT_REG_LLCNT (0x0310)
- ECT_REG_RXERR (0x0300)
- Physical Layer Error Counter (0x0301)
- ALStatusCode (0x0134)

In fact, I want to investigate about the causes of errors with the aim to distinguish if an error is caused by external random disturbances or due to electrical problems.

I tried to read the registers above of each slave from SOEM, but without success.
Is there any chance to read them directly from simple_test.c?
If not, how can I read them?

Waiting for a your response, thank you.
Best regards.

ArthurKetels, 2020-12-07:
> I tried to read the registers above of each slave from SOEM, but without success.
Is there any chance to read them directly from simple_test.c?
If not, how can I read them?

Ehh, can you explain what you tried to do? Preferably with some code?

If you want you can simply do an FPRD starting from 0x300 with the length you prefer (for example 16 bytes). It returns the address space of a slave register with the start and length you specified.  If you want to make it really luxurious you could also map that address space with a spare FMMU. And then add it to the IOmap. It is then part of the process data exchange.

Read up on the use of SM and FMMU and how to use them to get data transferred between slave and master.

mictas, 2020-12-09:
I'm sorry if my last explanation wasn't very clear, but I'm a beginner in these specific items.
From my EtherCAT Master node (based on simple_test.cpp), I want to read cyclically the values contained in the following registers of each slave connected:

ECT_REG_LLCNT (0x0310)
ECT_REG_RXERR (0x0300)
Physical Layer Error Counter (0x0301)

For example, in simple_test.cpp contained in the SOEM package, the value of ALStatusCode is printed as follows (rows 122-123):

printf("Slave %d State=0x%2.2x StatusCode=0x%4.4x : %s\n",
                            i, ec_slave[i].state, ec_slave[i].ALstatuscode, ec_ALstatuscode2string(ec_slave[i].ALstatuscode));

In this manner if an error occurred, it is possible to identify a possible cause reading according to the value of ALstatuscode.

From simple_test.cpp I would read and print the values of the three register listed above, in the same manner in which the value of ALstatuscode are printed.
Is it possible?

Otherwise, I'll try to configure the Ethercat slaves to send these values through TXPDO. 

Thank you very much.

nakarlsson, 2021-02-26:
You're referring to code that we don't have easy access to.

Anyway, you can read any registers as mentioned by @ArthurKetels . where you do it depends on the application. Probably less often than the PDO cycle. Run it from a diagostics task or do it last in the application cycle if you got spare time. Reading registers is done with quite little overhead since it simply reads the EtherCAT slave hardware, no action required by the stack.

Can we close this issue?

mictas, 2021-03-03:
Thank you for your suggestions.
I close this issue.

closed 2021-03-03 by mictas