OpenEtherCATsociety/SOEM #468: Can not reach state OP with no error when connect multiple motors

brutalman933, 2020-12-15:
Dear sir,

I met a problem that puzzled me for a long time and I couldn't find what the problem was. 

Based on the simple_test, I initialized the servo motor. But, when one motor is connected, the probability of entering OP is about 90%; when three motors are connected, the probability of entering OP is reduced to 50%; when 15 motors are connected, the OP cannot be reached at all with no error. I suspect it's the initialization configuration, or the initialization logic, but I haven't been able to locate the exact problem point, and of course I've been stuck here.

slave's ESI file:
[ZeroErr Driver3.1.6.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5694345/ZeroErr.Driver3.1.6.zip)

./slaveinfo -map :
[slaveinfo.txt](https://github.com/OpenEtherCATsociety/SOEM/files/5694183/slaveinfo.txt)

./eepromtool ifname 1 -i:
[eeprom.txt](https://github.com/OpenEtherCATsociety/SOEM/files/5694205/eeprom.txt)

uname -a:

> Linux hexrobot 5.4.19-rt11 #1SMP PREEMPT_RT Thu Dec 10 15:33:22 CST 2020 x86_64 x86_64 x86_64 GNU/Linux

wireshark of ./simpletest ifname:
[simple_test.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5694342/simple_test.zip)

a few minor changes:

>chk = 5000;         //chk = 200
>/* wait for all slaves to reach OP state */
>do
>{
>ec_send_processdata();
>ec_receive_processdata(EC_TIMEOUTRET);
>ec_statecheck(0, EC_STATE_OPERATIONAL, 1000);        //ec_statecheck(0, EC_STATE_OPERATIONAL, 50000);  
>}

Your help is very much appreciated. 

best regards

ArthurKetels, 2020-12-15:
From the ESI file the relevant timing requirements:

           <OpMode>
            <Name>DC</Name>
            <Desc>DC-Synchron</Desc>
            <AssignActivate>#x300</AssignActivate>
            <CycleTimeSync0 Factor="1">0</CycleTimeSync0>
            <ShiftTimeSync0>0</ShiftTimeSync0>
            <CycleTimeSync1 Factor="1">0</CycleTimeSync1>
          </OpMode>
          <OpMode>
            <Name>Synchron</Name>
            <Desc>SM-Synchron</Desc>
            <AssignActivate>#x0</AssignActivate>
            <CycleTimeSync0 Factor="1">0</CycleTimeSync0>
            <ShiftTimeSync0>0</ShiftTimeSync0>
            <CycleTimeSync1 Factor="1">0</CycleTimeSync1>
          </OpMode>
        </Dc>

These slaves need to have the PDO transfer synchronized. There is no free running mode allowed. I would advise to use the DC-Synchron mode and set-up the sync0 and sync1 on each slave and have a robust master timing set-up for the PDO transfer. Look at the red_test example and further guidance can be found here with similar issues users have posted here.

The wireshark trace also points in the same direction. All slaves go reliably in safe-OP but not all go to OP later on. Except for timing problems this is very rare. It looks like some slaves try to sync on the PDO transfers but do not manage to find a good lock.

brutalman933, 2020-12-17:
Thank you very much for your help. According to your advice, I have solved the problem.
best regards

brutalman933, 2021-01-01:
Dear sir,

I meet another  strange question, and it is probably also a matter of timing.

After power on the motors, no matter how long you wait, there are always several motors that cannot enter the OP state and no error is reported, when the program is executed for the first time. But, when the program is executed again, all motors can reach the OP state keeping the motors powered. Everything is normal after the first failure. 

My program is basically the same as red_test, with the addition of sync0 and Sync1 configurations between `ec_config_map(&IOmap)` and `ec_configdc()`. I wonder if Sync0 and Sync1 are configured incorrectly.

> ec_config_map(&IOmap);
...
> for (int slave = 1; slave <= ec_slavecount; slave++)
> {
> 	ec_dcsync01(slave, TRUE, 125 * 1000, 125 * 15 * 1000, 125 * 1000);
> }
...
> ec_configdc();

wireshark 
[op_failed.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5758786/op_failed.zip)
[op_reached.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5758790/op_reached.zip)

Your help is very much appreciated.

best regards

ArthurKetels, 2021-01-01:
Most likely a synchronization issue indeed. You must give both slaves and master time to synchronize before going to OP mode.
There are however also some things in your code you could improve.
1. First do ec_configdc() then ec_dcsync01(). ec_dcsync01() assumes the DC clock system is set-up already by ec_configdc().
2. The parameters of ec_sync01() are strange. From the ESI the cycle time factors are 1 for both sync0 and sync1. So I would assume you want to have 125us for both sync0 and 1. Further an offset of 125us makes no sense, you could just as well put in zero. I would advise to use half the cycle time to start with, so 62us. 

After ec_configdc() I would set dorun = 1; This will start the processdata IO, distribute the clocks over the slaves and synchronizes the master with the reference clock. I would then wait at least 1 second before going to OP. Depending on slave behavior it could be 10s.

When relying on synchronization you always should check the slave sync state to decide on starting operations. I recommend to have a look at the EtherCAT poster by ETG : https://www.ethercat.org/download/documents/EtherCAT_Device_Protocol_Poster.pdf
An important slave register to watch during DC operations is Register System Time Difference (0x092C:0x092F). The smaller the number the better the synchronization.

brutalman933, 2021-01-02:
Thanks very much for your quick reply.

Finally, i solve this problem with your advice and the parameters of `ec_dcsync01(slave, TRUE, 125 * 1000, 0, 0)`. In fact, i want to have 250us for both sync0 and sync 1, and the RT cycle time is 2ms. 

best regards

closed 2021-01-04 by brutalman933