OpenEtherCATsociety/SOEM #472: Fixed PDO mapped io module SyncManager watchdog error when moving to OP state

rhysseq, 2021-01-04:
Hi,

I'm new to ethercat/soem and am having trouble with a fixed PDO mapped I/O module. The module fails to enter OP state stating sync manager watchdog error, I can get it into OP state using twincat on windows but it fails using SOEM on linux.  

**Output from simple_test:**
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on enp1s0 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 12 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Not all slaves reached operational state.
Slave 1 State=0x14 StatusCode=0x001b : Sync manager watchdog

Request init state for all slaves
End simple test, close socket
End program

**Output from slaveinfo:**
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enp1s0 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:EtherCAT IO Module
 Output size: 16bits
 Input size: 80bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000a88 ID: 0a880101 Rev: 00000001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:  10 F:00010064 Type:4
 SM3 A:1400 L:   2 F:00010020 Type:3
 FMMU0 Ls:00000000 Ll:   2 Lsb:0 Leb:7 Ps:1400 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000002 Ll:  10 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 23 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program

wireshark capture from simple_test:
[simpletest.txt](https://github.com/OpenEtherCATsociety/SOEM/files/5763132/simpletest.txt)

From looking at similar issues I guess this is an error in the slave code but struggling to pinpoint whats going wrong and how I can fix it? any help and guidance would be greatly appreciated!

ArthurKetels, 2021-01-04:
Please post the raw wireshark capture (.pcap), the text file is useless.

rhysseq, 2021-01-05:
Sorry I couldnt upload that file type
[simpletest.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5767537/simpletest.zip)

ArthurKetels, 2021-01-05:
There are two things to observe from your Wireshark capture:
1) The wireshark timestamps are off by a factor of 1000. 1ms of real time is printed as 1000ms. What OS on what hardware are you using?
2) The slave somehow waits for a synchronization that is not happening. It could be that the 50ms PDO cycle for pre-OP to OP is too slow for this slave.

` Original code in simple test:

         chk = 200;
         /* wait for all slaves to reach OP state */
         do
         {
            ec_send_processdata();
            ec_receive_processdata(EC_TIMEOUTRET);
            ec_statecheck(0, EC_STATE_OPERATIONAL, 50000);
         }
         while (chk-- && (ec_slave[0].state != EC_STATE_OPERATIONAL));`

` Patched code in simple test:

         chk = 10000;
         /* wait for all slaves to reach OP state */
         do
         {
            ec_send_processdata();
            ec_receive_processdata(EC_TIMEOUTRET);
            ec_statecheck(0, EC_STATE_OPERATIONAL, 1000);
         }
         while (chk-- && (ec_slave[0].state != EC_STATE_OPERATIONAL));`

Please try the above modification. If the slave still goes to safe-OP+error then it needs sync0 activated. You could help by posting the ESI file of the slave.

rhysseq, 2021-01-05:
Hi Arthur thanks for the help,

1. OS is ubuntu

2.  output from simpletestEDIT:

SOEM (Simple Open EtherCAT Master)
Simple test EDIT
Starting simple test
ec_init on enp1s0 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 12 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Not all slaves reached operational state.
Slave 1 State=0x14 StatusCode=0x001b : Sync manager watchdog

Request init state for all slaves
End simple test, close socket
End program
sequor-base@sequorbase:~/soem/SOEM/build/test/linux/simple_test$ sudo ./simple_test enp1s0
SOEM (Simple Open EtherCAT Master)
Simple test EDIT
Starting simple test
ec_init on enp1s0 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 12 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Not all slaves reached operational state.
Slave 1 State=0x14 StatusCode=0x001b : Sync manager watchdog

Request init state for all slaves
End simple test, close socket
End program

**wireshark capture of simpletestEDIT:**
[simpletestEDIT.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5768946/simpletestEDIT.zip)

**ESI/XML file:**
[RTEC1616(1).zip](https://github.com/OpenEtherCATsociety/SOEM/files/5768958/RTEC1616.1.zip)

I had tried changing the timing before but with chk = 900 which still fails op state but with no error:
```
     chk = 900;
     /* wait for all slaves to reach OP state */
     do
     {
        ec_send_processdata();
        ec_receive_processdata(EC_TIMEOUTRET);
        ec_statecheck(0, EC_STATE_OPERATIONAL, 1000);
     }
     while (chk-- && (ec_slave[0].state != EC_STATE_OPERATIONAL));```
```
**output from simpletest chk =900:**
SOEM (Simple Open EtherCAT Master)
Simple test EDIT1
Starting simple test
ec_init on enp1s0 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 12 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Not all slaves reached operational state.
Slave 1 State=0x04 StatusCode=0x0000 : No error

Request init state for all slaves
End simple test, close socket
End program

**wireshark from simpletest chk = 900:**
[simpletestEDIT1.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5769018/simpletestEDIT1.zip)

ArthurKetels, 2021-01-05:
Ahh, thanks for the ESI. This slave needs a synchronized master with ec_sync01() activation. Minimum cycle time is 125us. To get this working I can refer to post #468 similar problem and I guess same solution.

rhysseq, 2021-01-05:
Thanks Arthur,

I implemented the ec_sync01() activation trying different cycle timings with the same result
```
         for (int slave = 1; slave <= ec_slavecount; slave++)
         {
             ec_dcsync01(slave, TRUE, 125000, 125000, 0);
         }
```
**output from simpletest sync01:**
SOEM (Simple Open EtherCAT Master)
Simple test EDITsync01
Starting simple test
ec_init on enp1s0 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 12 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Not all slaves reached operational state.
Slave 1 State=0x14 StatusCode=0x001b : Sync manager watchdog

Request init state for all slaves
End simple test, close socket
End program

wireshark from simpletestsync01:
[simpletestsync01.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5769334/simpletestsync01.zip)

ArthurKetels, 2021-01-05:
Please read post #468 carefully. You can not use the simpletest program but you have to switch to red_test. You have to synchronize the master PDO cycle with the slave sync0 cycle.

rhysseq, 2021-01-06:
Hi Arthur,

I added and played with the fixes and timing in post #468 with no success, then added the patch from post  #224 and it went straight into op mode. I'm still trying to learn the basics with timings and troubleshooting the wireshark capture but does this basically mean the code on the slave device needs to be patched as per #224 ? ive contacted the manufacturers previously and it is based off SSC 5.12 . Thanks for all your help

ArthurKetels, 2021-01-06:
Yes, SSC 5.12 has some bugs. And unfortunately patches have been slow. There is no way for a master to figure out what version of SSC (or even if) is used in  a slave. Therefore a quality master side patch is near to impossible. It is easy to reverse the SM assignment but some slaves really have odd SM assignments on purpose and work fine. 

I am thinking to add a flag in SOEM that users ca set. Something like ec_fix_sm_reversal = true;

There is also some blame at the slave manufacturer, they should better test their devices before putting them on the market. The argument "it works with TwinCAT" does not convey much confidence.

nakarlsson, 2021-02-09:
@rhysseq , can we close this issue?

closed 2021-02-09 by rhysseq