OpenEtherCATsociety/SOEM #473: SDO read succeed at first but failed soon after on Linux

foreversilver, 2021-01-08:
Hi everyone, 

I am using soem to read diffrent slaves on both Windows and Linux, but I found that when soem is running on Linux, I will get SDO read fail soon after all the slaves configured to OP state. Whatever the slave is, the issue will come up. When I run the program on window, every things works right. 
I am using a VMWare to run a virtual machine, the Linux version is:
Linux debian 3.2.0-4-686-pae #1 SMP Debian 3.2.65-1 i686 GNU/Linux

Attachment is the captured data frames from wireshark, which we run slaveinfo with -sdo, but after 0x3712, the SDO read all failed.
[linux_slaveinfo.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5785419/linux_slaveinfo.zip)

Any one encountered this issue before? Thanks a LOT!

foreversilver, 2021-01-08:
I debug and find the SDO read failed because in the function ecx_mbxempty(), SMstat read to be 9 while when SDO read success it should be 0.

ArthurKetels, 2021-01-08:
I looked at the capture. Nothing wrong from the master point of view (except your transmitted packets get echoed by the Linux stack). The slave however hangs in the SDO upload request. I guess a glitch in the slave firmware.

The difference with Windows vs Linux could be explained by the efficiency of Linux. It really bombards the slave with requests. You might try to put in some delay between SDO requests.

nakarlsson, 2021-05-28:
@foreversilver  , can this issue be closed?

foreversilver, 2021-05-28:
yes sure!

发自我的iPhone

closed 2021-05-28 by nakarlsson