OpenEtherCATsociety/SOEM #474: Emergency error when configuring map

LudoJ49, 2021-01-19:
Hi

I'm using Oriental motor controller AZD-KED.

For some obscure reason, I can not leave PRE_OP state.

as soon as I call ec_config_map, the controller goes into error and reports an error:
Time:1611046773.118 EMERGENCY slave:1 error:a000

Datasheet says:

• The unsupported communication cycle in the DC mode is set.
• The object that mapping is not possible is mapped in the PDO mapping.
• The object for TxPDO is mapped in RxPDO. Or the object for RxPDO is mapped in TxPDO.

I don't touch PDO mappings and keep them as default. So I guess it is related to DC?

about DC, xml config files says this:
     <Dc>
          <OpMode>
            <Name>DC Sync0</Name>
            <Desc>DC-Synchronous</Desc>
            <AssignActivate>#x0300</AssignActivate>
	          <CycleTimeSync0 Factor="1">0</CycleTimeSync0>
	          <ShiftTimeSync0>0</ShiftTimeSync0>
          </OpMode>
          <OpMode>
            <Name>Free Run</Name>
            <Desc>Free Run</Desc>
            <AssignActivate>#x0000</AssignActivate>
          </OpMode>
        </Dc>

Is there anything I have to do with this in SOEM ?

Any idea ?

nakarlsson, 2021-01-19:
Post a wireshark , what AlStatusCode do you get when it refuse to go to SAFEOP.

>I don't touch PDO mappings and keep them as default. So I guess it is related to DC?
On the other hand free runs is accaptable as well?

LudoJ49, 2021-01-19:
Don't know how to include a file :-s here is a we transfer link
https://we.tl/t-BDNZdUxSMs

the log starts just before I call this is where the controller goes into error
ec_slave[0].state = EC_STATE_OPERATIONAL;
ec_writestate(0);

i have set ecx_context.manualstatechange= true;

nakarlsson, 2021-01-19:
You can zip the wireshark and drag and drop. 
Add a read state to include the reading of the slave state & al status code

LudoJ49, 2021-01-19:
[wireshark3.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5836266/wireshark3.zip)

this is the complete wire shark trace don't know which packet you really want to see.

Thx for your help

nakarlsson, 2021-01-19:
You never request SAFEOP, try adding 

ec_slave[0].state = EC_STATE_SAFEOPERATIONAL;
ec_writestate(0);

Also, try adding an ec_readstate to recorde any issues going to SAFEOP

LudoJ49, 2021-01-19:
same thing, the error is triggered
just after calling 
`ec_slave[0].state = EC_STATE_SAFE_OP;
        ec_writestate(0);`

nakarlsson, 2021-01-19:
yes, but please add the ec_readstate

LudoJ49, 2021-01-19:
[wireshark4.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5836420/wireshark4.zip)

```
qDebug()<<ec_readstate();
ec_slave[0].state = EC_STATE_SAFE_OP;
ec_writestate(0);
qDebug()<<ec_readstate();
```
state is EC_STATE_PRE_OP in both calls

nakarlsson, 2021-01-19:
yes but no error flag is set, give it some time to "refuse" the state change. in the ESI/XML you might see required timeout's per state transition.

ec_statecheck(0, EC_STATE_xxxxxx, 5 * EC_TIMEOUTSTATE);

LudoJ49, 2021-01-19:
Well if I do this, the LED of the controller that says error still turn red  when I call ec_writeState
```
ec_slave[0].state = EC_STATE_SAFE_OP;
ec_writestate(0);
ec_statecheck(0, EC_STATE_SAFE_OP, 5 * EC_TIMEOUTSTATE);
```

Using the manufacturer tool, (with USB connection to the controller) I also have the same message "EtherCAT state machine" going from "PREOP" to "ERR PREOP" when I call writeState

It does not give more info unfortunatly :-(

I'm not so familliar with Ethercat. Could it be some registers that have to be set before ?
or something related to distributed clocks ?

nakarlsson, 2021-01-20:
Yes, but what is the Error? if you follow the instructions and take a Wireshark.

1. Do the state changes
2. Wait at least  5 * EC_TIMEOUTSTATE and hopefully that is enough for the slave to refuse the state change and present its error.
3. Do an ec_readstate that will read the Error from the status registers.

Also, run the test program slaveinfo. Post the console log of the program output. 
Run slaveinfo -map <- the Arg -map increase the amount of information retrieved.

Since the manufacture tool also have "trouble" entering OP, have you check the manual of the drive for any prerequisites need to be able to run? Any chapters on commissioning?

LudoJ49, 2021-01-22:
Hi, I got a feedback from the manufacturer: DC mode cycle has to be set correctly.

[image]

I tried this, but it was not working either:
```
ec_dcsync0(1, TRUE, 1000000U, 0);
ec_configdc();
```
May be I was doing something wrong ?

nakarlsson, 2021-01-22:
If free run is accapted and you don't activate DC that shouldn't be needed. If it is, you should as the **manufacturer**
If DC is needed you need to set it up, you can check serval issues here to get more details on setting up DC, cherck closed aswell.

LudoJ49, 2021-01-29:
Still having the issue. Tried free run mode as the manufacturer suggested. But still the same error.

Here is what slaveinfo says

```
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
gpio: init OK
******************************************************
Wiznet driver for RPI SOEM - V1.0
Author: Ho Tam  - thanhtam.h[at]gmail.com
******************************************************
Link Check...OK
ec_init on -map succeeded.
wkc = 0
1 slaves found and configured.
Calculated workcounter 3
Not all slaves reached safe operational state.
Slave 1 State=11 StatusCode=  50 : EEPROM no access

Slave:1
 Name:AZD-KED
 Output size: 56bits
 Input size: 56bits
 State: 17
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 000002be ID: 000013e5 Rev: 00000000
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1400 L:   7 F:00010064 Type:3
 SM3 A:1700 L:   7 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   7 Lsb:0 Leb:7 Ps:1400 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000007 Ll:   7 Lsb:0 Leb:7 Ps:1700 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:3 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 2f FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program
```

this triggers error 17 in the controller. No idea what  this is...

nakarlsson, 2021-01-29:
Take a wireshark log of the startup

LudoJ49, 2021-01-29:
Here is the complete wireshark trace of slaveinfo -map
[slaveinfo-map.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5893984/slaveinfo-map.zip)

ArthurKetels, 2021-01-29:
Your wireshark trace is not from the same run as your screenshot. So it is not possible to compare directly.

You use SOEM on rpi with wiznet driver. However this kills the -map request, ads now -map is used as network path name. Effectively you do not request slaveinfo to fetch and display the slave mapping. 

In you latest wireshark trace you do however initate the -map request and all mapping is correctly downloaded.

This sloppy and bit by bit information from you makes it difficult to help you.

Please post the ESI file of the slave, I seem not able download it from the OM website.

The wireshark trace shows an error going from pre-op to safe-op. The AL error = 0x1d "Invalid output configuration". The data in the output SM seem however to be correct. 

Your screenshot shows AL error 50 = "EEPROM no access". This means EEprom acces is not given back to PDI. But the wireshark trace it IS given back to PDI. Confusing.

I would suggest to try soem on some different platform than the rpi, perhaps we have strange packet errors.

LudoJ49, 2021-02-01:
Sorry my mistake I mixed the files. 
I tried SOEM on both windows and Rpi with the same errors

Here is the logs made on Windows

[slaveinfo-map_win32.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5903676/slaveinfo-map_win32.zip)

And here is the ESI file

[orientalmotor_azd-ked_rev0000.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5903683/orientalmotor_azd-ked_rev0000.zip)

ArthurKetels, 2021-02-01:
Thanks for the comprehensive data. At least I can tell you that from this data there is nothing wrong on the master side. Also all registers on the slave are programmed as required. The comparison with the ESI file is 100% match.

I think it is time to contact oriental motors and send them the same data set as in your above post. Their engineers could tell you what is wrong and what else the slave expects to resolve the "Invalid output configuration". I am starting to think their firmware expects something that is not written in the protocol but is coincidental behaviour from other masters (read TwinCAT).

LudoJ49, 2021-02-01:
Well, done that. All what they can say for now is that it is an issue with the master :-(

This is my guess too or a firmware issue...

LudoJ49, 2021-02-02:
I got  further info from the supplier. The stack used in the controller is based on Beckoff original stack. So it should be pretty standard...

Is there any know hack for Beckoff stack with SOEM ?

LudoJ49, 2021-02-02:
I have found something interesting based on another thread

If I do as below  in the setup function I can move to SafeOp state. then I got again an error. But at least, I am going further.

strange that I have to clear the registers and set them as they already are in default.

```
    write_u8(slave, 0x1c12, 0, 0);
    write_u8(slave, 0x1c13, 0, 0);

    write_u16(slave, 0x1c12, 1, 0x1600);
    write_u8 (slave, 0x1c12, 0, 1);

    write_u16(slave, 0x1c13, 1, 0x1A00);
    write_u8 (slave, 0x1c13, 0, 1);
```

ArthurKetels, 2021-02-02:
> I am starting to think their firmware expects something that is not written in the protocol but is coincidental behaviour from other masters (read TwinCAT).

Well, it looks I have been right. Tested with TwinCAT, not with other masters. That is how bugs stay in such slaves. 

If you can get to Safe-OP then most problems are gone. Now read the other threads how you can get a stable process data exchange going. Only then you can request OP mode. Most likely you now do not update the PDO cycle continuously and the slave watchdog timer kicks in.

Again, a wireshark trace will tell what is going on.

LudoJ49, 2021-02-02:
Thanks for your support.  After the little trick above, I finally managed to go to OP state.

No that I can move forward, I have few questions before I close this issue.

If I understood well, the thread will manage to update mapped PDOs into ec_slave[x].outputs and ec_slave[x].inputs.
thus, we can read position of the motor (or whatever is mapped) in real time without calling ec_SDOread or ec_SDOwrite

First question: is there some preconditions such as mutex to access them from somewhere else (specially from another thread) or is this handeled by SOEM ? 

2nd question: Just to make sure I'm not getting the oposite: ec_slave[x].outputs map RxPDOs from the slave and  ec_slave[x].inputs map TxPDOs. Or is the the opposite ?

ArthurKetels, 2021-02-02:
SOEM does not protect the IOmap structure, it is up to the user to find a suitable mode to operate safely. 

One method could be to have all data manipulation and computation in the same thread as the PDO exchange.
An other method is to make helper functions that, with the help of mutexes, transfer data in and out of the PDO thread.
The third method is to use an external library that does the IPC, like zeromq.
Which method is best depends on your application.

To your second question. I know it is confusing. EtherCAT is master-centric, inputs go into the master and outputs flow out of the master into the slaves. CANopen however is slave-centric, inputs go from master to slave and outputs go from slave to master. Therefore a CoE TxPDO goes out of the slave into the master and is called inputs in EtherCAT.

nakarlsson, 2021-02-09:
@LudoJ49 , can we close this issue?

LudoJ49, 2021-02-11:
Yes so far I can move on. I still have lots of work to do but I'm on a good track.

Thank you for your help.

closed 2021-02-11 by LudoJ49