OpenEtherCATsociety/SOEM #476: Leadshine EC882AC driver not working with SOEM

VitalSys, 2021-02-03:
Hi, 

I am having trouble getting the leadshine motor move via SOEM.  All my SDO and PDO settings are correct. The drive does not complain any error.  Drive gets enabled and the rotor gets servo locked.  But the target position data has no effect when I send motion data in the pdo object.  I can see the data at the right spot in the packets.  DC timing looks correct. Control and Status words are correct. 

The weird thing is that if I try to move the motor by TwinCat software, it can easily jog the motor.  I have attached wireshark packet captures of both the SOEM and TwinCAT software.  I will appreciate if the experts here can review the packets and let me know what I am doing wrong as I have already spent weeks trying to diagnose the issue.

Thank you.

[packets.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5919769/packets.zip)

ArthurKetels, 2021-02-03:
Well, I guess if you waited another 200000 days or so you would have found the issue ;-)
You accidentally put the sync0 start time (64bit) **before** the the actual system time (64 bit).  This will cause the first sync0 signal in the slave to be around 200000 days in the future (64bit wrap around).

From the wireshark trace:
line 718 : DC SysTime (0x910): 0x000002e6c267a4c0
line 720 : DC StartTime0 (0x990): 0x00000000c8681d00

Please rinse and retry...

VitalSys, 2021-02-10:
[capture.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5959055/capture.zip)
Thank you for your help. That was a trivial mistake that made through testing but we have fixed it. I tested different configurations over the past few days and have narrowed down the problem to when the DC config occurs.

According to the EtherCAT spec DC sync is supposed to take place in "safe-op" which works fine with other drives except for Omron and LeadShine. For these drives if the DC config happens during "pre-Op" then they work fine which is very odd. Other drives absolutely do not like the DC config to occur during "pre-Op"

I have shared the wireshark capture for both preOp (working) and safeOp (not working). The leadshine does go into OP state and the motor does enable but the motor does not move if any target position data is sent. There are no errors with the leadshine and the status word that it sends out is correct. But for the Omron SOEM gives an error - "dc sync timeout error".

The code between the two captures is the same except for when the DC config happens. I am doing this by changing the manualstatechange member of ecx_contextt to 1 for DC config during safe OP and 0 so we manually change the state to PreOp.

I am not sure if the wireshark capture will provide any clues.

closed 2021-02-11 by VitalSys