OpenEtherCATsociety/SOEM #477: The C++ Standard Library forbids macroizing the keyword "inline"

ilya-vl, 2021-02-05:
When using the library in a c++ project, I get the warning C4005 inline: macro redefinition, and error C1189 The C++ Standard Library forbids macroizing the keyword "inline"
in file  [pcap-stdinc.h](https://github.com/OpenEtherCATsociety/SOEM/blob/d9261e801d109eaf99b94907b6d24bac4a123968/oshw/win32/wpcap/Include/pcap-stdinc.h), in line 69
You may need to fix this.

In my case, the #ifndef directive helped
```
#ifndef __cplusplus
#define inline __inline
#endif
```

additional information:
MSVC 2019 compiler 
Project Tree:
```
Solution
│   CMakeLists.txt
│
├───CMakeProject
│       CMakeLists.txt
│       CMakeProject.cpp
│       CMakeProject.h
│
└───SOEM-master
          CMakeLists.txt
```

The Solution CMakeLists.txt contains the following:
```
add_subdirectory ("CMakeProject")
add_subdirectory ("SOEM-master")
```

the CMakeProject CMakeLists.txt contains the following:
```
include_directories( ../SOEM-master/osal )
include_directories( ../SOEM-master/osal/win32) 
include_directories( ../SOEM-master/oshw/win32)
include_directories( ../SOEM-master/oshw/win32/wpcap/Include)
link_directories( ../SOEM-master/oshw/win32/wpcap/Lib/x64)

add_executable (CMakeProject "CMakeProject.cpp" "CMakeProject.h")
target_link_libraries(CMakeProject soem)
```

Thanks

ArthurKetels, 2021-02-06:
Well, SOEM is a C library. For other languages, compile as a lib and link to it. When necessary ad wrappers to bind to your preferred language.
Second, strictly speaking winpcap is not part of SOEM, it is just included for convenience.  You can replace it by ncap if you like.

nakarlsson, 2021-02-26:
Won fix

closed 2021-02-26 by nakarlsson