OpenEtherCATsociety/SOEM #478: NIC Flags Question

FraserG01, 2021-02-08:
Hello I was just wondering why these flags are required when configuring the network interface:
https://github.com/OpenEtherCATsociety/SOEM/blob/master/oshw/linux/nicdrv.c#L163

ArthurKetels, 2021-02-08:
`ifr.ifr_flags = ifr.ifr_flags | IFF_PROMISC | IFF_BROADCAST;`
It is always a compromise how to set the network driver flags. It depends on the hardware layout and all kinds of network filters/firewalls. 

Promiscuous setting means that all network packets are allowed to pass. Some stacks filter out non IP packets (like EtherCAT). 

Broadcast setting means broadcast packets are allowed to pass. SOEM uses broadcast for destination MAC. This allows one to put a switch in-between and still have communication. However this is only for debugging. In production systems it is not recommended to have switches in-between the master and slaves. Another feature is that you could send EtherCAT over WiFi (totally not real-time).

FraserG01, 2021-02-08:
Thanks the explanation Arthur!

closed 2021-02-08 by FraserG01