OpenEtherCATsociety/SOEM #479: Problem with Encoder init and settings

Zoylamb, 2021-02-08:
Hello,

I'm a beginner on Ethercat, I'm trying to use an encoder with SOEM (only for setting it).
I tried the simple_test and slaveinfo examples but i'm facing some issues.

1/ I can't get the OP mode,
2/ I can't get or some parameters (maybe related to 1/).

Here is the encoder ESI file 
[ecat-2092t.xml.txt](https://github.com/OpenEtherCATsociety/SOEM/files/5943601/ecat-2092t.xml.txt)

Slaveinfo log : 

```
SOEM (Simple Open EtherCAT Master)
Starting slaveinfo
ec_init on \Device\NPF_{9E55C0D6-53DE-4A81-8D81-2D7D76F97137} succeeded.
Slave:1 Name:ECAT-2092T Output size:256bits Input size:160bits State:18 delay:0.1
         Out:6ce28f60,  32 In:6ce28f80,  20
1 slaves found and configured.
Calculated workcounter 3
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  25 : Invalid output mapping

Slave:1
 Name:ECAT-2092T
 Output size: 256bits
 Input size: 160bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00494350 ID: 00209254 Rev: 00010000
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:  32 F:00010064 Type:3
 SM3 A:1400 L:  20 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  32 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000020 Ll:  20 Lsb:0 Leb:7 Ps:1400 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 27 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x0000:0x00 0x01
  [0x0000.1] 0x7000:0x02 0x01 BOOLEAN      Reset index latch value
  [0x0000.2] 0x0000:0x00 0x01
  [0x0000.3] 0x7000:0x04 0x01 BOOLEAN      Set counter
  [0x0000.4] 0x0000:0x00 0x01
  [0x0000.5] 0x0000:0x00 0x01
  [0x0000.6] 0x0000:0x00 0x02
  [0x0001.0] 0x0000:0x00 0x08
  [0x0002.0] 0x7000:0x09 0x20 INTEGER32    Set counter value
  [0x0006.0] 0x7002:0x01 0x01 BOOLEAN      Enable compare
  [0x0006.1] 0x7002:0x02 0x01 BOOLEAN      Set auto increment compare direction
  [0x0006.2] 0x0000:0x00 0x01
  [0x0006.3] 0x0000:0x00 0x01
  [0x0006.4] 0x0000:0x00 0x04
  [0x0007.0] 0x0000:0x00 0x08
  [0x0008.0] 0x7002:0x07 0x20 INTEGER32    Set first compare position
  [0x000C.0] 0x7002:0x08 0x20 INTEGER32    Set auto increment compare value
  [0x0010.0] 0x0000:0x00 0x01
  [0x0010.1] 0x7010:0x02 0x01 BOOLEAN      Reset index latch value
  [0x0010.2] 0x0000:0x00 0x01
  [0x0010.3] 0x7010:0x04 0x01 BOOLEAN      Set counter
  [0x0010.4] 0x0000:0x00 0x01
  [0x0010.5] 0x0000:0x00 0x01
  [0x0010.6] 0x0000:0x00 0x02
  [0x0011.0] 0x0000:0x00 0x08
  [0x0012.0] 0x7010:0x09 0x20 INTEGER32    Set counter value
  [0x0016.0] 0x7012:0x01 0x01 BOOLEAN      Enable compare
  [0x0016.1] 0x7012:0x02 0x01 BOOLEAN      Set auto increment compare direction
  [0x0016.2] 0x0000:0x00 0x01
  [0x0016.3] 0x0000:0x00 0x01
  [0x0016.4] 0x0000:0x00 0x04
  [0x0017.0] 0x0000:0x00 0x08
  [0x0018.0] 0x7012:0x07 0x20 INTEGER32    Set first compare position
  [0x001C.0] 0x7012:0x08 0x20 INTEGER32    Set auto increment compare value
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0020.0] 0x6000:0x01 0x01 BOOLEAN      Set counter done
  [0x0020.1] 0x6000:0x02 0x01 BOOLEAN      Reset index latch value done
  [0x0020.2] 0x6000:0x03 0x01 BOOLEAN      Signal of input A
  [0x0020.3] 0x6000:0x04 0x01 BOOLEAN      Signal of input B
  [0x0020.4] 0x6000:0x05 0x01 BOOLEAN      Signal of input C
  [0x0020.5] 0x6000:0x06 0x01 BOOLEAN      Enable extern latch done
  [0x0020.6] 0x6000:0x07 0x01 BOOLEAN      Reset extern latch value done
  [0x0020.7] 0x6000:0x08 0x01 BOOLEAN      Enable compare done
  [0x0021.0] 0x0000:0x00 0x01
  [0x0021.1] 0x0000:0x00 0x01
  [0x0021.2] 0x0000:0x00 0x01
  [0x0021.3] 0x0000:0x00 0x01
  [0x0021.4] 0x0000:0x00 0x01
  [0x0021.5] 0x0000:0x00 0x01
  [0x0021.6] 0x6000:0x0F 0x01 BOOLEAN      Sync error
  [0x0021.7] 0x6000:0x10 0x01 BOOLEAN      TxPDO Toggle
  [0x0022.0] 0x6000:0x11 0x20 INTEGER32    Counter value
  [0x0026.0] 0x6002:0x01 0x20 INTEGER32    Get next compare position
  [0x002A.0] 0x6010:0x01 0x01 BOOLEAN      Set counter done
  [0x002A.1] 0x6010:0x02 0x01 BOOLEAN      Reset index latch value done
  [0x002A.2] 0x6010:0x03 0x01 BOOLEAN      Signal of input A
  [0x002A.3] 0x6010:0x04 0x01 BOOLEAN      Signal of input B
  [0x002A.4] 0x6010:0x05 0x01 BOOLEAN      Signal of input C
  [0x002A.5] 0x6010:0x06 0x01 BOOLEAN      Enable extern latch done
  [0x002A.6] 0x6010:0x07 0x01 BOOLEAN      Reset extern latch value done
  [0x002A.7] 0x6010:0x08 0x01 BOOLEAN      Enable compare done
  [0x002B.0] 0x0000:0x00 0x01
  [0x002B.1] 0x0000:0x00 0x01
  [0x002B.2] 0x0000:0x00 0x01
  [0x002B.3] 0x0000:0x00 0x01
  [0x002B.4] 0x0000:0x00 0x01
  [0x002B.5] 0x0000:0x00 0x01
  [0x002B.6] 0x6010:0x0F 0x01 BOOLEAN      Sync error
  [0x002B.7] 0x6010:0x10 0x01 BOOLEAN      TxPDO Toggle
  [0x002C.0] 0x6010:0x11 0x20 INTEGER32    Counter value
  [0x0030.0] 0x6012:0x01 0x20 INTEGER32    Get next compare position

```
Beside the OP isn't set, i could set some parameter (ie 0x8000 from 0x00 to 0x0A but can't at 0x8001).
What could be the problem of operation mode or not be able to set/get some parameters.

Thanks for your help.

nakarlsson, 2021-02-09:
Post a wireshark of the startup

Zoylamb, 2021-02-09:
Here is 2 captures, one with autoconfig and on which i change ethercatconfiglist

[Ecat2092T.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5949675/Ecat2092T.zip)

change in ethercatconfiglist : 

` {/*Man=*/0x00494350,/*ID=*/0x00209254,/*Name=*/"ECAT-2092T",/*dtype=*/0,/*Ibits=*/160,/*Obits=*/256,/*SM2a*/0x1100,/*SM2f*/0x00010064,/*SM3a*/0x1400,/*SM3f*/0x00010020,/*FM0ac*/1,/*FM1ac*/1},
      `

nakarlsson, 2021-02-09:
At first glance it looks like everthing match the default settings in OD and ESI., yet it complain abouts the output mappings.
Try doing the mapping explicitly, same as https://github.com/OpenEtherCATsociety/SOEM/issues/474#issuecomment-771508815, that is what TwinCAT would do I guess.

According to the ESI PDOUpload is not included and PdoAssign is  true.

Zoylamb, 2021-02-09:
Thank you for your feedback.

I've used the function as adviced.
Tried to use the information in documentation (0x1C for output and 0x20 for inputs) : 
[image]

but still error in mapping.

I've finally tried and found some values (4 for input and output) which is accepted and make device passed in SAFE_OP.
I don't know if those are the correct values for good behavior. 
I've got still issue when trying to modify settings but this could be device or method related.

ArthurKetels, 2021-02-09:
The problem with your object 0x8001:00 and :01 according to the slave is that they do not exist in the object dictionary. The error message is :` 0x06020000, "The object does not exist in the object directory"`
You ESI file does list this object, BUT the product version of the ESI is 0x00020000, and you slave has 0x0001000. So I guess you have to update your slave firmware. Good chance your other problems are solved too.

Zoylamb, 2021-02-09:
Thank you for this information, i didn't think of checking the FW version.
I'll try to get it the good version from my supplier.
Do the firm_update example could do the FW update operation or it is a simple example for developing slave?

ArthurKetels, 2021-02-09:
I advise to always follow the instructions of the slave manufacturer to update firmware. You could brick the device otherwise. But yes, the firm_update example is generic and should work with all slaves that support FoE firmware update.

Zoylamb, 2021-02-10:
Ok, thank you again for all advices.
Hoping the manufacturer will provide such tools.

You can consider this issue closed.

closed 2021-02-10 by Zoylamb