OpenEtherCATsociety/SOEM #484: Malformed CoE data when writing PDO

VitalSys, 2021-02-22:
Hello,
I have the SOEM stack running with multiple brands like omron, mitsubishi.. But now with one of the Chinese brands I am having trouble writing PDOs in Pre-OP.
Immediately after writing a PDO (does not matter which) I get the error 
`PACKET slave:1 index:1c12.00 error:1`

Never faced this particular issue with any of the other drives. TwinCat is able to run it hence it must be something wrong with what I am doing. Hopefully someone can help me out.

I am attaching the Wireshark trace and the ESI.
[Trace.zip](https://github.com/OpenEtherCATsociety/SOEM/files/6024299/Trace.zip)

ArthurKetels, 2021-02-22:
The only difference between TwinCAT and SOEM in the SDO download is the mailbox counter. TwinCAT starts at 0 and SOEM starts at 1. This counter is arbitrary and a slave should not expect any specific value or order. It is only used to implement the robust mailbox protocol (active retries on lost transmissions). 

The specific code in SOEM (ecatcoe.c):
```
      cnt = ec_nextmbxcnt(context->slavelist[Slave].mbx_cnt);
      context->slavelist[Slave].mbx_cnt = cnt;
      SDOp->MbxHeader.mbxtype = ECT_MBXT_COE + MBX_HDR_SET_CNT(cnt); /* CoE */
```
You could try to add cnt--; after the first line. ec_nextmbxcnt() returns a range of 1..7.

VitalSys, 2021-02-22:
I did notice that too. I tried the fix and it does not work its the same error. I am not even sure on how to begin to debug this issue there seems to be nothing wrong. Maybe something to do with ESC/EEPROM but twincat seems to be able to run it fine.

ArthurKetels, 2021-02-23:
I think I found the problem.
From the TwinCAT trace programming SM1:
```
SyncManager: 0014800022000100
    SM Start: 0x1400
    SM Length: 0x0080
    SM Ctrl/Status: 0x0022, OpMode: 1 buffer, Access: Read, PDI IRQ, 3 buffer state: 1. buffer
    SM Enable: 0x0001, Enable
```
SM1 from SOEM:
```
SyncManager: 0011800022000100
    SM Start: 0x1100
    SM Length: 0x0080
    SM Ctrl/Status: 0x0022, OpMode: 1 buffer, Access: Read, PDI IRQ, 3 buffer state: 1. buffer
    SM Enable: 0x0001, Enable
```
As you can see SOEM gets the wrong SM1 start address from the slave EEPROM. This explains why we would get garbage from a SDO request. 

Your solution is to manually set the slave SM1 address to 0x1400. This is a common problem with inexperienced slave vendors, not enough checks if their implementation is valid. They only test with TwinCAT and that is it. This is no fault of SOEM that just tries to configure the slave according to it's own source (eeprom) instead of an external ESI file.

From the ESI file for SM1:
`<Sm Enable="1" ControlByte="#x22" StartAddress="#x1400" DefaultSize="128" MaxSize="192" MinSize="32">MBoxIn</Sm>`

Best long term solution is of course a bug fix from the slave vendor.

VitalSys, 2021-03-04:
Thank you for the detailed response. This fixed the issue.

Is there a methodology you follow to analyze the Wireshark captures?

closed 2021-03-04 by VitalSys

ArthurKetels, 2021-03-04:
> Is there a methodology you follow to analyze the Wireshark captures?
What about experience from 15 years of looking at EtherCAT network traffic?
There are many layers of "functions" that need to be perfectly stacked in order to get slaves and master communicating together. Depending on the type of fault different layers are known to be the possible cause. 

I give lectures about EtherCAT ranging from slave stack implementation to long term stability management, from 5 fold redundant systems to  sub nanosecond synchronization. And each target has it's own set of rules and optimizations. Given this wide range of familiarity and time it is natural you build up a knowledge-base of solutions to problems.

Your type of problem lays most likely in protocol errors or configuration errors. For a slave mailbox  it's most critical configuration is in the Syncmanagers, so that is where I look for sure. The Wireshark dissector is very helpful for finding protocol errors. When I looked to your Wireshark traces I checked dozens of rules and parameters. It is like reading a book, the story is told in front of your eyes. I just look for irregularities in the story. And to learn the story, a good source is to read the source code of the configuration routines. Next to that the EtherCAT protocol specification.