OpenEtherCATsociety/SOEM #488: How to write octostring via ec_sdowrite?

kunijiwa03, 2021-03-01:
Hello, I am new to EtherCat and SOEM, and I am confused. I encountered a problem when using ec_SDOWrite. I don't know how to write octetstring to a subindex. For example, 0x8006:0x02 of EL6751 requires octetstring(12). Unfortunately, I did not find a corresponding solution.
My data is 00 81 01 00 00 FF 02 00 00 00 00 00, I am trying to deal with it this way:
     uint8 txpdo1[12] = {0x00,0x81, 0x01, 0x00, 0x00 ,0xFF, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00};
     int val = ec_SDOwrite(2, 0x8006, 0x01, FALSE, sizeof(txpdo1), &txpdo1, EC_TIMEOUTRXM);
What should I do?
good luck!

ArthurKetels, 2021-03-01:
Looks OK. What is your response? Anything written in the error list? Please post a Wireshark trace. Sometimes the SDO write is only possible in specific states of the slave (f.e. pre-OP). Did you check the slave manual?

kunijiwa03, 2021-03-01:
yeah, I wrote the SDO in the state of pre-OP just like the manual asked. And I tried to start the setup of EL6751 before  ec_config_map(), in this way the sampletest will end with "Slave 2 State=0x12 StatusCode=0x0025 : Invalid output mapping
". When I started the setup of EL6751 after the ec_configdc(), the sample will be executed but the PDOs is empty, and the can node status is incorrect. The slave address 0f EL6751 is 0x1002
 if (ec_init(ifname))

[wire.zip](https://github.com/OpenEtherCATsociety/SOEM/files/6060686/wire.zip)

   {
      printf("ec_init on %s succeeded.\n",ifname);
      /* find and auto-config slaves */

       if ( ec_config_init(FALSE) > 0 )
      {
         printf("%d slaves found and configured.\n",ec_slavecount);
         //

         ec_config_map(&IOmap);
         ec_configdc();
         EL6751setup();
         printf("Slaves mapped, state to SAFE_OP.\n");
         /* wait for all slaves to reach SAFE_OP state */
         ec_slave[2].state = EC_STATE_SAFE_OP;
         ec_writestate(2);
         ec_statecheck(0, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE * 4);
[wire.zip](https://github.com/OpenEtherCATsociety/SOEM/files/6060689/wire.zip)

kunijiwa03, 2021-03-03:
Hi, Artur, I've tried to configure the EL6751 as you said, and now my question is the simpletest will end with "State=0x12 StatusCode=0x001e : Invalid input configuration".  Here is my configure codes: int EL6751setup(uint16 slave)
{
    uint8 ubuf8;
    uint16 ubuf16;
    uint32 ubuf32;

    ECAT_WRITE(2, 0x1C12, 0x00, ubuf8, 0x00, "disable entry");
    ECAT_WRITE(2, 0x1C13, 0x00, ubuf8, 0x00, "");
    //sync//
    ECAT_WRITE(2, 0x1C32, 0x02,  ubuf32, 0x003D0900, "set to Cycle time");
    ECAT_WRITE(2, 0x1C33, 0x03,  ubuf32, 0x00249F00, "set to Shift time");
    //start the el6751//
    ec_SDOwrite(2, 0xF800, 0x00, TRUE, sizeof(map_xf800), &map_xf800, EC_TIMEOUTRXM);
    //printf(ec_elist2string());

    CanOpenSlave(1);//0x8000,0x8006,0x8008,0x8003
[wire.zip](https://github.com/OpenEtherCATsociety/SOEM/files/6073781/wire.zip)

    //download txpdo//

    ECAT_WRITE(2, 0x1C12, 0x01, ubuf16, 0x1600,"");
    ECAT_WRITE(2, 0x1C12, 0x00, ubuf8, 0x01,"set 1");

    //download rxpdo//

    ECAT_WRITE(2, 0x1C13, 0x01, ubuf16, 0x1A00, "");
    ECAT_WRITE(2, 0x1C13, 0x02, ubuf16, 0x1A82, "");
    ECAT_WRITE(2, 0x1C13, 0x03, ubuf16, 0x1A83, "");
    ECAT_WRITE(2, 0x1C13, 0x04, ubuf16, 0x1A84, "");
    ECAT_WRITE(2, 0x1C13, 0x00, ubuf8, 0x04, "set 4");

    return 0;

       if ( ec_config_init(FALSE) > 0 )
      {
         printf("%d slaves found and configured.\n",ec_slavecount);
         //
         if(ec_slave[2].state != EC_STATE_PRE_OP)
         {
             ec_slave[2].state = EC_STATE_PRE_OP;
             ec_writestate(2);
         }
         printf("slave 2 state= 0x%2.2x\n",ec_slave[2].state);

         ec_slave[2].PO2SOconfig = &EL6751setup;
         ec_config_map(&IOmap);
         ec_configdc();

         ECAT_READ(2,0x1A8A,0x00,ubuf8,"");
         ECAT_READ(2,0x8008,0x00,ubuf8,"");
         printf("Slaves mapped, state to SAFE_OP.\n");
         /* wait for all slaves to reach SAFE_OP state */
         ec_slave[2].state = EC_STATE_SAFE_OP;
         ec_writestate(2);
         ec_statecheck(0, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE * 4);

         slaveinfo();//print slaveinfo
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on ens33 succeeded.
2 slaves found and configured.
slave 2 state= 0x02
Slave: 2 - Read at 0x1a8a:0 => wkc: 1; data: 0x0 (0)	[]
Slave: 2 - Read at 0x8008:0 => wkc: 1; data: 0x4 (4)	[]
Slaves mapped, state to SAFE_OP.

Slave:1
 Name:EK1814
 Output size: 4bits
[... 88 lines cut ...]
  [0x002C.0] 0x0000:0x00 0x10
  [0x002E.0] 0xF108:0x21 0x08 UNSIGNED8    RX error counter
  [0x002F.0] 0xF108:0x22 0x08 UNSIGNED8    TX error counter
  [0x0030.0] 0xF108:0x01 0x01 BOOLEAN      Bus-Off
  [0x0030.1] 0xF108:0x02 0x01 BOOLEAN      warning limit reached
  [0x0030.2] 0xF108:0x03 0x01 BOOLEAN      RX overflow
  [0x0030.3] 0x0000:0x00 0x01
  [0x0030.4] 0xF108:0x05 0x01 BOOLEAN      TX overflow
  [0x0030.5] 0xF108:0x06 0x01 BOOLEAN      Ack error
  [0x0030.6] 0x0000:0x00 0x02
  [0x0031.0] 0x0000:0x00 0x08
  [0x0032.0] 0x0000:0x00 0x10
segments : 1 : 52 0 0 0
Request operational state for all slaves
Calculated workcounter 6
slave 2 state= 0x04
slave 2 state= 0x04
Not all slaves reached operational state.
Slave 2 State=0x12 StatusCode=0x001e : Invalid input configuration

Request init state for all slaves
End simple test, close socket
End program

And I faild to find out the difference between SOEM and Twincat, here is the wireshark.

ArthurKetels, 2021-03-03:
Your Wireshark files are filtered. Please do not do that. To be useful I need all data captured.

kunijiwa03, 2021-03-03:
[wire.zip](https://github.com/OpenEtherCATsociety/SOEM/files/6074648/wire.zip)
Thank you for your reply, it's a complete Wireshark file including SOEM and twincat.

ArthurKetels, 2021-03-03:
There is something wrong with SM3 size calculation. SOEM reads the mapping with CA (complete access). TwinCAT calculates 35 bytes correct and SOEM only 33 bytes. I have to dig deeper why, but for now disable CA.

After ec_config_init() but before ex_config_map() you can do
ec_slave[n].CoEdetails &= ~ECT_COEDET_SDOCA;
to clear the Complete Access bit in the CoE capabilities list for slave n.

nakarlsson, 2021-03-03:
The SM2,3 flags don't look to be correct. All is configured in one buffer mode?

ArthurKetels, 2021-03-03:
Actually slaves often use 1 buffer mode when in DC sync mode. There is no need for a triple buffer, master and slave are in sync after all :-)

kunijiwa03, 2021-03-04:
Hi Arthur, I tried the code as you said, it didn't fix this problem.

ArthurKetels, 2021-03-04:
Can you post a Wireshark trace? Perhaps one problem is solved and now we have another one.

kunijiwa03, 2021-03-04:
Hi Arthur, Inspired by the different sizes of SM3 you said, I fixed this problem. I checked the process Data in twincat and the pdo map in SOEM, found that 0x1A82 and 0x1A83 are not in the map list of SOEM. So I delete the "ECAT_WRITE(2, 0x1C13, 0x02, ubuf16, 0x1A82, "");
ECAT_WRITE(2, 0x1C13, 0x03, ubuf16, 0x1A83, "");".Since it's not key information in my system, I choose to give up.
Thank you a lot!

kunijiwa03, 2021-03-29:
Hi Arthur, recently, I encountered a new problem. After running for a period of time, my program will enter a strange state: using ec_ slave [3]. state reads that the slave status is op, but the slave LED indicates that the slave is in safe-OP. Do you know what happened? Thank you for any improvement!

ArthurKetels, 2021-03-29:
Did you first issue a ec_readstate(3) for the slave? Otherwise the state is not updated in SOEM.

kunijiwa03, 2021-03-30:
Thank you! I've fixed this problem.

closed 2021-03-30 by ArthurKetels