OpenEtherCATsociety/SOEM #489: tx_mutex use in nicdrv.c (and a little more)

mwh-newtec, 2021-03-03:
Hi,

Been using libSOEM for a while now, and stumbled upon some hiccups along the way, where most have been solved before I pulled myself together to patch upstream.

I have three concerns. I'm using linux, so my points will be regarding that platform.

1) Why is tx_mutex only locked on transition on secondary port in redundant mode? ecx_outframe does not lock the mutex.
This could be solved by making the mutex's recursive and locking in ecx_outframe as well.

2) The lock of tx_mutex and rx_mutex does not push cleanup handlers, but the scope of the lock includes cancellation points.

3) If send() in ecx_outframe/ecx_outframe_red fails, the rxbufstat[idx] is set to EC_BUF_EMPTY. This is also a race condition like the ones you have fixed in #275 (SHA ID: aed0f81724e5a2d482491bd6e5e9932a83a42646).  Instead of setting EC_BUF_EMPTY, EC_BUF_ALLOC is more correct. However, I suggest changing the sequence so that if send() >= 0, then set EC_BUF_TX, not before you know if it is sent. The reason for setting it before send(), is that due to the fact that send() is a cancellation point?

Best regards Mathias

nakarlsson, 2021-03-03:
On 3., check issue https://github.com/OpenEtherCATsociety/SOEM/issues/199 for why EC_BUF_TX is before. 
You're right EC_BUF_EMPTY vill give the same race condition fixed in #275

For me that doesn't work to much with linux, what do you mean with thread cancellation and the affects?

ArthurKetels, 2021-03-03:
After looking at it quite a bit I think the best solution is to simply remove
```
   if (rval == -1)
   {
      (*stack->rxbufstat)[idx] = EC_BUF_EMPTY;
   }
```
From ecx_outframe/ecx_outframe_red. There is no need to release the buffer, just assume the buffer has been send, it will get cleaned up later. This case should be rare. It could only happen when the NIC goes down. Then we do not care about efficiency anyway.

mwh-newtec, 2021-03-04:
Hi,

First, can you verify my answer to point 1. is correct?

Thanks for pointing to #199 for an explanation of point 3. I see the point that Arthur mentions, But I'd prefer the state to be reverted to EC_BUF_ALLOC, then following functions can differentiate and not do receiving if we don't expect reception due to the messages not sent.

@nakarlsson it is good practice to guard your locked mutex's with cleanup handlers when the locked scope includes cancellations points.
In this case, send/receive are cancellation points. So without cleanup handlers, if a thread is cancelled while having locked the mutex, the mutex is never released, thus causing the threads waiting for the mutex to wait forever (until they are cancelled).
Cleanup handlers are called upon thread cancellation. So if an cleanup handler was inserted that unlocked the mutex, then in the case described above, the mutex would be unlocked upon thread cancellation and the mutex would be free again.

Regarding point 4, it is basically the same as point 2, however it's the fact that rxbuf[idx] isn't set back to EC_BUF_EMPTY if a thread is cancelled while "owning" idx. Thus, eliminating the use of that idx for future communication. A standard cleanup handler is to simple as it would potentially invoke race conditions alike the ones mentioned here and in #275 (if thread A has sent, gets cancelled, sets rxbuf[idx] = EC_BUF_EMPTY, thread B gets the same idx, sends and receives thread A's answer) . So you'd need something that would cleanup after x time. That, and the fact the natural place to place the cleanup handlers is in ethercatbase.c, however pthreads are Linux and ethercatbase.c is platform independent.

Regards Mathias

nakarlsson, 2021-03-05:
1. is correct

3. For now, I think just removing the EC_BUF_EMPTY is best. The other scenarios is new functionallity and when adding such one should think on how to design that. Going to EC_BUF_ALLOC would not be obvious what have happende and what you can do. And EC_BUF_TX is never a garantuee that we'll receive a frame.

Ok, adding support for cancelation sounds like a work package for the future, not quickly fixed here and now. The need doesn't seem urgent.

mwh-newtec, 2021-03-05:
> For now, I think just removing the EC_BUF_EMPTY is best. The other scenarios is new functionallity and when adding such one should think on how to design that. Going to EC_BUF_ALLOC would not be [...]

The state of an unsent, yet reserved message is EC_BUF_ALLOC. Makes perfect sense to restore EC_BUF_ALLOC upon send failure.