OpenEtherCATsociety/SOEM #490: How to get the SOEM to modify certain output bytes

krishnaintell, 2021-03-09:
`
Hello Everyone/SOEM Team 

I am currently trying to use the SOEM on my laptop to connect to two Ethercat Slaves in my network. One of the Ethercat Slave is a CAN to Ethercat Gateway from HMS and the other slave is an ASi to Ethercat Gateway. 

I have gone through some example programs and tried to make minor changes in the already provided example code  to send my own data on a particular output byte. I am getting the below error(error mentioned at the end) when i try to do a SDO write directly to the Index and the sub index. Because the byte which wanted to modify and change resides in that index and subindex. Below is a snippet of the Output byte that i am trying to control. 

The error that i am getting is already clear. I am trying to write to an area which only has Read access. My question was as to how i can modify or overwrite this output byte using any other method.  I was unable to find other example programs. Please help. 

*A brief description of my setup where i am trying to implement the SOEM.   My setup comprises a Processor which is directly connected via CAN to the first Slave (CAN to Ethercat Gateway). This CAN to Ethercat Gateway is connected to a Ethercat to ASi Gateway.  Both these Gateways are connected to my PC, where i am trying to test the SOEM Stack. 

-------------------------------------

Snippet of the output for CAN to Ethercat Gateway

Slave:1
 Name:Anybus Communicator CAN
 Output size: 8bits
 Input size: 240bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
 Man: 0000001b ID: 0000001d Rev: 0001000c
 SM0 A:1c00 L: 192 F:00010026 Type:1
 SM1 A:1e00 L: 192 F:00010022 Type:2
 SM2 A:1000 L:   1 F:00010064 Type:3
 SM3 A:1600 L:  30 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   1 Lsb:0 Leb:7 Ps:1000 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000004b Ll:  30 Lsb:0 Leb:7 Ps:1600 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 192 rd: 192 MBX protocols : 04
 CoE details: 13 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
/********************************************************************/

A snippet of the SDO info for the Output byte that i am trying to modify and a point wise description to illustrate my understanding 

* I observed that the outputs are mapped via the index 0x1600 ( which i recollect from CANopen standards regarding the TxPDO , RXPDO assignment)

Index: 1c12 Datatype: 0006 Objectcode: 08 Name: Sync Manager RxPDO Assign
  Sub: 00 Datatype: 0005 Bitlength: 0008 Obj.access: 0007 Name: SubIndex 000
          Value :0x01 1
  Sub: 01 Datatype: 0006 Bitlength: 0010 Obj.access: 0007 Name: SubIndex 001
          Value :0x1600 5632

* When i look at the index 0x1600 , it tells me where i could find my output byte( that i was trying to modify)

 Index: 1600 Datatype: 0021 Objectcode: 09 Name: RxPDO
  Sub: 00 Datatype: 0005 Bitlength: 0008 Obj.access: 0007 Name: SubIndex 000
          Value :0x01 1
  Sub: 01 Datatype: 0007 Bitlength: 0020 Obj.access: 0007 Name: SubIndex 001
          Value :0x21000108 553648392

* The index 0x2100 , sub index 01 finally gives me the output byte that i was trying to overwrite
 

 Index: 2100 Datatype: 0005 Objectcode: 08 Name: Outputs
  Sub: 00 Datatype: 0005 Bitlength: 0008 Obj.access: 0007 Name: SubIndex 000
          Value :0x01 1
  Sub: 01 Datatype: 0005 Bitlength: 0008 Obj.access: 0047 Name: Output byte 0000
          Value :0x01 1

/****************************************************************************/

What am I trying to achieve here? 

1. The Output byte translates finally to a CAN message which my system would like to receive.  Further, this output byte has to be modified and communicated to my processor as and when my second Slave (ASi to Ethercat Gateway) records a certain temperature value in its process data image. 
2. I was therefore trying to test if i can modify the output byte via SDO write. 
 
Below is the code that i have written 

/***********************/

int Anybussetup(uint16 slave)
{
    int retval=0;
	uint8_t output_byte = 0xfe;  //  only one byte and giving it a value 0xfe
	
 
  
    retval += ec_SDOwrite(slave, 0x2100, 0x01, FALSE, sizeof(uint8_t), &output_byte, EC_TIMEOUTSAFE); // Trying to modify the variable present in the mentioned index 2100 and sub index 01
    
	while (EcatError) printf(" Error  : %s\n", ec_elist2string());

    printf("Anybus slave %d set, retval = %d\n", slave, retval);
    return 1;
}
/*************************/

A part of the already provided code by SOEM - with minor modifications: 

void simpletest(char* ifname)  // Obtained this from the SOEM Example code 
{
	int i, j, oloop, iloop, wkc_count, chk, slc; //slc stands for slave count ( just a count for a FOR loop)
	UINT mmResult;

	needlf = FALSE;
	inOP = FALSE;

	printf("Starting simple test\n");

	/* initialise SOEM, bind socket to ifname */
	if (ec_init(ifname))
	{
		printf("ec_init on %s succeeded.\n", ifname);
		/* find and auto-config slaves */

		if (ec_config_init(FALSE) > 0)
		{
			printf("%d SLAVES found and CONFIGURED.\n", ec_slavecount);

			if ((ec_slavecount > 1))
			{
				for (slc = 1; slc <= ec_slavecount; slc++)
				{
	
					// Now check if the HMS Gateway is found 
					if ((ec_slave[slc].eep_man == 0x0000001b) && (ec_slave[slc].eep_id == 0x0000001d))
					{
						printf("Found %s at position %d\n", ec_slave[slc].name, slc);
				
						ec_slave[slc].PO2SOconfig = &Anybussetup; // For the settings mentioned in the function int Anybussetup(uint16 slave)for my CAN to Ethercat Slave
					}
					
				}
			}
/*******************************************************************/   

The error that i am receiving 

Starting simple test
ec_init on \Device\NPF_{FF3C8305-7187-4612-AF66-54CFF7E52280} succeeded.
2 SLAVES found and CONFIGURED.
Found Anybus Communicator CAN at position 1
Found AS-i EtherCAT Gateway BWU3858 at position 2
Time:1615288149.509 SDO slave:1 index:2100.01 error:06010002 Attempt to write to a read only object

krishnaintell, 2021-03-09:
Thank you so much for the response. Yes, i have gone through the tutorial.txt . I understand that i need the start address of the output , which i can fetch via the ec_slave[slave].outputs and i found the below example.   

```
#define EL4001_1           2
...
void set_output_int16 (uint16 slave_no, uint8 module_index, int16 value)
{
   uint8 *data_ptr;
   data_ptr = ec_slave[slave_no].outputs;      // I obtain the start address for my outputs(RxPDOs, i presume) and since my slave's LSB reads 0 , i understand it is a byte access. 
   /* Move pointer to correct module index*/
   data_ptr += module_index * 2;                           //  I was unable to find an explanation to what a module_index is and the multiplication by 2. 

   /* Read value byte by byte since all targets can't handle misaligned
    * addresses
    */
   *data_ptr++ = (value >> 0) & 0xFF;  // for the first byte which is read( since the pointer is already pointing to the module_index given , shouldn't this be *data_ptr = (value >> 0) & 0xFF;  ???
   *data_ptr++ = (value >> 8) & 0xFF;  // for the next byte which is read ,  *data_ptr++ = (value>>8) & 0xff 
}
```
set_output_int16(EL4001_1,0,slave_EL4001_1.out1);

Thank you so much for providing me the opportunity to understand the SOEM stack and seeing how the Ethercat Master actually works.

nakarlsson, 2021-04-13:
@krishnaintell can we close this issue?

krishnaintell, 2021-04-13:
@nakarlsson  : Yes , we can close the issue. I don't think i will get any further information to my previous comment. Thank you so much.

closed 2021-04-13 by nakarlsson