OpenEtherCATsociety/SOEM #491: Thread safety of error handling

bkloster-sma, 2021-03-10:
I would like to use functions like `ecx_SDOread` and `ecx_SDOwrite` in a separate thread from the one calling `ecx_sendprocess_data` and `ecx_receive_processdata` in order to not block transmission of process data. While thread-safety is not discussed in the documentation, it is mentioned in other issues (#405, #466) that SOEM is thread-safe with respect to using SDO and PDO functions in separate threads. 

However, all of these functions use `ecx_pusherror`, and both threads would have to use `ecx_poperror` or similar for error handling. Looking at those functions' implementations, I cannot discern how they are thread-safe, especially on systems with a preemptive scheduler.

For example, say we have threads A and B, both calling `ecx_pusherror` at the same time while the error list is empty and `elist->head` is `0`. Thread A is slightly faster and writes its error to position `0`, but is then preempted before incrementing `elist->head`. Then, Thread B writes its own error, also to position `0` (overwriting A's error). Eventually, both threads will then increment `elist->head` to a total of 2. But the error list will contain only B's error at position `0` and garbage at position `1`.

A similar scenario is possible when querying errors with `ecx_poperror`.

Is there some additional synchronization I am overlooking?

nakarlsson, 2021-03-10:
Can you please clarify, does your PDO thread also use ecx_pusherror/ecx_poperror? I can't see that SOEM does that from ecx_sendprocess_data and ecx_receive_processdata.

bkloster-sma, 2021-03-10:
I apologize, I just assumed that `ecx_send_processdata` and `ecx_receive_processdata` also reported errors via the error list in addition to the WKC. But you are absolutely right, they don't. I'll close this issue then, sorry for the trouble.

closed 2021-03-10 by bkloster-sma