OpenEtherCATsociety/SOEM #492: Receiving values from Analog Inputs

MarcoMuellner, 2021-03-21:
Hi everyone!

First of all, SOEM is absolutely banger, thanks for making this. The work i am doing right now revolves around building a system to access various IOs through a pre defined configuration file. The Digital In- and Outputs seem to work nicely, and without large hickups. I do have a couple of issues with the analog inputs though (EL3202, EL3052 and EL3042). Here is the slavemap:

`
lave:10
 Name:EL3202
 Output size: 0bits
 Input size: 64bits
 State: 4
 Delay: 1350[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.0.0.0
 Configured address: 100a
 Man: 00000002 ID: 0c823052 Rev: 00100000
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   0 F:00000064 Type:3
 SM3 A:1180 L:   8 F:00010020 Type:4
 FMMU0 Ls:000000b4 Ll:   8 Lsb:0 Leb:7 Ps:1180 Psb:0 Ty:01 Act:01
 FMMUfunc 0:2 1:3 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 03 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 190[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x00B4.0] 0x6000:0x01 0x01 BOOLEAN      Underrange
  [0x00B4.1] 0x6000:0x02 0x01 BOOLEAN      Overrange
  [0x00B4.2] 0x6000:0x03 0x02 BIT2         Limit 1
  [0x00B4.4] 0x6000:0x05 0x02 BIT2         Limit 2
  [0x00B4.6] 0x6000:0x07 0x01 BOOLEAN      Error
  [0x00B4.7] 0x0000:0x00 0x07
  [0x00B5.6] 0x1800:0x07 0x01 BOOLEAN      TxPDO-State
  [0x00B5.7] 0x1800:0x09 0x01 BOOLEAN      TxPDO-Toggle
  [0x00B6.0] 0x6000:0x11 0x10 INTEGER16    Value
  [0x00B8.0] 0x6010:0x01 0x01 BOOLEAN      Underrange
  [0x00B8.1] 0x6010:0x02 0x01 BOOLEAN      Overrange
  [0x00B8.2] 0x6010:0x03 0x02 BIT2         Limit 1
  [0x00B8.4] 0x6010:0x05 0x02 BIT2         Limit 2
  [0x00B8.6] 0x6010:0x07 0x01 BOOLEAN      Error
  [0x00B8.7] 0x0000:0x00 0x07
  [0x00B9.6] 0x1801:0x07 0x01 BOOLEAN      TxPDO-State
  [0x00B9.7] 0x1801:0x09 0x01 BOOLEAN      TxPDO-Toggle
  [0x00BA.0] 0x6010:0x11 0x10 INTEGER16    Value
`

So, i am trying to access this through a struct, which looks like this:

`
typedef struct PACKED //used for the EL3* terminals (Analog input terminals). See the documentation of the Terminals for more information on the meaning of the individual objs.
{
    bool underRange;
    bool overRange;
    bool limit_1 : 2;
    bool limit_2 : 2;
    bool error;
    bool pdo_state;
    bool pdo_toggle;
    uint16_t value;
} in_EL3__single;

typedef struct PACKED //used for 2 IO EL3* terminals. in_1 corresponds to the first IO, in_2 to the second.
{
    /**First part of the input**/
    in_EL3__single in_1;
    /**Second part of the input**/
    in_EL3__single in_2;
} in_EL3__;
`
which imo seems to reflect the mapping above. I do have some issues with the _empty_ regions in there, but that is not the main issue i am having. If i am trying to read the content of these slaves using the following code 
`
in_EL3__* data_ptr = (in_EL3__*)ec_slave[data.m_Address.m_terminalID].inputs;
in_EL3__single retVal;
if (data.m_Address.m_bit == 0)
retVal = data_ptr->in_1;
else if (data.m_Address.m_bit == 1)
retVal = data_ptr->in_2;
`

The content for the first Input is completely nonsensical
[image]

I am sure that i am missing something in my struct above, but for the live of me i can't seem to figure it out. 

Thanks in advance!

ArthurKetels, 2021-03-21:
Hmm. Bool in C is not what you think. My advice, do not use them. 
https://yarchive.net/comp/linux/bool.html
This is what Linus has to say about them in relation to Linux.

In EtherCAT BOOLEAN is defined as a 1 bit type.
If you want to build a struct on the EtherCAT mapping you might want to look at the first column (addr.b). Your data types need to align on a whole byte. In this case the common underlying type for the bit combination is a uint16_t. 
Your struct will become:
```
 typedef struct PACKED //used for the EL3* terminals (Analog input terminals). See the documentation of the Terminals for more information on the meaning of the individual objs.
{
    uint16_t underRange : 1;
    uint16_t overRange : 1;
    uint16_t limit_1 : 2;
    uint16_t limit_2 : 2;
    uint16_t error : 1;
    uint16_t pad1 : 7;
    uint16_t pdo_state : 1;
    uint16_t pdo_toggle : 1;
    uint16_t value;
} in_EL3__single;

```
Keep in mind this only works for little-endian machines. For big-endian you need to reverse the bit fields.

Your decision to split the slave map in two structs is yours. I would not bother and just create one only. I also would skip the bit field and just have one uint16_t flags. Then use bit masks as needed in your code.

MarcoMuellner, 2021-03-29:
Hu. Learned something today. I guess the lesson is, do not use booleans in these kinds of structs then. Thanks a lot!

closed 2021-03-29 by MarcoMuellner