OpenEtherCATsociety/SOEM #493: SOEM not interacting with LAN9252

reactiondynamicsadam, 2021-03-30:
I'm having an issue using the SOEM simple test tool interfacing with a LAN9252 that is configured with the device emulation flag set (no external MCU). When I run the simple test tool I get the following output

> SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on \Device\NPF_{2FC9FD5B-E187-4422-979A-2BC89451D974} succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 0 0 0 0
Request operational state for all slaves
Calculated workcounter 0
Operational state reached for all slaves.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.

And this continues forever. This is the output of slave_info:
>Slave:1
 Name:Microchip PIC32 Slaves
 Output size: 0bits
 Input size: 0bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: e00004d8 ID: 00009252 Rev: 00000001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00000000 Type:2
 SM2 A:1100 L:   4 F:00010064 Type:3
 SM3 A:1400 L:   4 F:00010020 Type:4
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0e
 CoE details: 03 FoE details: 01 EoE details: 01 SoE details: 11
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program

I'm honestly not sure if this is an SOEM issue or an issue with my EEPROM config, but any help would be greatly appreciated

nakarlsson, 2021-03-30:
Post a wireshark. The printout from slave_info indicate the EEPROM is setup to use an MCU, you got mailboxes and support SDO/SDO Info. Does your EEPROm contain PDO data?

nakarlsson, 2021-05-10:
@reactiondynamicsadam , is this still an issue?

nakarlsson, 2021-05-28:
no activity

closed 2021-05-28 by nakarlsson