OpenEtherCATsociety/SOEM #497: DC master SysTime incorrect calculation

alexkalmuk, 2021-04-06:
https://github.com/OpenEtherCATsociety/SOEM/blob/25e36e44f37e396f359f748978878bcc4c6bc90b/soem/ethercatdc.c#L271-L272

But in osal/linux osal_current_time is implemented using CLOCK_MONOTONIC (it calls osal_gettimeofday which uses monotonic clock):

https://github.com/OpenEtherCATsociety/SOEM/blob/25e36e44f37e396f359f748978878bcc4c6bc90b/osal/linux/osal.c#L30-L36

In Linux https://linux.die.net/man/3/clock_gettime (btw, not only for Linux but for POSIX):

```
CLOCK_MONOTONIC
    Clock that cannot be set and represents monotonic time since some unspecified starting point.
```

It means ethercatdc.c can not rely on Epoch time with current osal_current_time implementation.

nakarlsson, 2021-04-06:
that's true, an improvement would be to do as for win32. MONOTONIC is important and a must for the timers

alexkalmuk, 2021-04-06:
Do you mean this part?

https://github.com/OpenEtherCATsociety/SOEM/blob/25e36e44f37e396f359f748978878bcc4c6bc90b/osal/win32/osal.c#L37-L47

But the difference with Linux here that `GetSystemTimeAsFileTime` returns the time from Windows epoch, while MONOTONIC on Linux returns time from an unspecified moment in the past (e.g. system power on). By the way, after reading this description (https://docs.microsoft.com/en-us/windows/win32/sysinfo/system-time, the last paragraph) I am in doubts that `GetSystemTimeAsFileTime` returns monotonic clocks.

If we want to have a single function on Linux which maintains a monotonic clock and returns nanoseconds from 1 Jan, 2000, then probably we can use `gettimeofday` only once, store the result, and then add this result to MONOTONIC clocks.

nakarlsson, 2021-04-06:
I was refering to win32 having a ”int osal_getrelativetime” that would use the MONOTONIC clock on Linux , that is used for the timers. 

Get current_time/timeofday can be clock  REALTIME that doesn’t need to be strictly increasing.

alexkalmuk, 2021-04-06:
Ah, I see.

closed 2021-04-08 by nakarlsson