OpenEtherCATsociety/SOEM #500: SDO communication not working after cycling the slave state

fennel-labs, 2021-04-08:
Hi everyone,
I am trying to connect a Synapticon Somanet drive with SOEM. So far I was able to spin up the motor and read some sensor values via PDO. 

However, there is one problem: When I connect the motor to power (cold boot) and then run my program for the first time, everything performs well. The program simply configures PDOs and reads some SDO values periodically (for debug purposes, I'm aware that this is not real-time). After 5 seconds, the program terminates and right before that happens, the slave is set to PREOP state. When I restart the program then, the SDO communication fails entirely with an undefined error (ec_elist2string() does not output anything).

I went through the code, and for me it seems like a mailbox cannot be allocated. Furthermore, I was able to determine, that the issues vanishes if I do not read an 8 byte SDO (a string to be more precise) periodically.

I've attached a Wireshark capture of the scenario described above. The code was not altered between the two runs. The second run starts at message #12178.

Thanks for your help in advance.
[issue_github.pcapng.zip](https://github.com/OpenEtherCATsociety/SOEM/files/6279651/issue_github.pcapng.zip)

fennel-labs, 2021-04-08:
So, if I remove the periodical SDO queries it seems to work properly. However, this is not intended behavior in my opinion.

ArthurKetels, 2021-04-08:
From the wireshark capture it looks like a stuck mailbox handler in the slave. That is a bug in the slave firmware. Contact the manufacturer and ask for a fix.

fennel-labs, 2021-04-08:
Many thanks for the quick response!

I will contact the drive manufacturer tomorrow. Unfortunately, I guess that this will take some time because they purchased the ESC as well.

In the meantime I am thinking about resetting the slave. Is there any option for that using the Slave registers? For me it seems like there is, since TwinCAT is able to recover the slave when it gets stuck. 

Additionally, do you have any idea, why the 4 byte string SDO exchange is causing the malfunction? What also bothers me here is that TwinCAT is not having those lockup issues, even when the checkbox for periodic SDO updates is set.

ArthurKetels, 2021-04-08:
> What also bothers me here is that TwinCAT is not having those lockup issues
Well, that makes sense. Most firmware developers only check their code against TwinCAT. If that passes they think their work is done. Don't bother to make code that just works. 

> In the meantime I am thinking about resetting the slave
That is what SOEM does when setting the slave back to init and clearing the SM and FMMU registers. That is the official way to reset a slave.

nakarlsson, 2021-05-28:
@fennel-labs , can this issue be closed?

fennel-labs, 2021-05-28:
@nakarlsson The vendor has forwarded this issue including the Wireshark capture to its supplier. The ticket there is still work in progress.

nakarlsson, 2021-05-28:
Ok, feel free to close this and re-open Or close it when you can verify the new slave software

nakarlsson, 2021-11-09:
close due to inactivity

closed 2021-11-09 by nakarlsson