OpenEtherCATsociety/SOEM #503: Regarding the compatibility of the SOEM and SOES

sinamr, 2021-04-22:
Hi Everyone, 

I have a naïve question, but I couldn't find the answer in the documentations.
Are these packages compatible with any PCI card? or is there a list of the  manufactures/products which aremost compatible with these two packages?

Thanks a lot and Best Regards, 
Sina

ArthurKetels, 2021-04-22:
SOEM is an EtherCAT master library. SOES is an EtherCAT slave library. The master runs normally on commodity PC's, the slave on special designed embedded hardware. 

SOES has support for a few different slave controllers (Beckhoff and those using the Beckhoff IP, and Microchip). For other slave hardware you have to write your own driver. 

SOEM has support for Linux, Windows and some other real time OS. For Linux and Windows all NIC drivers are supported as long they can do 100Mb/s. For bear metal systems you have to build your own driver.

sinamr, 2021-04-22:
Thanks a lot for the answer. I think my understanding is wrong then. 
We have a robotic arm which has both EtherCat master and slave cards. We want to communicate to this robot via EtherCat and through a linux machine; send commands to the robot and receives sensory information.   

We thought by using SOEM, one can send commands from the Linux machine via  a PCI card to the EtherCat slave on the  robot. Then the robot, in the same cycle, sends back the sensory information to the same PCI card on the Linux machine. Would this make sense?

and if yes, would if it be realizable by using any NIC PCI over 100Mb/s on our linux machine?

ArthurKetels, 2021-04-22:
Simple answer, yes. And it does not need to be a PCI NIC. Any NIC will do, even an USB NIC works fine.

sinamr, 2021-04-30:
Thx  a lot for answers.

closed 2021-04-30 by sinamr