OpenEtherCATsociety/SOEM #504: SAMD51-LAN9252 can't reach PREOP state

Nicolas-Cetec, 2021-04-23:
Hello everyone,

I know this problem has been already discussed on other places but the main solution which is often brought doesn't work for me. 

So to explain : I am using SAMD51-LAN9252 as EtherCAT slave and TwinCAT on my PC as master. 
I follow the tutorial from github : Creating your first EtherCAT Application : https://github.com/Microchip-MPLAB-Harmony/ethercat/wiki/create-your-first-ethercat-application

So I run the program on MPLAB IDE and then I switch on TWINCAT. But when I create a new project and I scan I/O, there are 2 error messages on my console which are : 
'INIT to PREOP' timeout (3000 ms) reached! Error : set device state to INIT	
state change aborted (requested 'PREOP', back to 'INIT')

And I have tried to change the state of my slave by clicking on I/O / device 2 / online / right click and select "request PREOP state" but it doesn't work
And if I click on I/O / device2 / box 1 / InfoData / State / Online, the value is 65 and not 1 to show that the slave is ready to be used. 

I saw on several forums that coeappl.c and coeappl.h have to be changed with these lines : 
TOBJ1C00 sSyncmanagertype = {0x04, {__builtin_bswap16(0x0102), __builtin_bswap16(0x0304)}}; 
instead of
TOBJ1C00 sSyncmanagertype = {0x04, {0x0102, 0x0304}};

But it changes nothing and I don't know how to fix my issue. 

Does someone have an idea to solve this problem ? I have noticed several people are also in trouble with LAN9252 and SSC Tool 5.12 

Many thanks in advance for your help.

Best regards,

Nicolas

nakarlsson, 2021-04-23:
Is the EEPROM generated from that source? If so, you'll have to make sure to re-program the EEPROM as well.

Nicolas-Cetec, 2021-04-23:
Thank you for you reply.

Sorry but I am not very sure that I understand exactly what you mean.
I have generated a program from SSC and run the program from MPLAB IDE on my EtherCAT slave (LAN9252). I am supposed to scan to find my Master and Slave on TWINCAT and then I am supposed to update EEPROM. But I can't update EEPROM because scan doesn't work.

Nicolas-Cetec, 2021-04-23:
I have noticed that frames are sent from master and received in the slave board (light blinks on the connector Ethernet) and TwinCAT shows the number of the frame sent (screen 1 attached) .
[image]

If I stop sending frame, this LED doesn't blink anymore.

And as you can see on the same screen, the state of the slave is "INIT_ERR".

Another thing : on the screen 2 if I click on Pre-op for example to change the slave's state, the value of state 65 will be changed into 1 for 5s but as this state is unachievable the value of state get back to 65. 
[image]

Sorry for french words on my screen. 

Best regards,

Nicolas

nakarlsson, 2021-04-26:
The fix you're talking about fixes an issue of a "functional" slave. This slave doesn't seem to be functional at all. 
I'm not sure this is the forum for this issue, you need to go back a couple of steps and make sure you've followed the instructions given by Microchip.

Nicolas-Cetec, 2021-04-26:
Hi nakarlsson,
Thank you for your reply.
The fact is, if I run my program on MAPLAB IDE it works so it means that my PC recognizes my slave. But when I switch on Twincat, these mistakes comes when I scan. 
So in my opinion, my slave is ready but connection doesn't work. 
Someone has just a little idea of what is going on ?
Thank you everyone.
Best regards,
Nicolas

nakarlsson, 2021-04-26:
run the slaveinfo application and do a wireshark on that startup.

>The fact is, if I run my program on MAPLAB IDE it works so it means that my PC recognizes my slave
I wouldn't say it is a guarantee it is a functional EtherCAT slave

nakarlsson, 2021-05-10:
@Nicolas-Cetec , is this still an issue?

closed 2021-05-11 by Nicolas-Cetec

wkDriver, 2022-11-07:
> Hello Nakarlsson, Sorry. This issue is resolved thank you, it was a software issue from MPLAB. Many thanks for your help. I close this file. Best regards, Nicolas

May I ask what is the cause of this?  I'm having the exact same problem you're having

Rodolfo270281, 2023-03-03:
Hello Nicolas I have thje same issue than you, as you mnetiin it was a ISSUE of mplab, can you pleaese let me know what isssue is that due I have same, Problem thanks in advance

kanakathalli, 2023-11-09:
Hi Nicolas-Cetec, we are facing same problem. Can you please share how you resolved your issue
Thank you

kanakathalli, 2023-12-06:
> Hello Nakarlsson, Sorry. This issue is resolved thank you, it was a software issue from MPLAB. Many thanks for your help. I close this file. Best regards, Nicolas

We are also facing the same issue. Can you please tell what is the issue from MPLAB that after fixing it worked for you
Thank you

maheemouriya, 2024-09-23:
LAN9252_SAMD51 does not go into the OP mode. 

Any workarounds to resolve this error?

[image]

Aviatron, 2025-07-25:
What was the problem?