OpenEtherCATsociety/SOEM #507: EEPROM data

LudoJ49, 2021-05-04:
Hi,

Not really an issue. But a question. How is the EEPROM organized ? I mean the data rad and written by eepromtool.

The only info I could find is this : https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_io_intro/1358008331.html&id=5054579963582410224

But it is not very detailed.

Is there any datasheet that explains it in more details ? Or public tools to generate the files ?

Thx

mheden, 2021-05-04:
See chapter 5.4 of ETG.1000.6 (Application Layer protocol specification)

LudoJ49, 2021-05-04:
do you have a link to dowload it ?

mheden, 2021-05-04:
Here you go: https://www.ethercat.org/en/downloads/downloads_A02E436C7A97479F9261FDFA8A6D71E5.htm

You need to be an ETG member to access them.

LudoJ49, 2021-05-04:
Well, I'm not :-s

LudoJ49, 2021-09-17:
Ok, we're finally ETG member :) Thanks for your support. I just need to read the datasheet now :)

closed 2021-09-17 by LudoJ49