OpenEtherCATsociety/SOEM #508: Slave Address

gftrobots, 2021-05-06:
Currently we are identifying slaves by their position in the ethercat communication ring (e.g. slave 1,2,3,n).
However, this is very impractical, if the slave position is changing (e.g. by inserting a new slave).

How can we identify from a slave which physical node it represents (e.g. Axis 10) independent of its position in communication ring? The eep_man and eep_id are not sufficient as they will tell us only the device type but not the device id.

We are using Maxon EPOS4 Slaves that have a DIP Switch. However we are not able to read the Switch via SOEM.

Please be so kind and advise.

ArthurKetels, 2021-05-06:
There are two options to identify a slave in a unique way:
a) Read the slave serial number stored in eeprom.
b) Read the Alias Address at register location 0x0012.

Caveats.
a) It is rare for slaves to have the serial number set to something else than 0. You might try to overwrite the eeprom data with your own serial number.
b) The Alias Address can be used in stead of the Configured Address (by SOEM). The other option is to read the Alias Address and compare with a known table in your application. When changing the dip switch, and rebooting, the address should change accordingly. The big unknown is off course if there are duplicates set or someone changes the dip switch unintentionally.

gftrobots, 2021-05-07:
Thanks for the quick response. We are using the maxon EPOS4 controller and the alias address is always zero. Maxon does not provide documentation, if this address can be set.
But good news is, that the serial number is properly configured. Is there an easy why to read the serial number, as it is not presented in ec_slave? The only way we found is eeprom_read function implemented in the eepromtool.

ArthurKetels, 2021-05-07:
To read the slave serial from eeprom you can use:
`uint32 ec_readeeprom(uint16 slave, uint16 eeproma, int timeout);`
For the address of the serial number use eeproma = 0x000e;
Eeprom is slow, so use enough timeout.
After reading eeprom data from a slave do not forget to give eeprom control back to the slave with:
`int ec_eeprom2pdi(uint16 slave);`

gftrobots, 2021-05-08:
This works already good, I get the lower 32 bit of the 64 bit serial number.
However i do not find the higher 32bit of the serial number.
eeproma = (0x000e)+2  does not do the job

ArthurKetels, 2021-05-08:
Serial number in eeprom is only 32bit.

The ID section in eeprom has:
```
uint32_t manufacturer
uint32_t product id
uint32_t version (split in high and low part)
uint32_t serial number
```
Perhaps Maxon uses the version + serial or product id + serial as hardware serial number.

closed 2021-05-11 by gftrobots