OpenEtherCATsociety/SOEM #510: how to change the servo mode from CSP to HM

ilovewall, 2021-05-13:
dear all,if the servo had worked in CSP mode first,when the move had finished,i  wanted to changed HM mode .Then, I modified the 0x6098 to 35,and continued changed the 0x6060 from 8 to 6,the servo would had an error. but if I disabled the servo power,and changed the ox6060 to 6,it would be OK,and when home target reached ,i  moditied the 0x6060 to 8,and enable the servo power,it could work well in CSP mode,so whether disable the servo power when changed the CSP mode to HM ,i hope anyone can give me some suggestions,thank you!

ArthurKetels, 2021-05-13:
Although I understand asking the question, it is not the right place to do that. This place is for asking questions about SOEM, not about how to configure/control my servo drive. The forum (if it exists) of the drive manufacturer would be more appropriate.

You also do not give a lot of information about the implementation details. It is hard to give a proper answer. Cyclic synchronous position mode on drives in general often come with limitations. So not being able to transition to homing mode is not strange. Perhaps there are specific constraints to be met. Again, ask the vendor, they know best.

ilovewall, 2021-05-14:
thank you for your suggestion,I doubted may be something wrong with the configuration in SOEM originally,I donot know whether you do the same test before,and I will try to contract the vendor.

closed 2021-06-09 by ilovewall