OpenEtherCATsociety/SOEM #511: SDO read/write operation error when connect multiple motion slaves

Hikipedia, 2021-05-13:
Hello, sir. I have a problem about SDO read/write in the case of connecting multiple motion slaves. It is posted as follow:
  when i use a single motion slave, i can use SDO read and write operation normally, but when i connect two same motion slaves, the master can't capture the respone came from the second slave.

 
/* Init Master */
`ec_init(ifname);`
`ec_config_init(FALSE);`
……
……
/* Read SDO */
`ec_SDOread(1, 0x1600, 0x01, FALSE, &buf_size, &buf_32, EC_TIMEOUTRET)`;
`ec_SDOread(2, 0x1600, 0x01, FALSE, &buf_size, &buf_32, EC_TIMEOUTRET)`;
……

then the result is

[image]

the frame captured by wireshark is below

[wireshark_data.zip](https://github.com/OpenEtherCATsociety/SOEM/files/6472365/wireshark_data.zip)

any one can help me? thany you.

ArthurKetels, 2021-05-13:
Well, 2 issues,
a) You have to check if all slaves have transitioned to pre-op before doing any SDO communication.
b) Your time-out values are much too low for SDO requests. Use at least 100ms.

Hikipedia, 2021-05-14:
Thank you for you reply, but i found that the slave can't enter PRE_OP after `ec_init()` and `ec_config_init()`.  And when I use `ec_config_init`, all slave can enter PRE_OP,  but the second slave still have no respone by setting 1s timeout. 

Beside, i found that when i replace the first slave by other same motion slave, they run normally. is something woring in first slave? (the first slave can run normally in single).

nakarlsson, 2022-01-14:
@Hikipedia , can this issue be closed?

nakarlsson, 2022-11-17:
Close due to inactivity

closed 2022-11-17 by nakarlsson