OpenEtherCATsociety/SOEM #512: CMake Error

osporui, 2021-05-16:
Hi everyone, 
I am trying to install the SOEM library in Ubuntu 16.04 in a Virtual Machine, but when I do the "cmake .." command, i am getting this error:

> CMake Error: Could not find cmake module file: CMakeDetermineDESCRIPTIONCompiler.cmake
> CMake Error: Error required internal CMake variable not set, cmake may be not be built correctly.
> Missing variable is:
> CMAKE_DESCRIPTION_COMPILER_ENV_VAR
> CMake Error: Error required internal CMake variable not set, cmake may be not be built correctly.
> Missing variable is:
> CMAKE_DESCRIPTION_COMPILER
> CMake Error: Could not find cmake module file: /home/osporui/SOEM-master/build/CMakeFiles/3.5.1/CMakeDESCRIPTIONCompiler.cmake
> CMake Error: Could not find cmake module file: CMakeDetermineSimple Open EtherCAT MasterCompiler.cmake
> CMake Error: Error required internal CMake variable not set, cmake may be not be built correctly.
> Missing variable is:
> CMAKE_Simple Open EtherCAT Master_COMPILER_ENV_VAR
> CMake Error: Error required internal CMake variable not set, cmake may be not be built correctly.
> Missing variable is:
> CMAKE_Simple Open EtherCAT Master_COMPILER
> CMake Error: Could not find cmake module file: /home/osporui/SOEM-master/build/CMakeFiles/3.5.1/CMakeSimple Open EtherCAT MasterCompiler.cmake
> -- The C compiler identification is GNU 5.5.0
> CMake Error at CMakeLists.txt:4 (project):
>   No CMAKE_DESCRIPTION_COMPILER could be found.
> 
>   Tell CMake where to find the compiler by setting the CMake cache entry
>   CMAKE_DESCRIPTION_COMPILER to the full path to the compiler, or to the
>   compiler name if it is in the PATH.
> 
> 
> CMake Error: Could not find cmake module file: CMakeDESCRIPTIONInformation.cmake
> CMake Error at CMakeLists.txt:4 (project):
>   No CMAKE_Simple Open EtherCAT Master_COMPILER could be found.
> 
>   Tell CMake where to find the compiler by setting the CMake cache entry
>   CMAKE_Simple Open EtherCAT Master_COMPILER to the full path to the
>   compiler, or to the compiler name if it is in the PATH.
> 
> 
> CMake Error: Could not find cmake module file: CMakeSimple Open EtherCAT MasterInformation.cmake
> -- Check for working C compiler: /usr/bin/cc
> -- Check for working C compiler: /usr/bin/cc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> CMake Error: CMAKE_DESCRIPTION_COMPILER not set, after EnableLanguage
> CMake Error: CMAKE_Simple Open EtherCAT Master_COMPILER not set, after EnableLanguage
> -- Configuring incomplete, errors occurred!
> See also "/home/osporui/SOEM-master/build/CMakeFiles/CMakeOutput.log".

Have someone experienced this issue or which could be the problem?

nakarlsson, 2021-05-17:
I kind of looks like there is a CMake CMake issue, not a SOEM CMake issue?

ArthurKetels, 2021-05-17:
Perhaps the "build-essential" package is not installed?