OpenEtherCATsociety/SOEM #513: Issues compiling SOEM with MinGW on Windows 10

MarcoMuellner, 2021-05-17:
Hi everyone!

I need to compile SOEM (through the desired usecase) with MinGW, but i stumbled upon a couple of issues along the way. 

This my cmake call:

```
cmake .. -G "MinGW Makefiles"
```
```
-- The C compiler identification is GNU 5.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Qt/Qt5.9.8/Tools/mingw530_32/bin/gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
OS is win32
LIB_DIR: lib
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/Marco/Documents/code/psystem-ethercat/ethercat/soem/build
```

which seems fine at a first glance. However, when building i get the following error (this already includes me changing /D _CRT_SECURE_WARNINGS to -D _CRT_SECURE_WARNINGS:

```
mingw32-make
[  5%] Building C object CMakeFiles/soem.dir/soem/ethercatbase.c.obj
gcc.exe: error: unrecognized command line option '-WX'
CMakeFiles\soem.dir\build.make:75: recipe for target 'CMakeFiles/soem.dir/soem/ethercatbase.c.obj' failed
mingw32-make[2]: *** [CMakeFiles/soem.dir/soem/ethercatbase.c.obj] Error 1
CMakeFiles\Makefile2:134: recipe for target 'CMakeFiles/soem.dir/all' failed
mingw32-make[1]: *** [CMakeFiles/soem.dir/all] Error 2
Makefile:134: recipe for target 'all' failed
mingw32-make: *** [all] Error 2
```
Afterwards i removed the WX flag, which lead to the following errors:
```
mingw32-make
[  5%] Building C object CMakeFiles/soem.dir/soem/ethercatbase.c.obj
ethercatbase.c:1:0: warning: -fPIC ignored for target (all code is position independent)
 /*
 ^
In file included from soem/osal/osal.h:15:0,
                 from ethercattype.h:26,
                 from soem/oshw/win32/oshw.h:19,
                 from ethercatbase.c:19:
soem/osal/win32/stdint.h:33:2: error: #error "Use this header only with Microsoft Visual C++ compilers!"
 #error "Use this header only with Microsoft Visual C++ compilers!"
  ^
In file included from soem/osal/osal.h:14:0,
                 from ethercattype.h:26,
                 from soem/oshw/win32/oshw.h:19,
                 from ethercatbase.c:19:
soem/osal/win32/osal_defs.h:24:31: error: unknown type name 'pack'
 #define PACKED_BEGIN __pragma(pack(push, 1))
                               ^
ethercattype.h:90:1: note: in expansion of macro 'PACKED_BEGIN'
 PACKED_BEGIN
 ^
In file included from soem/oshw/win32/oshw.h:19:0,
                 from ethercatbase.c:19:
ethercattype.h:99:3: warning: data definition has no type or storage class
 } ec_etherheadert;
   ^
ethercattype.h:99:3: warning: type defaults to 'int' in declaration of 'ec_etherheadert' [-Wimplicit-int]
In file included from soem/osal/osal.h:14:0,
                 from ethercattype.h:26,
                 from soem/oshw/win32/oshw.h:19,
                 from ethercatbase.c:19:
[... 329 lines cut ...]
                         ^
ethercatbase.c: In function 'ecx_adddatagram':
ethercatbase.c:108:13: error: 'datagramP' undeclared (first use in this function)
    ec_comt *datagramP;
             ^
ethercatbase.c:115:25: error: expected expression before ')' token
    datagramP = (ec_comt*)&frameP[ETH_HEADERSIZE];
                         ^
ethercatbase.c:121:25: error: expected expression before ')' token
    datagramP = (ec_comt*)&frameP[prevlength - EC_ELENGTHSIZE];
                         ^
CMakeFiles\soem.dir\build.make:75: recipe for target 'CMakeFiles/soem.dir/soem/ethercatbase.c.obj' failed
mingw32-make[2]: *** [CMakeFiles/soem.dir/soem/ethercatbase.c.obj] Error 1
CMakeFiles\Makefile2:134: recipe for target 'CMakeFiles/soem.dir/all' failed
mingw32-make[1]: *** [CMakeFiles/soem.dir/all] Error 2
Makefile:134: recipe for target 'all' failed
mingw32-make: *** [all] Error 2
```

I find this one especially odd:
```
soem/osal/win32/stdint.h:33:2: error: #error "Use this header only with Microsoft Visual C++ compilers!"
```

Can anyone help me with this? Do you have any experience with building the code with MinGW on Windows?

ArthurKetels, 2021-05-18:
MinGW is an odd toolchain. It behaves like Linux, but targets Windows. SOEM does not have support for MinGW.
I guess you could make it work. But take the Linux target as basis and only modify for Windows libs where necessary. SOEM has a very good OS abstraction layer, but it gets confused by MinGW. Neither Linux nor Windows target fits.

MarcoMuellner, 2021-05-21:
I don't think the ethercat code itself is the issue, but rather the abstraction through osal and oshw. These are, as far as i am gathering, platform specific and hence cause the error above. So when trying to just use the sources for linux on windows, i get the following error:

```
C:\SOEM-master\build>mingw32-make
[  4%] Building C object CMakeFiles/soem.dir/soem/ethercatbase.c.obj
[  9%] Building C object CMakeFiles/soem.dir/soem/ethercatcoe.c.obj
[ 14%] Building C object CMakeFiles/soem.dir/soem/ethercatconfig.c.obj
[ 19%] Building C object CMakeFiles/soem.dir/soem/ethercatdc.c.obj
[ 23%] Building C object CMakeFiles/soem.dir/soem/ethercateoe.c.obj
[ 28%] Building C object CMakeFiles/soem.dir/soem/ethercatfoe.c.obj
[ 33%] Building C object CMakeFiles/soem.dir/soem/ethercatmain.c.obj
[ 38%] Building C object CMakeFiles/soem.dir/soem/ethercatprint.c.obj
[ 42%] Building C object CMakeFiles/soem.dir/soem/ethercatsoe.c.obj
[ 47%] Building C object CMakeFiles/soem.dir/osal/linux/osal.c.obj
C:\SOEM-master\osal\linux\osal.c: In function 'osal_timer_start':
C:\SOEM-master\osal\linux\osal.c:72:4: error: implicit declaration of function 'timeradd' [-Werror=implicit-function-declaration]
    timeradd(&start_time, &timeout, &stop_time);
    ^
cc1.exe: all warnings being treated as errors
CMakeFiles\soem.dir\build.make:210: recipe for target 'CMakeFiles/soem.dir/osal/linux/osal.c.obj' failed
mingw32-make[2]: *** [CMakeFiles/soem.dir/osal/linux/osal.c.obj] Error 1
CMakeFiles\Makefile2:152: recipe for target 'CMakeFiles/soem.dir/all' failed
mingw32-make[1]: *** [CMakeFiles/soem.dir/all] Error 2
Makefile:134: recipe for target 'all' failed
mingw32-make: *** [all] Error 2
```

Because timeradd doesn't exist in the windows header files for mingw. Do you have any suggestions to circumvent this?

ArthurKetels, 2021-05-22:
You can find the macro definition of timeradd() here : https://github.com/OpenEtherCATsociety/SOEM/blob/master/osal/win32/osal_win32.h
But it won't help you much. Do you know what the MinGW implementations are of mutex, time, and raw sockets?  That is what the SOEM abstraction layer is trying to capture. And even if the API of MinGW is similar of that of Linux, the underlying implementation can/will still be Windows. How do you want to implement nicdrv.c? It is the most important module of SOEM that sends and receives raw ethernet packets through the network card. 

As far as I know MinGW uses the winsock v2 driver. And as of windows Vista it no longer supports raw packets. It is possible to use WinPcap, and that is exactly the path nicdrv.c for windows takes. 

Also windows timing capabilities are nowhere near those of other OS. Even when you can glue it all together so it compiles, the performance will be mediocre. 

> I need to compile SOEM (through the desired usecase) with MinGW

So here is your real problem. You first need to answer the question if MinGW is capable to do what you want. You could also build with VisualC and use the resulting library in MinGW. Still lousy performance, but much quicker to implement.

MarcoMuellner, 2021-05-23:
Well, if i could choose i'd use a different compiler as well. Sadly due to customer requirements MinGW is required. Just building the Lib with the MSVC compiler probably won't work either, simply due to how the header files are included. I would probably need to move another layer between soem and the actual code to make this work i guess.

I am accutly aware that the timing under windows is abhorent in comparison to Linux and Unix systems, and the application i am building is cross plattform anyway. So yeah, this requires some more work on my side to make it work :(

MarcoMuellner, 2021-06-22:
Short update on this: 

It seems that due to the different name wrangling of the mvsc compiler and mingw, it seems impossible to link the library. So if i can't build the library with MinGW, i don't seem to be able to come to a satisfying result.

Do you have any more suggestions that i could try?

MarcoMuellner, 2021-09-07:
For posterities sake, and for future visitors to this issue, i was able to fix the problem and create a working version of SOEM, that is buildable with MinGW. It is actually rather straight forward, and if the Maintainers of this repo would be okay with it, i would make a fork, and link it with this issue?

ArthurKetels, 2021-09-07:
Of course you can make a fork, that is the whole point of open source software. And if the changes are maintainable I would not mind a pull request either. 

For curiosity's sake, how do you rate the performance of the MinGW port?

MarcoMuellner, 2021-09-07:
Well the point would be to link it here in this issue for people to find it. 

The changes are a bit crude (and they break the MSCV building of the code, which i don't mind cause i don't need that), so a pull request probably won't make much sense. Maybe at some future point this could be expanded in order for it to work properly with both compilers at the same code base.

I should note that the current application of the code isn't on something to demanding (yet), so i can't really talk much about performance (yet). From my first experiences though it seems roughly similar to what i get when i run the same code on linux.

zivke, 2023-04-19:
> Well the point would be to link it here in this issue for people to find it.
>
> The changes are a bit crude (and they break the MSCV building of the code, which i don't mind cause i don't need that), [...]

@MarcoMuellner I'm also interested in trying to build SOEM using MinGW, but I didn't find any resource on how to do so successfully.

Could you please publish the changes you've ade to SOEM in order to make it successfully compile using MinGW?

Thanks!

nakarlsson, 2023-04-19:
We’ve received a patch offline for MinGW not yet commited due to lack of time. It was from a non GitHub user provided ASIS. If you’d like you can get it and complete it here om GitHub?

zivke, 2023-04-19:
@nakarlsson Yes, I would very much like to take a look at the patch, and yes, maybe even create a PR if everything turns out well enough.