OpenEtherCATsociety/SOEM #514: failed to build with gcc 11

pingplug, 2021-05-25:
gcc -v
```
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/11.1.0/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++,d --with-isl --with-linker-hash-style=gnu --with-system-zlib --enable-__cxa_atexit --enable-cet=auto --enable-checking=release --enable-clocale=gnu --enable-default-pie --enable-default-ssp --enable-gnu-indirect-function --enable-gnu-unique-object --enable-install-libiberty --enable-linker-build-id --enable-lto --enable-multilib --enable-plugin --enable-shared --enable-threads=posix --disable-libssp --disable-libstdcxx-pch --disable-libunwind-exceptions --disable-werror gdc_include_dir=/usr/include/dlang/gdc
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.1.0 (GCC) 
```
error:
```
[  4%] Building C object CMakeFiles/soem.dir/soem/ethercatbase.c.o
[  9%] Building C object CMakeFiles/soem.dir/soem/ethercatcoe.c.o
[ 14%] Building C object CMakeFiles/soem.dir/soem/ethercatconfig.c.o
[ 19%] Building C object CMakeFiles/soem.dir/soem/ethercatdc.c.o
[ 23%] Building C object CMakeFiles/soem.dir/soem/ethercateoe.c.o
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c: In function ‘ecx_EOEsetIp’:
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:82:8: error: array subscript ‘ec_EOEt[0]’ is partly outside array bounds of ‘ec_mbxbuft’ {aka ‘unsigned char[1487]’} [-Werror=array-bounds]
   82 |    EOEp->mbxheader.address = htoes(0x0000);
      |        ^~
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:70:22: note: while referencing ‘MbxOut’
   70 |    ec_mbxbuft MbxIn, MbxOut;
      |                      ^~~~~~
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:83:8: error: array subscript ‘ec_EOEt[0]’ is partly outside array bounds of ‘ec_mbxbuft’ {aka ‘unsigned char[1487]’} [-Werror=array-bounds]
   83 |    EOEp->mbxheader.priority = 0x00;
      |        ^~
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:70:22: note: while referencing ‘MbxOut’
   70 |    ec_mbxbuft MbxIn, MbxOut;
      |                      ^~~~~~
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:90:8: error: array subscript ‘ec_EOEt[0]’ is partly outside array bounds of ‘ec_mbxbuft’ {aka ‘unsigned char[1487]’} [-Werror=array-bounds]
   90 |    EOEp->mbxheader.mbxtype = ECT_MBXT_EOE + MBX_HDR_SET_CNT(cnt); /* EoE */
      |        ^~
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:70:22: note: while referencing ‘MbxOut’
   70 |    ec_mbxbuft MbxIn, MbxOut;
      |                      ^~~~~~
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:92:8: error: array subscript ‘ec_EOEt[0]’ is partly outside array bounds of ‘ec_mbxbuft’ {aka ‘unsigned char[1487]’} [-Werror=array-bounds]
   92 |    EOEp->frameinfo1 = htoes(EOE_HDR_FRAME_TYPE_SET(EOE_INIT_REQ) |
      |        ^~
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:70:22: note: while referencing ‘MbxOut’
   70 |    ec_mbxbuft MbxIn, MbxOut;
      |                      ^~~~~~
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:95:8: error: array subscript ‘ec_EOEt[0]’ is partly outside array bounds of ‘ec_mbxbuft’ {aka ‘unsigned char[1487]’} [-Werror=array-bounds]
   95 |    EOEp->frameinfo2 = 0;
      |        ^~
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:70:22: note: while referencing ‘MbxOut’
   70 |    ec_mbxbuft MbxIn, MbxOut;
      |                      ^~~~~~
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:106:43: error: array subscript ‘ec_EOEt[0]’ is partly outside array bounds of ‘ec_mbxbuft’ {aka ‘unsigned char[1487]’} [-Werror=array-bounds]
  106 |       EOE_ip_uint32_to_byte(&ipparam->ip, &EOEp->data[data_offset]);
      |                                           ^~~~~~~~~~~~~~~~~~~~~~~~
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:70:22: note: while referencing ‘MbxOut’
   70 |    ec_mbxbuft MbxIn, MbxOut;
      |                      ^~~~~~
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:30:15: error: array subscript ‘ec_EOEt[0]’ is partly outside array bounds of ‘ec_mbxbuft’ {aka ‘unsigned char[1487]’} [-Werror=array-bounds]
   30 |    byte_ip[0] = eoe_ip4_addr4(ip); /* 4th octet */
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:70:22: note: while referencing ‘MbxOut’
   70 |    ec_mbxbuft MbxIn, MbxOut;
      |                      ^~~~~~
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:112:47: error: array subscript ‘ec_EOEt[0]’ is partly outside array bounds of ‘ec_mbxbuft’ {aka ‘unsigned char[1487]’} [-Werror=array-bounds]
  112 |       EOE_ip_uint32_to_byte(&ipparam->subnet, &EOEp->data[data_offset]);
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:70:22: note: while referencing ‘MbxOut’
   70 |    ec_mbxbuft MbxIn, MbxOut;
      |                      ^~~~~~
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:30:15: error: array subscript ‘ec_EOEt[0]’ is partly outside array bounds of ‘ec_mbxbuft’ {aka ‘unsigned char[1487]’} [-Werror=array-bounds]
   30 |    byte_ip[0] = eoe_ip4_addr4(ip); /* 4th octet */
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:70:22: note: while referencing ‘MbxOut’
   70 |    ec_mbxbuft MbxIn, MbxOut;
[... 297 lines cut ...]
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:434:34: error: array subscript ‘ec_EOEt[0]’ is partly outside array bounds of ‘ec_mbxbuft’ {aka ‘unsigned char[1487]’} [-Werror=array-bounds]
  434 |          frameinfo1 = etohs(aEOEp->frameinfo1);
      |                                  ^~
/home/share/makepkg/soem/src/SOEM/soem/ethercattype.h:540:21: note: in definition of macro ‘etohs’
  540 |   #define etohs(A) (A)
      |                     ^
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:411:15: note: while referencing ‘MbxIn’
  411 |    ec_mbxbuft MbxIn;
      |               ^~~~~
In file included from /home/share/makepkg/soem/src/SOEM/oshw/linux/oshw.h:18,
                 from /home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:18:
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:435:34: error: array subscript ‘ec_EOEt[0]’ is partly outside array bounds of ‘ec_mbxbuft’ {aka ‘unsigned char[1487]’} [-Werror=array-bounds]
  435 |          frameinfo2 = etohs(aEOEp->frameinfo2);
      |                                  ^~
/home/share/makepkg/soem/src/SOEM/soem/ethercattype.h:540:21: note: in definition of macro ‘etohs’
  540 |   #define etohs(A) (A)
      |                     ^
/home/share/makepkg/soem/src/SOEM/soem/ethercateoe.c:411:15: note: while referencing ‘MbxIn’
  411 |    ec_mbxbuft MbxIn;
      |               ^~~~~
cc1: all warnings being treated as errors
make[2]: *** [CMakeFiles/soem.dir/build.make:132: CMakeFiles/soem.dir/soem/ethercateoe.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:154: CMakeFiles/soem.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

nakarlsson, 2021-05-25:
Ok, thanks. We'll take a look at it.

nakarlsson, 2021-05-26:
I couldn't reproduce it using Ubuntu Desktop 21.04 and gcc-11,  I'm not sure why but that is less important.
Does @ArthurKetels fix fix your problem , @pingplug

pingplug, 2021-05-28:
Arch Linux use this flags in `makepkg.conf`
```
CPPFLAGS=""
CFLAGS="-march=native -mtune=native -O2 -pipe -fno-plt -fexceptions \
        -Wp,-D_FORTIFY_SOURCE=2,-D_GLIBCXX_ASSERTIONS \
        -Wformat -Werror=format-security \
        -fstack-clash-protection -fcf-protection"
CXXFLAGS="$CFLAGS"
LDFLAGS="-Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now"
```
maybe it is caused by these flags? @nakarlsson

PR #521 (2021-06-02): Adjust size of ec_EOE_t to max mailbox size
Consider size of mailbox header and frameinfo1 and 2 fields when calculating the data portion of ec_EOE_t

fix #514
commit 9de08898 (2021-06-02): Adjust size of ec_EOE_t to max mailbox size

fix #514
closed 2021-06-02 by ArthurKetels