OpenEtherCATsociety/SOEM #515: ec_DCtime = 0

DamonTTSA, 2021-05-27:
Good day

I have adapted SOEM to work on baremetal OS for a microcontroller.
I have been successful in generating a master time.
I would like to know how the ec_DCtime parameter of the ecx_context global variable should be populated?
I see that it is printed in line 106 of linux/simple_test/simple_test.c, but I cannot find where this value is set in the code.

I do not have a linux system to test out the original code, but I can't see how ec_DCtime's value is ever anything other than zero.
Could someone please advise me on how this should be set? 
Should I set it to the reference clock  (ie clock of the first DC capable slave), and if so, when?

Kind regards
DamonTTSA

ArthurKetels, 2021-05-27:
ec_DCtime is a global uin64_t variable. If VER1 is set then it is put in the context struct and is updated each time ecx_receive_processdata_group() is called. See code from ethercatmain.c below.
```
               memcpy(&le_DCtime, &(rxbuf[idx][idxstack->dcoffset[pos]]), sizeof(le_DCtime));
               *(context->DCtime) = etohll(le_DCtime);
```
This makes ec_DCtime the slave reference clock time as sampled by the latest PDO exchange. It's use is to keep the master time in relation to the slave reference clock.

DamonTTSA, 2021-05-28:
Great, thank you for such a quick and insightful response! 
That answered my question perfectly.

Kind regards
DamonTTSA

closed 2021-05-28 by DamonTTSA

joserubenh, 2022-09-05:
according to ethercat_main  ist a int64 type.
could you please confirm?

/** slave group structure */
extern ec_groupt   ec_group[EC_MAXGROUP];
extern boolean     EcatError;
extern int64       ec_DCtime;

ArthurKetels, 2022-09-06:
To be perfectly correct, the DC time of EtherCAT is uint64_t (unsigned) and is expressed in units of ns. The slave clocks are updated in increments of 9-10-11ns to be able to track a reference. In SOEM this is configured as the first DC capable slave in the current group. 

In SOEM this reference slave clock is referenced in each PDO cycle and stored in a int64_t (signed). This is formally not correct but allows users to do simple subtractions when needing time differences. This is deliberate because I do not count on everyone having his/her integer maths done perfectly. So yes, the current definition of ec_DCtime runs into problems in about 270 years. Please remind me when the time is there ;-)