OpenEtherCATsociety/SOEM #516: ecx_receive_processdata_group() of ethercatmain.c not using group argument

DamonTTSA, 2021-05-28:
Good day

In the ecx_receive_processdata_group() function in ethercatmain.c, the uint8 'group' argument is not used in a meaningful manner.
This is confirmed by the comment "just to prevent compiler warning for unused group."

It seems to me like this argument is not needed (perhaps it is deprecated?), and that there is no reason to have the ecx_receive_processdata() function as a wrapper for the ecx_receive_processdata_group() function.
Am I misunderstanding something?

How would I send process data for a specific group only?
It seems to me that I can't do that, since no matter the 'group' argument value, the process data will still be sent and received for all of the slaves.

Kind regards
Damon McLoughlin

DamonTTSA, 2021-05-28:
I have just seen that ecx_send_processdata_group( ) does use the 'group' argument in a meaningful way, so I understand how to send process data for a specific group now.

I am still uncertain about the purpose of the 'group' argument in the ecx_receive_processdata_group() function though.
Although, I see it could be that it was included for consistency's sake?

Is this correct:
You send process data for a specific slave group using ecx_send_processdata_group( groupNum, etc ), and then call ecx_receive_processdata_group, which will then obviously only receive the returning frames that were just sent, thereby only receiving process data back from the 'groupNum' slave group.

nakarlsson, 2021-05-28:
It is described in the tutorial, https://openethercatsociety.github.io/doc/soem/tutorial_8txt.html

DamonTTSA, 2021-05-28:
> It is described in the tutorial, https://openethercatsociety.github.io/doc/soem/tutorial_8txt.html

Hi @nakarlsson, thank you for your response.

I see that the tutorial describes the sending and receiving of groups via the 'group' argument.
I am just wondering whether the 'group' argument is not used at all in the receiving for the reason I gave above.

nakarlsson, 2021-05-28:
It was used in the first implementation of groups, due to fixes/changes in the groups implementation it has become unused.

DamonTTSA, 2021-05-28:
Great, thank you very much @nakarlsson.

closed 2021-05-28 by DamonTTSA