OpenEtherCATsociety/SOEM #519: some problems with "the linux sync point 50us later than DC sync"

huyu168, 2021-05-31:
I have seen that the example red_test.c set the linux sync point 50us later than DC sync
`

void ec_sync(int64 reftime, int64 cycletime , int64 *offsettime){   static int64 integral = 0;   int64 delta;   /* set linux sync point 50us later than DC sync, just as example */   delta = (reftime - 50000) % cycletime;   if(delta&gt; (cycletime / 2)) { delta= delta - cycletime; }   if(delta&gt;0){ integral++; }   if(delta&lt;0){ integral--; }   *offsettime = -(delta / 100) - (integral / 20);   gl_delta = delta;}

`
I would know can i set the linux sync point before than  DC sync? Or the reasons for the  50us later than DC sync?
and i find the picture below:
[image]
i think the sm2 event  is the linux sync point.
it seems that "set the linux sync point before than  DC sync" works.

ArthurKetels, 2021-05-31:
From the code: 
/ set linux sync point 50us later than DC sync, _just as example_
You can set the sync point as you like. F.e. if the cycle time is 1000us, you can set it at 0, 50, 500, 999. Of course -200 is equal to +800, it is a relative time. It is up to you to find the optimal offset.

The Linux sync point is slightly earlier than SM2 event. The packet first has to be assembled, transferred over the wire and accepted by the EtherCAT slave (delay depending on number of slaves before and wire length).

huyu168, 2021-06-01:
Thank you very much for your reply！

closed 2021-06-01 by huyu168