OpenEtherCATsociety/SOEM #520: DC not activating

DamonTTSA, 2021-05-31:
The `ec_dcsync0` could be used after you configured the DC of the slaves (i.e. after `ec_configdc()` if you use the `simple_test` example).

If you are sniffing the network you should see something like the following sequence:
1. read `ECT_REG_DCSYSTIME (0x0910)`
2. write start time to `ECT_REG_DCSTART0 (0x0990)` (in your case it should something in the area of the previous read value + 1_000_000)
3. write cycle time to `ECT_REG_DCCYCLE0 (0x09A0)` (i.e. 1_000_000)
4. activate the DC `sync0` by writing `3` (bit 0 and 1) to `ECT_REG_DCSYNCACT (0x0981)`

_Originally posted by @mheden in https://github.com/OpenEtherCATsociety/SOEM/issues/29#issuecomment-183268228_

ArthurKetels, 2021-05-31:
If ec_dcsync0() returns with a WKC>0 then it is working in the slave. It is directly programming the hardware in the slave ESC. But the questions is, have you programmed it the way you want? Please read https://github.com/OpenEtherCATsociety/SOEM/issues/487#issuecomment-786245585

DamonTTSA, 2021-05-31:
> If ec_dcsync0() returns with a WKC>0 then it is working in the slave. It is directly programming the hardware in the slave ESC. But the questions is, have you programmed it the way you want? Please read [#487 (comment)](https://github.com/OpenEtherCATsociety/SOEM/issues/487#issuecomment-786245585)

Hi @ArthurKetels , thank you so much for your quick response.
I read your comment and I think it will take some time for me to check that I have done all the steps you outlined.

However, just to clarify, I have a question about this part:
"After the slave clock is synchronized to the reference clock it will take only a few observations per time unit to keep it very tight. As a rule of thumb 100 observations per second is more than enough. But to get a decent lock quickly it is recommended to send at least 10.000 clock distributions as quick as possible. This is what TwinCAT does. It will take less than a second.

The best way to make sure all slave clocks are locked in is to measure the internal closed loop parameters. Register System Time Difference (0x092C:0x092F) is the obvious one to test. When it's value drops below one microsecond and stays below it then you have reached reasonable synchronization. It is the task of the application programmer (and not SOEM) to monitor this for all slaves. Only when this constraint is satisfied try to start the sync0/1 generation."

Does this mean that I should rather move the ec_dcsync0 function call out of the PRE-OP to SAFE-OP hook, and only call it later once I have done 10,000 cycles of ec_send_processdata() and ec_receive_processdata() function calls? 
So should I call ec_configdc() in PRE-OP or SAFE-OP? And then ec_sync0 in the OP state after the 10,000 cycles of process data I/O?

ArthurKetels, 2021-05-31:
The thing is when you call ec_dcsync0() before ec_configdc() that the latter will overwrite the slave reference clock. If the first sync0 did not trigger before that then it most likely will not trigger for a long time. The reason is that the first sync0 will fire at an absolute time (64bit). So make sure this absolute time is always a bit in the future. 

The correct timing to start sync0 generation is AFTER slave AND master clock synchronization, but BEFORE going to OP. And, to be sure the slave has time to validate proper master and sync0 timing, wait a few seconds before going to OP. ec_configdc() should be called at pre-OP. Then run the 10.000 cycles, then sync the master clock. There are a few weird slaves out there that require to set-up sync0 on the transition from pre-OP to safe-OP. This is violating the EtherCAT protocol specification but what can you do? It is like "I was programmed without understanding the standards and only tested with TwinCAT and now I am confused and going to crash".

DamonTTSA, 2021-06-01:
Thank you again @ArthurKetels 

I'm going to break what you said down once more, to confirm I am understanding you correctly:
1. Call ec_configdc() in PRE-OP state
2. Transition to SAFE-OP (some slaves require starting sync0 here, but this is a violation of the EtherCAT protocol spec so it is not the default).
3. Do 10,000 process data cycles 
4. Synchronise the slave and master clock by setting the slave reference clock = master clock. Do this by writing the master clock 64 bit value into the slave's local time register (ECT_REG_DCSYSTIME).
5. Call ec_dcsync0() for the reference clock slave
6. Wait for a few seconds
7. Transition to OP state

Thank you very much for your help so far!

ArthurKetels, 2021-06-01:
The order should be:
1. To add: do slave specific configurations, then map all slaves.
3.
4.
2.
5.
6.
7.

DamonTTSA, 2021-06-01:
> The order should be:
> 
> 1. To add: do slave specific configurations, then map all slaves.

By 'map all slaves' do you mean call ec_config_map()? 
My current understanding is that this causes the transition from PRE-OP to SAFE-OP, so if I do this first, then ec_configdc() will only be called in the SAFE-OP state.

Thank you @ArthurKetels

ArthurKetels, 2021-06-01:
You can suppress the automatic state change by setting:
ecx_context.manualstatechange = 1;
Of course you have to initiate the state change later by your own code.

DamonTTSA, 2021-06-01:
Okay so the order should be:

1. Do slave-specific configurations via SDO communication
2. Map PDOs for all slaves by calling ec_config_map(). Set ecx_context.manualstatechange = 1.
3. Call ec_configdc() in PRE-OP state
4. Manually call the transition to SAFE-OP (some slaves require starting sync0 here, but this is a violation of the EtherCAT protocol spec so it is not the default)
5. Do 10,000 process data cycles
6. Synchronise the slave and master clock by setting the master clock = slave reference clock
7. Call ec_dcsync0() for the reference clock slave
8. Wait for a few seconds
9. Transition to OP state

I have two more questions:

1. Why does calling ec_config_map() before ec_configdc() make a difference for synchronisation?
2. Am I correct that synchronizing the master clock to the slave clock (step 6 above) would just involve setting it equal to the reference clock? And then I would must adjust it each cycle, as shown in red_test.c?

Thank you @ArthurKetels !

ArthurKetels, 2021-06-02:
Sigh, the web interface automatically renumbered my corrected order. Sorry for the confusion. Always take #487 as reference. But you managed to come up with a very acceptable order anyway.

There are two different strategies you could follow.
a) Separate slave configuration in pre-OP and synchronisation in safe-OP.
b) Do slave configuration and synchronisation both in pre-OP and work around some details.

Method a) is the preferred one and follows the standard, method b) has a greater chance to work with all slaves out there. TwinCAT used to take method a) but changed to method b) later. To be clear both methods are allowed in the standard for the master, it is just that slaves have to accept method a) (and they not always do).

To keep things simple I would first try to implement a). 99% chance it will work.

I would place 4 after 5. The reason is that the preferred way to configure special slave settings is via the preop2safeop hook. The configuration is then automatically executed at this state transition by SOEM. But it can take quite  a while. The ec_configdc() function resets all slave clock registers to default and does the set-up to set absolute time from the master clock. From that moment the slave clocks start to drift again. The time between ec_configdc() and the rapid sync sequence should be as short as possible. Mostly it is not that critical, but better be safe than sorry. So step 4 and 6 should be close in time. 

6b) should then be to start the regular PDO transfer at the proper interval. Step 7 then takes care that the PDO transmit event is both synchronized and offset the proper way in relation to sync0. When that is all checked and proper do step 8. This way you make sure the first sync0 generated is already perfect. You give the slave no room to complain.

To answer your question 2 : Yes, but this is not the complete story. You can only set a clock once to a specific absolute time. From then on the clock starts to drift in relation to other clocks. This is true for all clocks in the world. The moment you reset your clock again you create a discontinuity. This is really bad. The best solution is to keep track of your clock drift and compensate each time you reference your clock. This is what the example in red_test.c does. Managing clocks and time events is just as difficult as data synchronisation and locking schemes.

DamonTTSA, 2021-06-02:
Thank you @ArthurKetels !

I will implement method A.

I have a few more questions.
You said this: "6b) should then be to start the regular PDO transfer at the proper interval. Step 7 then takes care that the PDO transmit event is both synchronized and offset the proper way in relation to sync0. When that is all checked and proper do step 8. "

I don't understand how I would start the regular PDO transfer here, since I have not yet transitioned to the OP state. 
Or is the idea that I transition to OP (9 & 10), before I start this PDO transfer (6b)?

Also, what is the 'proper' interval here? Is this purely application-specific, and does it relate to the SYNC0 cycle time?
So for example, let's say I want my PDO cycles to be 1ms apart. Does that mean that my SYNC0 cycle time must be 1ms and synchronised with my PDO cycles? So whenever my PDO cycle begins, it must be at the same time that a SYNC0 pulse is generated?

ArthurKetels, 2021-06-02:
The regular PDO transfer should run in its own real-time task. See the example red_test.c. All the configuration and set-up mentioned above is non-real-time and should run in the main task. A simple notification from the main task could notify the real-time task to start sending and receiving PDO packets. The right moment to do this is at step 6b. The state of the slaves are not relevant. But only at state OP the slaves can accept RxPDO's, so in pre-OP the WKC is lower. It does allow servo drives to check the timing of the master in relation to the sync0 signals. 

The slave with sync0 dictates the lower and upper bounds of timing interval. Some slaves have very wide ranges others allow only one setting. Read the manual. The master must find a setting that satisfies all slaves and the main application. This is one reason for using slave groups. If no acceptable timing can be found that satisfies all then split the slaves up in two or more groups and configure them in their separate domains. 

That is another reason why you should not try to modify a clock. If you only track the clock drift you can use it in multiple time domains each with their own drift characteristics and still synchronize to all of them. SOEM is a powerful tool if you know how to wield it.

closed 2021-07-09 by DamonTTSA

madhu-dhanasekaran, 2025-02-24:
Hi @DamonTTSA, hope you have achieved DC sync with your setup. I have tried the similar steps that you have reworked collectively with @ArthurKetels. But, in my case I do not know how exactly to verify if DC sync is achieved. Can you may be let me know how you have done it? I currently use the System Time Diff register (0x092C:0x092F) for this. But, I see it to be 0 all the time. Summary of my setup is made here: https://github.com/OpenEtherCATsociety/SOEM/issues/888

Kindly let me know your inputs. Thank you.